Koenkk/zigbee2mqtt #15193: [New device support]: 3 Phase Electricity Meter Unsupported device model 'TS0601' and manufacturer name '_TZE200_ywbltqzw'

evsmorodin, 2022-11-26:
### Link

https://www.aliexpress.com/item/1005004883393092.html

### Database entry

{"id":21,"type":"Router","ieeeAddr":"0xa4c138d5be107401","nwkAddr":27327,"manufId":4417,"manufName":"_TZE200_ywbltqzw","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i","65506":48,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_ywbltqzw","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1669465017123,"defaultSendRequestWhen":"immediate"}

### Comments

The device does not show any parameters and they cannot be controlled
[image]
[image]
[image]
[image]
[image]

evsmorodin, 2022-12-27:
This issue has not been resolved yet

Koenkk, 2023-01-29:
To add support for this device, the datapoints need to be known: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

evsmorodin, 2023-01-30:
Hello @Koenkk

[image]

- Forward Energy Total - `{code: "1"}`
- Month energy - `{code: "3"}`
- Daily energy - `{code: "4"}`
- Phase A - `{code: "6"}`
- Phase B - `{code: "7"}`
- Phase C - `{code: "8"}`
- Total Power - `{code: "9"}`
- Fault - `{code: "10"}`
- Frozen set - `{code: "11"}`
- Switch - `{code: "16"}`
- Alarm set - `{code: "17"}`
- Meter ID - `{code: "18"}`
- Clear event - `{code: "20"}`
- Froward Energy T1 - `{code: "21"}`
- Froward Energy T2 - `{code: "22"}`
- Froward Energy T3 - `{code: "23"}`
- Froward Energy T4 - `{code: "24"}`

Koenkk, 2023-01-30:
Based on this it should be possible to create an external converter as mentioned in: https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html

You can find an example which closely matches your device here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/efe19c4c7e343a065154c9373301179df21ee46d/devices/tuya.js#L3938

evsmorodin, 2023-03-02:
I'm sorry, I haven't checked yet (
the request should not be closed yet

closed 2023-04-11 by github-actions[bot]

evsmorodin, 2023-07-17:
I tried to create a converter
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_ywbltqzw',
        },
    ],
    model: 'PJ-DAC7361C',
    vendor: 'TuYa',
    description: 'Zigbee DIN 3-phase energy meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    whiteLabel: [{vendor: 'XOCA', model: 'DAC7361C'}],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.voltage(), e.current(), e.power(), e.energy(),
        tuya.exposes.switch(),
        tuya.exposes.energyWithPhase('a'), tuya.exposes.energyWithPhase('b'), tuya.exposes.energyWithPhase('c'),
        tuya.exposes.voltageWithPhase('a'), tuya.exposes.voltageWithPhase('b'), tuya.exposes.voltageWithPhase('c'),
        tuya.exposes.powerWithPhase('a'), tuya.exposes.powerWithPhase('b'), tuya.exposes.powerWithPhase('c'),
        tuya.exposes.currentWithPhase('a'), tuya.exposes.currentWithPhase('b'), tuya.exposes.currentWithPhase('c'),
        tuya.exposes.powerFactorWithPhase('a'), tuya.exposes.powerFactorWithPhase('b'), tuya.exposes.powerFactorWithPhase('c'),
        exposes.enum('fault', ea.STATE, ['clear', 'over_current_threshold', 'over_power_threshold',
            'over_voltage threshold', 'wrong_frequency_threshold']).withDescription('Fault status of the device (clear = nothing)'),
        exposes.enum('threshold_1', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
            .withDescription('State of threshold_1'),
        exposes.binary('threshold_1_protection', ea.STATE, 'ON', 'OFF')
            .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
        exposes.numeric('threshold_1_value', ea.STATE)
            .withDescription('Can be in Volt or Ampere depending on threshold setting. Setup the value on the device'),
        exposes.enum('threshold_2', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
            .withDescription('State of threshold_2'),
        exposes.binary('threshold_2_protection', ea.STATE, 'ON', 'OFF')
            .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
        exposes.numeric('threshold_2_value', ea.STATE)
            .withDescription('Setup value on the device'),
        exposes.binary('clear_fault', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Turn ON to clear last the fault'),
            exposes.text('meter_id', ea.STATE).withDescription('Meter ID (ID of device)'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [3, null, null], // Monthly, but sends data only after request
            [4, null, null], // Dayly, but sends data only after request
            [6, null, tuya.valueConverter.phaseVariant2WithPhase('a')],
            [7, null, tuya.valueConverter.phaseVariant2WithPhase('b')],
            [8, null, tuya.valueConverter.phaseVariant2WithPhase('c')],
            [9, 'power', tuya.valueConverter.raw],
            [10, 'fault', tuya.valueConverterBasic.lookup({'clear': 0, 'over_current_threshold': 1,
                'over_power_threshold': 2, 'over_voltage_threshold': 4, 'wrong_frequency_threshold': 8})],
            [11, null, null], // Frozen - strange function, in native app - nothing is clear
            [16, 'state', tuya.valueConverter.onOff],
            [17, null, tuya.valueConverter.threshold], // It's settable, but can't write converter
            [18, 'meter_id', tuya.valueConverter.raw],
            [20, 'clear_fault', tuya.valueConverter.onOff], // Clear fault
            [21, null, null], // Forward Energy T1 - don't know what this
            [22, null, null], // Forward Energy T2 - don't know what this
            [23, null, null], // Forward Energy T3 - don't know what this
            [24, null, null], // Forward Energy T4 - don't know what this
        ],
    },
};

module.exports = definition;
```
It based on this: https://github.com/Koenkk/zigbee-herdsman-converters/blob/efe19c4c7e343a065154c9373301179df21ee46d/devices/tuya.js#L2726C63-L2726C63

evsmorodin, 2023-07-20:
@Koenkk have you any ideas what's wrong?