Koenkk/zigbee2mqtt #15196: TuYa TS0012_switch_module changing states of both switches even if only one was triggered

Mickroz, 2022-11-26:
### What happened?

TuYa TS0012_switch_module 2 gang switch module - (without neutral)
When I click on one of the two switches on the dashboard (State Left or State Right) that refer to that switch, both loads turn on, not just one.
The fisical buttons work as expected, however turning them on or off from openHAB i get the exact result as above.

### What did you expect to happen?

Each of the switches (Left or Right) will only control its own load.

### Zigbee2MQTT version

 1.28.2 commit: 360a777

### Adapter firmware version

 20220219

### Adapter

Slaesh's CC2652RB stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10096449/log.txt)

entry in data/database.db
```
{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c13831b8aaa0ae","nwkAddr":49732,"manufId":4417,"manufName":"_TZ3000_jl7qyupf","powerSource":"Mains (single phase)","modelId":"TS0012","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAAAgAA"}},"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001936aa42","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001936aa42","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":918350781},"lastSeen":1669471891603,"defaultSendRequestWhen":"immediate"}
```

i did change from a Texas Instruments CC2531 to the Slaesh's CC2652RB stick without changing anything, just restarted z2m, but this switch is new, and the only thing on the network atm.

danieledwardgeorgehitchcock, 2022-11-28:
This should be fixed in the `Dev` branch of Z2M and will be available in the release at the start of next month

Mickroz, 2022-11-28:
Thanks, will wait then till release.

Mickroz, 2022-12-02:
Latest update fixes this issue, thanks 👍

closed 2022-12-02 by Mickroz