Koenkk/zigbee2mqtt #15197: Energy, power and device temperature stoped working for auara switch SSM-U01 since zigbee2mqtt version 1.23.0

pdalak, 2022-11-26:
### What happened?

Energy, power and device temperature stoped working for auara switch SSM-U01 since zigbee2mqtt version 1.23.0 

### In version 1.22.2 and earlier: 
* Metrics are showed correctly
[image]

### In version 1.23.0:
* frontend looks same, but values are not working anymore
[image]

### In version 1.24.0 and later:
* There are new fields in frontend, voltage, temperature, current. All of them are also N/A. 
[image]

### This is exact model of aqara switch:
**Zigbee Model:** lumi.switch.n0agl1
**Description:** Aqara single switch module T1 (with neutral) 
**Firmware build date:** Dec 30 2021 (Same behavior was with older FW, I do not remember exact date of previous FW)
**Manufacturer:** Xiaomi
**Model:** [SSM-U01](https://www.zigbee2mqtt.io/devices/SSM-U01.html#xiaomi-ssm-u01)

> Just a note, I have another auara smart plug SP-EUC01 and it work correctly in all versions of zigbee2mqtt

### What did you expect to happen?

Energy, power and device temperature should be reported correctly as before.

### How to reproduce it (minimal and precise)

* Pair auara switch device into zigbee2mqtt version 1.22.2 or any earlier - It works as it should. 
* Pair aquara switch to any version later than 1.23.0. Now Energy, power and device temperature are not working anymore, and zigbee2mqtt frontend shows N/A. 

### Zigbee2MQTT version

1.23.0 and any later

### Adapter firmware version

0x26660700

### Adapter

Deconz ConBee2/RaspBee2

pdalak, 2023-01-01:
Remove stale.

Koenkk, 2023-01-02:
Try updating the conbee firmware to at least [0x26720700](http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26720700.bin.GCF) and repair the device.

pdalak, 2023-01-03:
Thank You for an advice. It works now. Sorry, I did not found this advice anywhere before.

closed 2023-01-03 by pdalak