Koenkk/zigbee2mqtt #15201: Aqara E1 TRV publish set error using Sonoff Zigbee Dongle Plus V2 (Dongle-E)

xEcho1, 2022-11-26:
### What happened?

When sending MQTT Publish set to a device a error occurs. This happens randomly to a device, the effected device changes as does the number of effected devices. While a device encounters this error the frontend is still able to update manual inputs to the device.

I have considered updating the firmware of the coordinator but as of now this process seems too intimidating to me.

Sorry if this was the wrong place to post this.

### What did you expect to happen?

No error

### How to reproduce it (minimal and precise)

Send any MQTT publish to the device and the error will occur at seemingly random times.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle E (Type EZSP) | SoC: EFR32MG21

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10096577/log.txt)

tbrtje, 2022-12-11:
I am experiencing the same issue with my Conbee II.

Were you able to resolve it?

xEcho1, 2022-12-11:
> I am experiencing the same issue with my Conbee II.
> 
> Were you able to resolve it?

Yes but i dont think this would help you.

https://community.home-assistant.io/t/zigbee2mqtt-aqara-e1-trv-publish-set-error-using-sonoff-zigbee-dongle-plus-v2-dongle-e/493393/4?u=xecho1

closed 2022-12-11 by xEcho1