Koenkk/zigbee2mqtt #15205: Zigbee2mqtt freeze

DrArut, 2022-11-27:
### What happened?

After normal operation (after update to Zigbee2MQTT 1.28.2-1), zigbee2mqtt hangs after a while. Restarting the plugin helps. But after a while it crashes again. 
node index.js load one kernel of CPU on 100%
My system information and logs:

**Raspberry Pi 4 2Gb**
Home Assistant 2022.11.4
Supervisor 2022.11.2
Operating System 9.3
Frontend 20221108.0 - latest
Mosquitto broker 6.1.3
Zigbee2MQTT 1.28.2-1

**Zigbee2Mqtt logs:**
error 2022-11-27 09:28:28: Failed to setup reporting for '0x000d6f00131c5a57' - Error: Bind 0x000d6f00131c5a57/1 genOnOff from '0x00124b002291d5f0/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
error 2022-11-27 09:28:38: Failed to setup reporting for '0xcc86ecfffe56af3a' - Error: Bind 0xcc86ecfffe56af3a/1 genOnOff from '0x00124b002291d5f0/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
error 2022-11-27 10:53:38: Publish 'set' 'state' to 'LED strip koridor' failed: 'Error: Command 0xa4c1386144f1c384/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2022-11-27 10:59:03: Publish 'set' 'state' to 'LED strip koridor' failed: 'Error: Command 0xa4c1386144f1c384/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

### What did you expect to happen?

to stop freezing

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

20220301

### Adapter

Egony Stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10097770/log.txt)

DrArut, 2022-11-27:
Update log. Now i see:
Zigbee2MQTT:error 2022-11-27 14:44:32: Not connected to MQTT server!
Zigbee2MQTT:error 2022-11-27 14:45:33: Not connected to MQTT server!
Zigbee2MQTT:error 2022-11-27 14:46:35: Not connected to MQTT server!
Zigbee2MQTT:error 2022-11-27 14:46:35: Cannot send message: topic: 'zigbee2mqtt/Moes thermostat child', payload: '{"child_lock":"UNLOCK","current_heating_setpoint":25,"deadzone_temperature":0,"heat":"OFF","linkquality":98,"local_temperature":25.5,"local_temperature_calibration":-1,"max_temperature_limit":27,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","running_state":"idle","sensor":"IN","system_mode":"heat"}
Zigbee2MQTT:error 2022-11-27 14:46:35: Not connected to MQTT server!
Zigbee2MQTT:error 2022-11-27 14:47:31: Not connected to MQTT server!

ingar195, 2022-11-27:
Have you checked if your MQTT server is actually running? 
Recommend [MQTT explorer](http://mqtt-explorer.com/) to check 

If ist running recheck you credentials

DrArut, 2022-11-27:
Yes, MQTT server work. I tested with mqtt explorer, and work very well.  I recheck my credentials, all OK.

DrArut, 2022-11-27:
After restart i see new error
Zigbee2MQTT:error 2022-11-27 20:59:19: Failed to disable reporting for '0xcc86ecfffe933808' - Error: Unbind 0xcc86ecfffe933808/1 genOnOff from '0x00124b002291d5f0/1' failed (AREQ - ZDO - unbindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
Zigbee2MQTT:error 2022-11-27 20:59:25: Failed to disable reporting for '0xa4c138add1f44d1c' - Error: ConfigureReporting 0xa4c138add1f44d1c/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":65535,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'NOT_FOUND')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Report.setupReporting (/app/lib/extension/legacy/report.ts:107:25)
    at Report.start (/app/lib/extension/legacy/report.ts:179:17)
    at Controller.callExtensions (/app/lib/controller.ts:314:17)
    at Controller.start (/app/lib/controller.ts:153:9)
    at start (/app/index.js:109:5)

DrArut, 2022-11-27:
**Log with zigbee_herdsman_debug**

2022-11-27T17:27:14.801Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,21,36,1,21,192,1,1,10,0,111,0,30,11,24,108,1,7,0,0,35,136,146,22,43,227]
2022-11-27T17:27:14.802Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,5,0,2,0,7,0,105,188,28,110]
2022-11-27T17:27:14.803Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,10,0,21,192,1,1,0,65,0,230,133,134,0,0,5,0,2,0,7,0,105,188,28,110]
2022-11-27T17:27:14.803Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,10,0,21,192,1,1,0,65,0,230,133,134,0,0,5,0,2,0,7,0,105,188,28] - 110
2022-11-27T17:27:14.804Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":10,"srcaddr":49173,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":65,"securityuse":0,"timestamp":8816102,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[0,2,0,7,0]}}
2022-11-27T17:27:14.805Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":2,"manufacturerCode":null,"commandIdentifier":0},"Payload":[{"attrId":7}],"Command":{"ID":0,"name":"read","parameters":[{"name":"attrId","type":33}],"response":1}},"address":49173,"endpoint":1,"linkquality":65,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-11-27T17:27:14.807Z zigbee-herdsman:controller:endpoint ReadResponse 0x847127fffe1dc07d/1 genTime({"localTime":722899635}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-11-27T17:27:14.811Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-27T17:27:14.824Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-11-27T17:27:14.824Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-11-27T17:27:14.824Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-11-27T17:27:14.825Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-11-27T17:27:14.825Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-27T17:27:14.827Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,225,1,109,74]
2022-11-27T17:27:14.828Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,225,1,109,74]
2022-11-27T17:27:14.828Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [225,1,109] - 74
2022-11-27T17:27:14.828Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":225,"endpoint":1,"transid":109}
2022-11-27T17:27:14.829Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-27T17:27:14.829Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x9035eafffeb5e72e:3914,225,0)
2022-11-27T17:27:14.849Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,111,169]
2022-11-27T17:27:14.849Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,111,169]
2022-11-27T17:27:14.850Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,111] - 169
2022-11-27T17:27:14.850Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":111}
2022-11-27T17:27:14.851Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-27T17:27:14.854Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x847127fffe1dc07d:49173/1 (0,0,896)
2022-11-27T17:27:14.854Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":49173,"destendpoint":1,"srcendpoint":1,"clusterid":10,"transid":112,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[24,108,1,7,0,0,35,136,146,22,43]}}
2022-11-27T17:27:14.855Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,21,36,1,21,192,1,1,10,0,112,0,30,11,24,108,1,7,0,0,35,136,146,22,43,252]
2022-11-27T17:27:14.874Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-11-27T17:27:14.875Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-11-27T17:27:14.875Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-11-27T17:27:14.876Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-11-27T17:27:14.876Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-27T17:27:14.887Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,33,68,129,0,0,0,239,74,15,1,1,0,47,0,109,160,134,0,0,13,9,127,1,0,0,18,2,0,4,0,0]
2022-11-27T17:27:14.888Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,33,68,129,0,0,0,239,74,15,1,1,0,47,0,109,160,134,0,0,13,9,127,1,0,0,18,2,0,4,0,0]
2022-11-27T17:27:14.889Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,251,74,15,29,231]
2022-11-27T17:27:14.890Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,33,68,129,0,0,0,239,74,15,1,1,0,47,0,109,160,134,0,0,13,9,127,1,0,0,18,2,0,4,0,0,0,251,74,15,29,231]
2022-11-27T17:27:14.890Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,74,15,1,1,0,47,0,109,160,134,0,0,13,9,127,1,0,0,18,2,0,4,0,0,0,251,74,15,29] - 231
[... cut ...] :47,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-11-27T17:27:15.694Z zigbee-herdsman:controller:endpoint DefaultResponse 0x9035eafffeb5e72e/1 61184(1, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-11-27T17:27:15.703Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

closed 2023-01-04 by github-actions[bot]