Koenkk/zigbee2mqtt #15211: Philips Hue Switch V2 on-hold actions

hpio, 2022-11-27:
### What happened?

I have got the Philips Hue Switch V2 model `RWL022` running `2.45.2_hF4400CA` firmware. I noticed that `on_hold` action incorrectly sends `on_press` event too. This causes all sorts of issues with automations as triggers ones that listen to `on_press` events too

Also, the issue affects all buttons, not only the on/off one

The issue was previously reported here https://github.com/Koenkk/zigbee2mqtt/issues/10233

### What did you expect to happen?

I expected just the `on_hold` payload to be produced without `on_press` being sent

### How to reproduce it (minimal and precise)

Press and hold a button, instead of `on_hold` event being produced there's also `on_press`

```
fo 2022-11-27 14:26:45MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch', payload '{"action":"on_press","battery":100,"linkquality":200,"update":{"state":"idle"},"update_available":null}'
info 2022-11-27 14:26:45MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch/action', payload 'on_press'
info 2022-11-27 14:26:46MQTT publish: topic 'zigbee2mqtt/Bedroom spotlight 1', payload '{"brightness":253,"color_mode":"color_temp","color_temp":187,"linkquality":204,"state":"ON","update":{"state":"idle"},"update_available":null}'
info 2022-11-27 14:26:46MQTT publish: topic 'zigbee2mqtt/light_bedroom_spotlights', payload '{"brightness":253,"color_mode":"color_temp","color_temp":187,"state":"ON"}'
info 2022-11-27 14:26:46MQTT publish: topic 'zigbee2mqtt/Bedroom spotlight 2', payload '{"brightness":253,"color_mode":"color_temp","color_temp":187,"linkquality":116,"state":"ON","update":{"state":"idle"},"update_available":null}'
info 2022-11-27 14:26:46MQTT publish: topic 'zigbee2mqtt/Bedroom spotlight 4', payload '{"brightness":253,"color_mode":"color_temp","color_temp":187,"linkquality":200,"state":"ON","update":{"state":"idle"},"update_available":null}'
info 2022-11-27 14:26:46MQTT publish: topic 'zigbee2mqtt/Bedroom spotlight 3', payload '{"brightness":253,"color_mode":"color_temp","color_temp":187,"linkquality":204,"state":"ON","update":{"state":"idle"},"update_available":null}'
info 2022-11-27 14:26:46MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch', payload '{"action":"on_hold","action_duration":0.794,"battery":100,"linkquality":200,"update":{"state":"idle"},"update_available":null}'
info 2022-11-27 14:26:46MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch/action', payload 'on_hold'
info 2022-11-27 14:26:46MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch', payload '{"action":"on_hold_release","battery":100,"linkquality":204,"update":{"state":"idle"},"update_available":null}'
info 2022-11-27 14:26:46MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch/action', payload 'on_hold_release'
```

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EFR32MG21

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10098508/log.txt)

zegl, 2022-11-30:
@hpio Could you please include debug logs of when you're just pressing (not holding) the button?

piotrhryszko-img, 2022-12-02:
apologies @zegl for delayed response, here are requested debug logs from a single press of a button 
```
debug 2022-12-02 09:03:36Received Zigbee message from 'Bedroom dimmer switch', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info 2022-12-02 09:03:36MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch', payload '{"action":"on_press","battery":100,"linkquality":200,"update":{"state":"idle"},"update_available":null}'
info 2022-12-02 09:03:36MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch/action', payload 'on_press'
debug 2022-12-02 09:03:36Received Zigbee message from 'Bedroom dimmer switch', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info 2022-12-02 09:03:36MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch', payload '{"action":"on_press_release","battery":100,"linkquality":200,"update":{"state":"idle"},"update_available":null}'
info 2022-12-02 09:03:36MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch/action', payload 'on_press_release'
debug 2022-12-02 09:03:37Received Zigbee message from 'Bedroom dimmer switch', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID 0
```

hpio, 2022-12-02:
apologies for delayed response @zegl , here are logs from a single press event 
```
debug 2022-12-02 09:07:57Received Zigbee message from 'Bedroom dimmer switch', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info 2022-12-02 09:07:57MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch', payload '{"action":"on_press","battery":100,"linkquality":200,"update":{"state":"idle"},"update_available":null}'
info 2022-12-02 09:07:57MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch/action', payload 'on_press'
debug 2022-12-02 09:07:57Received Zigbee message from 'Bedroom dimmer switch', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info 2022-12-02 09:07:57MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch', payload '{"action":"on_press_release","battery":100,"linkquality":196,"update":{"state":"idle"},"update_available":null}'
info 2022-12-02 09:07:57MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch/action', payload 'on_press_release'
debug 2022-12-02 09:07:58Received Zigbee message from 'Bedroom dimmer switch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
```

and for comparison taken just now logs from long hold of the same button
```
debug 2022-12-02 09:08:39Received Zigbee message from 'Bedroom dimmer switch', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info 2022-12-02 09:08:39MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch', payload '{"action":"on_press","battery":100,"linkquality":192,"update":{"state":"idle"},"update_available":null}'
info 2022-12-02 09:08:39MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch/action', payload 'on_press'
debug 2022-12-02 09:08:40Received Zigbee message from 'Bedroom dimmer switch', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":8,"type":1,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info 2022-12-02 09:08:40MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch', payload '{"action":"on_hold","action_duration":0.759,"battery":100,"linkquality":192,"update":{"state":"idle"},"update_available":null}'
info 2022-12-02 09:08:40MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch/action', payload 'on_hold'
debug 2022-12-02 09:08:40Received Zigbee message from 'Bedroom dimmer switch', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":10,"type":3,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info 2022-12-02 09:08:40MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch', payload '{"action":"on_hold_release","battery":100,"linkquality":196,"update":{"state":"idle"},"update_available":null}'
info 2022-12-02 09:08:40MQTT publish: topic 'zigbee2mqtt/Bedroom dimmer switch/action', payload 'on_hold_release'
```

hpio, 2023-01-03:
this is still an issue, anything I can do to help debug this?

zegl, 2023-01-04:
Hey. It's hard to say if this is working as intended or not, it seems like the correct events are sent:

Short press: `on_press` -> `on_press_release`
Long hold: `on_press` -> `on_hold` -> `on_hold_release`

To separate between the two, you can hook up your automations to the `on_press_release` and `on_hold` / `on_hold_release` events.

kapciuch, 2023-01-30:
I've got the same problem. If I hold it would do 4 actions:
Press
Hold
Brightness Down
Release 

All together!
Please 🙏

d-EScape, 2023-03-05:
The issue still exists and also applies to model 324131137411 (not just V2).
I don't think that this is working as intended, because it will work just fine in legacy mode.

closed 2023-04-14 by github-actions[bot]