Koenkk/zigbee2mqtt #15212: [New device support]: NOUS A1Z 2putqrmw

Mjauu, 2022-11-27:
### Link

https://nous.technology/product/a1z-1.html

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0xa4c138046a8b95e8","nwkAddr":54401,"manufId":4417,"manufName":"_TZ3000_2putqrmw","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u0000\u0000\u0000\u0012&D\u0015+\u0012m\u0002\u0000\u0000\u0012�N\u0015+\u0012t(\u0016+\u0003","65506":54,"65508":0,"stackVersion":0,"dateCode":"","appVersion":192,"hwVersion":1,"zclVersion":3}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":235,"rmsCurrent":8132,"activePower":1915}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,209]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c3c755","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c3c755","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c3c755","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1669558336514,"defaultSendRequestWhen":"immediate"}

### Comments

**Zigbee Model**: TS011F
**Zigbee Manufacturer**: _TZ3000_2putqrmw

I bought the device on 23.11.12 directly from [NOUS](https://nous.technology/product/a1z-1.html). It looks to me that it is a upgraded version, because the NOUS A1Z has already been added in the past, see #11863.

For this particular model the manufacturer entry is missing in [tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/defa34ef114514788e7236739dca684d6318f6ee/devices/tuya.js#L18-L25), but I tried exactly [this converer](https://github.com/Koenkk/zigbee2mqtt/issues/11863#issuecomment-1068944350) and the device worked immediately. See home assistant entities card below.

[image]

The device come to me with a firmware 1.0.13 and thanks to OTA firmware upgarde feature I could update it to 3.0.0. Many thanks for all the work you put in Zigbee2Mqtt!

### External converter

```shell
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const globalStore = require('zigbee-herdsman-converters/lib/store')
const ota = require('zigbee-herdsman-converters//lib/ota');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['TS011F'],
        model: 'A1Z_2putqrmw',
        description: 'Smart plug (with power monitoring)',
        vendor: 'Nous',
        ota: ota.zigbeeOTA,
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory,
            fz.ts011f_plug_indicator_mode, fz.ts011f_plug_child_mode],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode, tz.ts011f_plug_child_mode],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage'),
            exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
                .withDescription('Plug LED indicator mode'), e.child_lock()],
    },
];
```

svb91, 2022-11-27:
Having the same device (already was about to add the issue here).
Your converter does not update the power and current value on my side - otherwise it worked well for me.

Mjauu, 2022-11-27:
I'm not sure if it's the same problem, but I had null in the following entities in firmware version 1.0.13:
- power
- current
- energy

I then did a firmware update to version 3.0.0 using the OTA feature from Zigbee2Mqtt and everything works very well.

See the graph for my laptop power supply. When I unplug the cable, the power drops to zero (17:42, 17:53).
[image]

At this point I can not say anything more. I used the device only a few hours. Maybe try to update the firmware version as well?

numerobis08, 2022-11-29:
Hello , I'm facing same issue I have the same model and device is not recognized into Zigbee2mqtt. I'm lost on how you made it work , can you help me ? what the change to be applied in order to make it work ? 
Will it be updated in the next update of zigbee2mqtt ? 
Thank you in advance

svb91, 2022-11-29:
@Mjauu: Firmeware-Update did the trick. Thanks!

@numerobis08: I'm using the Edge-Branch of Zigbee2MQTT and created a custom converter. You'll find the code a few comments above and [here](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html) how you can add it to your instance of Z2M.

Mjauu, 2022-12-03:
NOUS A1Z 2putqrmw is supported since 1.28.3 release.

See https://github.com/Koenkk/zigbee2mqtt/issues/7405#issuecomment-1334107151

closed 2022-12-03 by Mjauu

vKnmnn, 2023-09-06:
Just came by to say that i have a device 2putqrmw that is not a NUOS A1Z, but a  [ts011F](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1)