Koenkk/zigbee2mqtt #15222: Zigbee2mqtt crashes during interview and has problem restarting

tkarlsson, 2022-11-28:
### What happened?

Diring interview of Aquara Thermostat E1 Zigbee2mqtt crashed and then failed to restart. After restarting 2-3 times the system comes up.

### Zigbee2MQTT version

1.28.2 commit: 360a777

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Debug log

**Error during inclusion**
Zigbee2MQTT:debug 2022-11-28 12:39:07: Received Zigbee message from '0x54ef44100057ba30', type 'attributeReport', cluster 'genBasic', data '{"appVersion":25,"modelId":"lumi.airrtc.agl001"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-11-28 12:39:07: Skipping message, definition is undefined and still interviewing
Error: Failure send setManufacturerCode:{"type":"Buffer","data":[69,0,1,21,0,95,17]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:512:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

**Error during restart**
Zigbee2MQTT:debug 2022-11-28 12:43:33: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:error 2022-11-28 12:43:36: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-11-28 12:43:36: Failed to start zigbee
Zigbee2MQTT:error 2022-11-28 12:43:36: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-28 12:43:36: Exiting...
Zigbee2MQTT:error 2022-11-28 12:43:36: Error: Failure send getNetworkParameters:{"type":"Buffer","data":[23,0,1,40,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:512:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

kirovilya, 2022-11-28:
try using dev|edge version of z2m

tkarlsson, 2022-11-28:
Dev doesn't crash, however, still get a error:

Zigbee2MQTT:error 2022-11-28 13:17:32: Failed to configure '0x54ef44100057ba30', attempt 1 (Error: Read 0x54ef44100057ba30/1 aqaraOpple([1034], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 22198 - 1 - 3 - 64704 - 1 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:564:17)
    at processTimers (node:internal/timers:507:7))
Zigbee2MQTT:info  2022-11-28 13:17:32: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x54ef44100057ba30"},"error":"Failed to configure (Read 0x54ef44100057ba30/1 aqaraOpple([1034], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 22198 - 1 - 3 - 64704 - 1 after 10000ms))","status":"error","transaction":"usdo7-1"}'
Zigbee2MQTT:warn  2022-11-28 13:17:32: Device '0x54ef44100057ba30' left the network

closed 2023-01-05 by github-actions[bot]