Koenkk/zigbee2mqtt #15223: [New device support]: AxoX / EGLO 12251 bulb

miksen, 2022-11-28:
### Link

https://www.eglo.com/en/bulb-12251.html

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0xa4c13883d0df8965","nwkAddr":46801,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorMode":2,"colorTemperature":370}},"genIdentify":{"attributes":{"identifyTime":0}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1669639049824,"defaultSendRequestWhen":"immediate"}

### Comments

This is an RGB light bulb, but it's recognized as a different bulb, which is white-only:

```
info  2022-11-17 13: 45: 34: Successfully interviewed '0xa4c13883d0df8965', device has successfully been paired
info  2022-11-17 13: 45: 34: Device '0xa4c13883d0df8965' is supported, identified as: AwoX LED white (33951/33948)
```

The converter below seems to fix it.

### External converter

```shell
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const extend = require("zigbee-herdsman-converters/lib/extend");
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  zigbeeModel: ["TLSR82xx"],
  model: "12251",
  vendor: "Awox",
  description: "LED bulb with RGB color and temperature",
  extend: extend.light_onoff_brightness_colortemp_color({
    colorTempRange: [153, 370],
    supportsHS: true,
  }),
  fingerprint: [
    {
      type: "Router",
      manufacturerName: "AwoX",
      modelID: "TLSR82xx",
      endpoints: [
        {
          ID: 1,
          profileID: 260,
          inputClusters: [0, 3, 4, 5, 6, 8, 768, 4096, 64599],
          outputClusters: [6],
        },
        {
          ID: 3,
          profileID: 4751,
          inputClusters: [65360, 65361],
          outputClusters: [65360, 65361],
        },
        { ID: 242, profileID: 41440, inputClusters: [], outputClusters: [33] },
      ],
    },
  ],
};

module.exports = definition;
```

### Supported color modes

color temperature, color rgb

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}

miksen, 2022-12-01:
Updated to v1.28.3, force removed and re-paired. The new database.db entry -- now with `"devId": 269,` is:

{"id":6,"type":"Router","ieeeAddr":"0xa4c13883d0df8965","nwkAddr":46801,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"appVersion":0,"hwVersion":0}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1669919213074,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-12-01:
Is it detected as supported now?

miksen, 2022-12-01:
No, i tried first with my converter disabled, and then it still came back in as a while led.

I added the devid to my converter above, and that is working fine.

Koenkk, 2022-12-02:
The device should be now match with https://github.com/Koenkk/zigbee-herdsman-converters/blob/4f060b03c14f85b75da5e83804eb4553c5cac3f9/devices/awox.js#L114 , I can't see why it is not. Can you add the deviceIDs to your external converter step by step and see where it goes wrong?

runejuhl, 2022-12-03:
Saw this issue yesterday and just tried updating to latest zigbee2mqtt and everything seems to be in order now. Thanks!

Koenkk, 2022-12-07:
@miksen can you provide the database.db entries of both the bulb detected as `33943/33944/33946` and `12251 `?

miksen, 2022-12-07:
After another reboot of z2m, the `12251` seems to have changed to `33943/33944/33946`. 

I'll close this, thanks for all the replies!

closed 2022-12-07 by miksen