Koenkk/zigbee2mqtt #15224: xiaomy gateway 3 fails to start z2m

ju0k, 2022-11-28:
### What happened?

Hello, I am neew to HA and trying to run zigbee2mqtt on xiaomi gateway 3 but the log shows that it hangs on stage "Starting zigbee-herdsman (0.14.76)"

'[15:32:18] INFO: Preparing to start... [15:32:18] INFO: Socat not enabled [15:32:18] INFO: Starting Zigbee2MQTT... Zigbee2MQTT:info 2022-11-28 15:32:20: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-28.15-32-20' filename: log.txt Zigbee2MQTT:info 2022-11-28 15:32:20: Starting Zigbee2MQTT version 1.28.2-dev (commit #9289a26) Zigbee2MQTT:info 2022-11-28 15:32:20: Starting zigbee-herdsman (0.14.76)'

### What did you expect to happen?

I want to run z2m.

### Zigbee2MQTT version

1.22.2-1

### Adapter firmware version

2022.11.2

### Adapter

xiaomy gateway 3

jjpvdpoel, 2022-11-28:
Did you use this https://github.com/AlexxIT/XiaomiGateway3 ?

ju0k, 2022-11-28:
> Did you use this https://github.com/AlexxIT/XiaomiGateway3 ?

Yes, I did.

closed 2023-01-05 by github-actions[bot]