Koenkk/zigbee2mqtt #15227: [New device support]: Samotech SM308-2ch w/ power monitoring

kurtis-poole, 2022-11-28:
### Link

https://www.samotech.co.uk/products/zigbee-switch-sm308-2ch/

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x9035eafffeef8a34","nwkAddr":30043,"manufId":4107,"manufName":"Samotech","powerSource":"Mains (single phase)","modelId":"SM308-2CH","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,1794,2820,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SM308-2CH","manufacturerName":"Samotech","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.5.3_r2"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0}},"haElectricalMeasurement":{"attributes":{"rmsCurrent":5,"activePower":1,"rmsVoltage":2363,"acVoltageMultiplier":1,"acVoltageDivisor":10,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":10}},"seMetering":{"attributes":{"currentSummDelivered":[0,3906704],"multiplier":1,"divisor":3600000}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":2820,"attrId":1288,"minRepIntval":10,"maxRepIntval":3600,"repChange":10},{"cluster":2820,"attrId":1285,"minRepIntval":10,"maxRepIntval":3600,"repChange":1},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6,1794,2820,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"genOnOff":{"attributes":{"onOff":0}},"haElectricalMeasurement":{"attributes":{"rmsCurrent":0,"activePower":0,"rmsVoltage":2361}},"seMetering":{"attributes":{"currentSummDelivered":[0,3906324]}}},"binds":[{"cluster":7,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69ecc","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.5.3_r2","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1717205429},"lastSeen":1669651239045,"defaultSendRequestWhen":"immediate"}

### Comments

Switch functionality and overall energy consumption works. Is there a way to force per endpoint figures to overall?

I'm stuck at getting the power figures to show as overall figures rather than needing to stick them under a specific endpoint. The device confusingly reports them per-endpoint but it can only send figures summed for the whole device.

I've tried to take them from just the first endpoint, and I get eg. state is reported as below with the below converter:

```
{
    "current_l1": 0.22,
    "current_l2": 224,
    "energy": 1.09,
    "linkquality": 93,
    "power_l1": 41.1,
    "power_l2": 411,
    "state": "ON",
    "state_1": "OFF",
    "state_l1": "ON",
    "state_l2": "ON",
    "voltage_l1": 235.8,
    "voltage_l2": 2356,
    "current": null,
    "voltage": null,
    "power": null
}
```

This shows up in z2mqtt but other than the overall energy, it means the voltage/current/power is under l1. 

I'm happy to PR with the below converter if this is as close as it can get. Thanks!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['SM308-2CH'],
        model: 'SM308-2CH',
        vendor: 'Samotech',
        description: 'Zigbee in wall smart switch, two channel',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report],
        toZigbee: [tz.on_off],
        exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.energy(), e.power().withEndpoint('l1'), e.current().withEndpoint('l1'), e.voltage().withEndpoint('l1')],
        endpoint: (device) => {
            return { 'l1': 1, 'l2': 2 };
        },
        meta: { multiEndpoint: true },
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1)
            const endpoint2 = device.getEndpoint(2)
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genBasic', 'genIdentify', 'genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['genBasic', 'genIdentify', 'genOnOff']);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint1);
            await reporting.activePower(endpoint1);
            await reporting.rmsCurrent(endpoint1, { min: 10, change: 10 });
            await reporting.rmsVoltage(endpoint1, { min: 10 });
            await reporting.readMeteringMultiplierDivisor(endpoint1);
            await reporting.currentSummDelivered(endpoint1);
        },
    },
];
```

Koenkk, 2022-11-28:
Try removing `meta: { multiEndpoint: true },` and change `fz.on_off` to `fz.on_off_force_multiendpoint`

amkurtis, 2022-11-28:
That all works! Thanks!