Koenkk/zigbee2mqtt #15231: zigbee2mqtt stop working

zombie241088, 2022-11-28:
### What happened?

Addon suddenly stopped working after working fine for several weeks.
Does anyone know what the problem could be?
ZHA recognizes the dongle but it seems that zigbee2mqtt does not.

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

2039602615

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10107086/log.txt)

mh166, 2022-11-28:
Hi @zombie241088,
how are you running zigbee2mqtt? If you are able, please provide the contents of the `database.db` file. It seems that it's contents are corrupted, leading the startup process to abort loading the data.

zombie241088, 2022-11-29:
Hi @mh166,
I'm running this on Home Assistant OS as an add-on.
How can I share its content?

zombie241088, 2022-12-01:
Any update anyone?

closed 2023-01-08 by github-actions[bot]