Koenkk/zigbee2mqtt #15234: Lidl HG08164

mreldd, 2022-11-28:
### What happened?

Activating the Smart Button leaves the following 2lines as debug message at the log

Received Zigbee message from 'XXXXXXXMYDEVICE', type 'raw', cluster 'genOnOff', data '{"data":[1,36,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
No converter available for 'HG08164' with cluster 'genOnOff' and type 'raw' and data '{"data":[1,36,253,0],"type":"Buffer"}'

### What did you expect to happen?

I would like to receive the correct action in HA

### How to reproduce it (minimal and precise)

just click the button 

### Zigbee2MQTT version

 1.28.2

### Adapter firmware version

20180507

### Adapter

CC2531

JochemLangerak, 2022-12-12:
Hi, Ie got the same issue.

lzahradil, 2022-12-19:
{
    **"action": null,**
    "battery": 58,
    "last_seen": "2022-12-19T22:50:13+01:00",
    "linkquality": 52
}

ErikT80, 2022-12-29:
Same issue here. Same adapter and firmware version

closed 2023-02-06 by github-actions[bot]