Koenkk/zigbee2mqtt #15235: Ikea OTA is just not working

balihb, 2022-11-29:
### What happened?

trying to update Ikea stuff via a Xiaomi GW is just now working. Or am I doing something wrong? (sorry if I do)
[image]
[image]

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

6.7.10.0 build 423

### Adapter

https://github.com/zvldz/mgl03_fw/tree/main/firmware/custom/mgl03_1.5.0_0102_mod20210816

### Debug log

debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"0x14b457fffe4dcf80","transaction":"hmwma-1"}'
info Updating '0x14b457fffe4dcf80' to latest firmware
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating '0x14b457fffe4dcf80' to latest firmware","meta":{"device":"0x14b457fffe4dcf80","status":"update_in_progress"},"type":"ota_update"}'
debug Updating to latest '0x14b457fffe4dcf80' (TRADFRI remote control)
debug Using endpoint '1'
debug Update of '0x14b457fffe4dcf80' failed (Error: Device didn't respond to OTA request)
info MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4dcf80', payload '{"battery":74,"linkquality":255,"update":{"state":"available"},"update_available":true}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of '0x14b457fffe4dcf80' failed (Device didn't respond to OTA request)","meta":{"device":"0x14b457fffe4dcf80","status":"update_failed"},"type":"ota_update"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x14b457fffe4dcf80"},"error":"Update of '0x14b457fffe4dcf80' failed (Device didn't respond to OTA request)","status":"error","transaction":"hmwma-1"}'
error Update of '0x14b457fffe4dcf80' failed (Device didn't respond to OTA request)
debug Error: Device didn't respond to OTA request at requestOTA (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:196:15) at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:283:21) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:256:21)
debug Received Zigbee message from '0x14b457fffe4dcf80', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":60}' from endpoint 1 with groupID 0
info MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4dcf80', payload '{"battery":60,"linkquality":252,"update":{"state":"available"},"update_available":true}'
debug Received Zigbee message from '0x14b457fffe4dcf80', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 58508
info MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4dcf80', payload '{"action":"toggle","battery":60,"linkquality":255,"update":{"state":"available"},"update_available":true}'
debug Received Zigbee message from '0x14b457fffe4dcf80', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 58508
info MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4dcf80', payload '{"action":"toggle","battery":60,"linkquality":184,"update":{"state":"available"},"update_available":true}'
debug Received Zigbee message from '0x14b457fffe4dcf80', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 58508
info MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4dcf80', payload '{"action":"toggle","battery":60,"linkquality":228,"update":{"state":"available"},"update_available":true}'
debug Received Zigbee message from '0x14b457fffe4dcf80', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 58508
info MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4dcf80', payload '{"action":"toggle","battery":60,"linkquality":248,"update":{"state":"available"},"update_available":true}'
debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"0x14b457fffe4dcf80","transaction":"hmwma-2"}'
info Updating '0x14b457fffe4dcf80' to latest firmware
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating '0x14b457fffe4dcf80' to latest firmware","meta":{"device":"0x14b457fffe4dcf80","status":"update_in_progress"},"type":"ota_update"}'
debug Updating to latest '0x14b457fffe4dcf80' (TRADFRI remote control)
debug Using endpoint '1'
debug Update of '0x14b457fffe4dcf80' failed (Error: Device didn't respond to OTA request)
info MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4dcf80', payload '{"battery":60,"linkquality":248,"update":{"state":"available"},"update_available":true}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of '0x14b457fffe4dcf80' failed (Device didn't respond to OTA request)","meta":{"device":"0x14b457fffe4dcf80","status":"update_failed"},"type":"ota_update"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x14b457fffe4dcf80"},"error":"Update of '0x14b457fffe4dcf80' failed (Device didn't respond to OTA request)","status":"error","transaction":"hmwma-2"}'
error Update of '0x14b457fffe4dcf80' failed (Device didn't respond to OTA request)
debug Error: Device didn't respond to OTA request at requestOTA (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:196:15) at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:283:21) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:256:21)
debug Saving state to file /app/data/state.json
debug Saving state to file /app/data/state.json

danieledwardgeorgehitchcock, 2022-11-29:
Are you waking the device before trying to apply the update? The above error only usually occurs when the device is in a sleep state...

balihb, 2022-11-29:
yes I do. I had access to another gw and ikea switch pairing with the same
error and I have this same error for more then a year (never worked). the
update works with the ikea gw with the same version of switch.

balihb, 2023-01-04:
the problem is still present? can I somehow give any more information that would help?

closed 2023-02-12 by github-actions[bot]

Suschibomber, 2024-02-01:
I have the same Error. Any solutions how to update via Z2M?

PlasticRhino, 2024-03-08:
> I have the same Error. Any solutions how to update via Z2M?

Yes - make sure you are using Zigbee2MQTT 1.36 (or higher if available).
Power cycle the mains adapter and then immediately try the update. It works for me.

Suschibomber, 2024-03-08:
I've finally done it. The secret, which is nowhere on the Internet, is that the remote control has to be very close to the coordinator when you initiate the update and shortly beforehand Press a couple of buttons so that the remote control is not in sleep mode. Then the update works straight away! Hope this helps someone who has the same problem.

PlasticRhino, 2024-03-08:
Mine was about as far away as it can be in my house but maybe I have better coverage with my coordinator. I suspect that the significant action was bringing it out of standby.

MarkuZP, 2024-09-09:
same issue here with the ikea air quality sensor. Tried reboot before update. Bring it closer ti zigbee Hub. Press pairing button before. Nothing works!

MarkuZP, 2024-09-09:
info 2024-09-09 17:38:19z2m: Updating 'Ikea Luftqualitätsmesser' to latest firmware
info 2024-09-09 17:39:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ikea Luftqualitätsmesser', payload '{"humidity":59,"linkquality":51,"pm25":2,"temperature":20,"update":{"installed_version":65552,"latest_version":16777233,"state":"available"},"update_available":null,"voc_index":42}'
info 2024-09-09 17:39:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Ikea Luftqualitätsmesser"},"error":"Update of 'Ikea Luftqualitätsmesser' failed (Device didn't respond to OTA request)","status":"error","transaction":"m3zmu-1"}'
error 2024-09-09 17:39:03z2m: Update of 'Ikea Luftqualitätsmesser' failed (Device didn't respond to OTA request)
info 2024-09-09 17:39:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ikea Repeater DG Vorne Links', payload '{"linkquality":51,"update":{"installed_version":587753009,"latest_version":587753009,"state":"idle"},"update_available":null}'