Koenkk/zigbee2mqtt #15236: Moes ZS-EUB_1gang Switch_type could not be changed

Murada99, 2022-11-29:
### What happened?

Trying to get the swith type fialed and same problem with the write:

Zigbee2MQTT:error 2022-11-29 10:06:44: Publish 'get' 'switch_type' to 'dressing light' failed: 'Error: Read 0xa4c1382afc14d2b9/1 manuSpecificTuya_3(["switchType"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:debug 2022-11-29 10:06:44: Error: Read 0xa4c1382afc14d2b9/1 manuSpecificTuya_3(["switchType"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at Object.convertGet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:3909:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:273:21)
Zigbee2MQTT:info  2022-11-29 10:06:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'switch_type' to 'dressing light' failed: 'Error: Read 0xa4c1382afc14d2b9/1 manuSpecificTuya_3([\"switchType\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"dressing light"},"type":"zigbee_publish_error"}'

WRITING (SET)

Zigbee2MQTT:debug 2022-11-29 10:09:01: Received MQTT message on 'zigbee2mqtt/0xa4c1382afc14d2b9/set' with data '{"switch_type": "toggle"}'
Zigbee2MQTT:debug 2022-11-29 10:09:01: Publishing 'set' 'switch_type' to 'dressing light'
Zigbee2MQTT:error 2022-11-29 10:09:01: Publish 'set' 'switch_type' to 'dressing light' failed: 'Error: Write 0xa4c1382afc14d2b9/1 manuSpecificTuya_3({"switchType":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:debug 2022-11-29 10:09:01: Error: Write 0xa4c1382afc14d2b9/1 manuSpecificTuya_3({"switchType":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:3905:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
Zigbee2MQTT:info  2022-11-29 10:09:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'switch_type' to 'dressing light' failed: 'Error: Write 0xa4c1382afc14d2b9/1 manuSpecificTuya_3({\"switchType\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"dressing light"},"type":"zigbee_publish_error"}'

### What did you expect to happen?

Change switch type.

### How to reproduce it (minimal and precise)

Reading the value publishing a message to topic zigbee2mqtt/FRIENDLY_NAME/get with payload {"switch_type": ""}. 
Writing a value publishing a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"switch_type": NEW_VALUE}. The tested values are: toggle, state, momentary

### Zigbee2MQTT version

 1.28.2

### Adapter firmware version

20201127

### Adapter

CC2531

Koenkk, 2022-11-29:
Seems the version you have does not support this. (` (Status 'UNSUPPORTED_ATTRIBUTE')'`)

Murada99, 2022-11-30:
Hi Koenkk, it seems to be the right one, I bought it last week.
It leaves errors in HA log : Invalid option for select.second_light_switch_type: '' (valid options: ['toggle', 'state', 'momentary'])

[image]

closed 2023-01-07 by github-actions[bot]