Koenkk/zigbee2mqtt #15238: IKEA 5-button (E1524/E1810) remote falls stops responding, and have to be repaired.

phellarv, 2022-11-29:
### What happened?

After a "while" of no use, the 5-button remote from IKEA have to be re-paired to work. It has not gone missing from the list of devices in Z2M, but it will not work whichever button I try, and no matter how many times I press the buttons.
After a re-pairing everything is back to normal.

Only thing I get from the logs are 1 entry from last night when the device reported battery status:
Zigbee2MQTT:info  2022-11-28 23:56:00: MQTT publish: topic 'zigbee2mqtt/remote_2_kontor', payload '{"action":null,"battery":74,"linkquality":131,"update":{"state"
:"idle"},"update_available":null}'

And then nothing before I re-paired the device.
No other devices have this behavior, but I do not have any other 5-button remotes from IKEA, so I cannot say if this is a behavior of this specific device, or it is common for 5-button remotes from IKEA. 
Debug-logging is hard to get, since the behavior seems a bit erratic. But I can try to get logs by enabling debug, and restart Z2M the next time it happens.

### How to reproduce it (minimal and precise)

Do not use the device for a while. At least a couple of days.

### Zigbee2MQTT version

 1.28.2

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

therealeldaria, 2022-12-02:
I have several of these remotes, and they are notorious of dropping of the network when battery starts going down. and they are really sensitive to bad batteries. I had batteries from Swedish store Kjell, and they would last one or two weeks before the remote gave up. Since I switched to Duracell Long Life (Black and Brown) They will last much longer, months compared to weeks. However they still tend to drop of the network when battery drops below 30-50%. So I'm actually looking to replace mine with something else.

mozzbozz, 2023-01-05:
This doesn't seem to be a problem with the Ikea Gateway (at least the Dirigera gateway), though. So more likely not a hardware limitation.

closed 2023-02-13 by github-actions[bot]