Koenkk/zigbee2mqtt #15250: [New device support]: Eglo 900024 ceiling light with color temperature, brightness and RGB

Delphius7, 2022-11-29:
### Link

https://www.eglo.com/at/deckenleuchte-saliteras-z-900024.html

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0xa4c1388eb2ded68d","nwkAddr":29269,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"EGLO_ZM_RGB_TW","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"EGLO_ZM_RGB_TW","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1669744268969,"defaultSendRequestWhen":"immediate"}

### Comments

I am already using the 1.28.2 dev branch which solved the "deviceID is missing" issues for some of my lights [#15223 ](https://github.com/Koenkk/zigbee2mqtt/issues/15223).
Thanks for this!

However this RGB light which was recognized as Awox 33943/33944/33946 before is now unsupported.
I have tried to add the device and create a converter with the debug messages but I don't fully understand how this works.

Could you please help me to add this as new light or fix this again?
Thanks!

Debug Log with some "no converter available" messages:
```
Zigbee2MQTT:info  2022-11-29 19:46:12: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T19:46:12+01:00","linkquality":72,"power_on_behavior":null,"state":"OFF"}'
Zigbee2MQTT:debug 2022-11-29 19:46:16: Received Zigbee message from '0xa4c1388eb2ded68d', type 'raw', cluster '65360', data '{"data":[1,114,1,0,0],"type":"Buffer"}' from endpoint 3 with groupID 0
Zigbee2MQTT:debug 2022-11-29 19:46:16: No converter available for '900024' with cluster '65360' and type 'raw' and data '{"data":[1,114,1,0,0],"type":"Buffer"}'
Zigbee2MQTT:info  2022-11-29 19:46:16: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T19:46:16+01:00","linkquality":69,"power_on_behavior":null,"state":"OFF"}'
Zigbee2MQTT:debug 2022-11-29 19:47:09: Received Zigbee message from '0xa4c1388eb2ded68d', type 'read', cluster 'genTime', data '["time","timeStatus","timeZone","lastSetTime","validUntilTime"]' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-11-29 19:47:09: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T19:47:09+01:00","linkquality":76,"power_on_behavior":null,"state":"OFF"}'
```

Here the interview log:
```
Zigbee2MQTT:info  2022-11-29 17:10:14: Device '0xa4c1388eb2ded68d' joined
Zigbee2MQTT:info  2022-11-29 17:10:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1388eb2ded68d","ieee_address":"0xa4c1388eb2ded68d"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-11-29 17:10:14: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d/availability', payload 'online'
Zigbee2MQTT:info  2022-11-29 17:10:14: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:14+01:00","linkquality":87,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:14: Starting interview of '0xa4c1388eb2ded68d'
Zigbee2MQTT:info  2022-11-29 17:10:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1388eb2ded68d","ieee_address":"0xa4c1388eb2ded68d","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-11-29 17:10:14: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:14+01:00","linkquality":87,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1388eb2ded68d","ieee_address":"0xa4c1388eb2ded68d"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-11-29 17:10:15: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:15+01:00","linkquality":40,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:15: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:15+01:00","linkquality":40,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:15: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:15+01:00","linkquality":40,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:15: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:15+01:00","linkquality":36,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:15: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:15+01:00","linkquality":40,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:15: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:15+01:00","linkquality":36,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:17: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:17+01:00","linkquality":36,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:17: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:17+01:00","linkquality":40,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:17: MQTT publish: topic 'zigbee2mqtt/0xa4c1388eb2ded68d', payload '{"brightness":216,"color":{"h":37,"hue":37,"s":64,"saturation":64,"x":0.4087,"y":0.3924},"color_mode":"color_temp","color_temp":290,"last_seen":"2022-11-29T17:10:17+01:00","linkquality":36,"state":"OFF"}'
Zigbee2MQTT:info  2022-11-29 17:10:17: Successfully interviewed '0xa4c1388eb2ded68d', device has successfully been paired
Zigbee2MQTT:warn  2022-11-29 17:10:17: Device '0xa4c1388eb2ded68d' with Zigbee model 'EGLO_ZM_RGB_TW' and manufacturer name 'AwoX' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:info  2022-11-29 17:10:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c1388eb2ded68d","ieee_address":"0xa4c1388eb2ded68d","status":"successful","supported":false},"type":"device_interview"}'
[... cut ...]
### External converter

```shell
// Not sure if this is correct and for sure not complete
const definition = {
    zigbeeModel: ['EGLO_ZM_RGB_TW'],
    model: '900024',
    vendor: 'Eglo',
    description: 'LED light with RGB & brightness & color temperature',
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 370]}),
};
```

### Supported color modes

color temperature, color (RGB), brightness

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}

Koenkk, 2022-11-30:
Converter looks good, added it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-30 by Koenkk

Delphius7, 2022-11-30:
Thanks, works great!

One more question:
I have 4 other ceiling light models from this vendor which are all recognized wrong as model [33955](https://www.zigbee2mqtt.io/devices/33955.html#awox-33955) but they work just fine.
Is there any benefit or value of doing the same and adding them to your supported device list?

The only issue I can spot is the following in HA which didn't a exist with oder Z2M or HA releases:
```
2022-11-30 23:20:49.738 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.bathroom_light
```

Koenkk, 2022-12-01:
What is the actual model?

Delphius7, 2022-12-01:
Thanks for your effort! Let me know if there is anything else I can provide or prepare.

1. The one producing the HA errors is the following:

```
model: '99548',
vendor: 'EGLO',
description: 'SALITERAS-Z ceiling light',
```
[Product Link](https://www.eglo.com/at/deckenleuchte-herrora-z-99548.html)
Database.db: `{"id":4,"type":"Router","ieeeAddr":"0xa4c138302b07813c","nwkAddr":18666,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"EGLO_ZM_TW","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"EGLO_ZM_TW","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1669926470389,"defaultSendRequestWhen":"immediate"}`
 

 ---------------------------------------------------------------
Here also the other 3 models which are also recognized wrong as model [33955](https://www.zigbee2mqtt.io/devices/33955.html#awox-33955):

**2. 900316**

```
model: '900316',
vendor: 'EGLO',
description: 'PARANDAY-Z ceiling light',
```
[Product Link](https://www.eglo.com/at/deckenleuchte-paranday-z-900316.html)
Database.db: `{"id":9,"type":"Router","ieeeAddr":"0xa4c1386b79257abe","nwkAddr":15175,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"EGLO_ZM_TW","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"EGLO_ZM_TW","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1669926459026,"defaultSendRequestWhen":"immediate"}`
 
---------------------------------------------------------------

**3. 900317**

```
model: '900317',
vendor: 'EGLO',
description: 'PARANDAY-Z ceiling light',
```
[Product Link](https://www.eglo.com/at/deckenleuchte-paranday-z-900317.html)
Database.db: `{"id":18,"type":"Router","ieeeAddr":"0xa4c138a416e3ad39","nwkAddr":6361,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"EGLO_ZM_TW","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"EGLO_ZM_TW","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorMode":2,"colorTemperature":333}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1669926470384,"defaultSendRequestWhen":"immediate"}`

 ---------------------------------------------------------------

**4. 900053**

```
model: '900053',
vendor: 'EGLO',
description: 'SALOBRENA-Z ceiling light',
```
[Product Link](https://www.eglo.com/at/deckenleuchte-salobrena-z-900053.html)
Database.db: `{"id":17,"type":"Router","ieeeAddr":"0xa4c138fe9282d0c7","nwkAddr":34667,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"EGLO_ZM_TW","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"EGLO_ZM_TW","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysic [... cut ...] epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1669844184866,"defaultSendRequestWhen":"immediate"}`

Delphius7, 2022-12-04:
I have no groups with this light, but I have found the issue.
It seems like the light was sending invalid data. See the `"color_mode":"xy"` in the log:

`Zigbee2MQTT:info  2022-12-04 00:26:28: MQTT publish: topic 'zigbee2mqtt/0xa4c138302b07813c', payload '{"color_mode":"xy","last_seen":"2022-12-04T00:26:28+01:00","linkquality":36,"power_on_behavior":null,"state":null}'`

After changing the color temperature in the vendor app (via Bluetooth) the issue is gone and it reports a valid `"color_mode":"color_temp"` again. Maybe this was caused by me doing a factory reset over and over again.

Thanks again for all your help!!