Koenkk/zigbee2mqtt #15253: Ikea E1810 remote pressing power button switches xiaomi WS-USC04 smart switch

Johnny140, 2022-11-30:
### What happened?

Just a few days ago, setup my first zigbee devices using zigbee2mqtt (Zigbee Sonoff MiniL & Xiaomi 2 button switch). Yesterday I added (2) of the Ikea E1810 5 button remotes. I do not have any automation, blueprints, scripts running on the system. By pressing the power button on either of Ikea remotes, the first (upper) Xiaomi button of the dual smart switch toggles its state. I have not done any linking of any sort either. I am running zigbee2mqtt on Ubuntu 22.04 in a VM.

### What did you expect to happen?

I expected nothing to happen since I have not done any setup for the remotes to do anything yet.

### How to reproduce it (minimal and precise)

Have existing Xiaomi smart switch installed and setup. Pair an Ikea E1810 remote. When configured, just press the power button.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3 USB Dongle E

### Debug log

2022-11-29 19:42:24MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-01', payload '{"action":"","battery":74,"linkquality":94,"update":{"state":"idle"},"update_available":null}'
info 2022-11-29 19:42:24MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-01/action', payload 'toggle'
info 2022-11-29 19:42:24MQTT publish: topic 'zigbee2mqtt/Laundry-Exhaust-Light&Fan (shared with kitchen)', payload '{"linkquality":94,"state_bottom":"OFF","state_top":"OFF","update":{"state":"idle"},"update_available":null}'
info 2022-11-29 19:42:25MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-01', payload '{"action":"toggle","battery":74,"linkquality":98,"update":{"state":"idle"},"update_available":null}'
info 2022-11-29 19:42:25MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-01', payload '{"action":"","battery":74,"linkquality":98,"update":{"state":"idle"},"update_available":null}'
info 2022-11-29 19:42:25MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-01/action', payload 'toggle'
info 2022-11-29 19:42:25MQTT publish: topic 'zigbee2mqtt/Laundry-Exhaust-Light&Fan (shared with kitchen)', payload '{"linkquality":91,"state_bottom":"OFF","state_top":"ON","update":{"state":"idle"},"update_available":null}'
info 2022-11-29 19:42:26MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-02', payload '{"action":"toggle","battery":87,"linkquality":94,"update":{"state":"idle"},"update_available":null}'
info 2022-11-29 19:42:26MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-02', payload '{"action":"","battery":87,"linkquality":94,"update":{"state":"idle"},"update_available":null}'
info 2022-11-29 19:42:26MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-02/action', payload 'toggle'
info 2022-11-29 19:42:26MQTT publish: topic 'zigbee2mqtt/Laundry-Exhaust-Light&Fan (shared with kitchen)', payload '{"linkquality":94,"state_bottom":"OFF","state_top":"OFF","update":{"state":"idle"},"update_available":null}'
info 2022-11-29 19:42:26MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-02', payload '{"action":"toggle","battery":87,"linkquality":98,"update":{"state":"idle"},"update_available":null}'
info 2022-11-29 19:42:26MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-02', payload '{"action":"","battery":87,"linkquality":98,"update":{"state":"idle"},"update_available":null}'
info 2022-11-29 19:42:26MQTT publish: topic 'zigbee2mqtt/Ikea-Remote-02/action', payload 'toggle'
info 2022-11-29 19:42:26MQTT publish: topic 'zigbee2mqtt/Laundry-Exhaust-Light&Fan (shared with kitchen)', payload '{"linkquality":87,"state_bottom":"OFF","state_top":"ON","update":{"state":"idle"},"update_available":null}'

zegl, 2022-11-30:
This means that the zigbee devices are bound to each other, and are communicating peer-to-peer and bypassing the controller.

You can fix this by unbinding the devices from each other, see this guide: https://www.zigbee2mqtt.io/guide/usage/binding.html

Johnny140, 2022-11-30:
> This means that the zigbee devices are bound to each other, and are communicating peer-to-peer and bypassing the controller.
> 
> You can fix this by unbinding the devices from each other, see this guide: https://www.zigbee2mqtt.io/guide/usage/binding.html

Hi, thanks for your response. I am very new to the zigee2mqtt side of things.
Here is what I have done and the response n the logs. The remote and the switch are still linked, the remote toggles the power switch. Am I missing some steps or just blind. Have tried multiple times.

Thanks.

Using MQTT explorer, executed:
Topic -> zigbee2mqtt/bridge/request/device/unbind
{
  "from": "Ikea-Remote-01",
  "to": "Laundry-Exhaust-Light&Fan" , "skip_disable_reporting": true
}

Logs from Zigbee2MQTT show:
info 2022-11-30 13:26:15Successfully unbound cluster 'genScenes' from 'Ikea-Remote-01' to 'Laundry-Exhaust-Light&Fan'
info 2022-11-30 13:26:19Successfully unbound cluster 'genOnOff' from 'Ikea-Remote-01' to 'Laundry-Exhaust-Light&Fan'
info 2022-11-30 13:26:19MQTT publish: topic 'zigbee2mqtt/bridge/response/device/unbind', payload '{"data":{"clusters":["genScenes","genOnOff"],"failed":[],"from":"Ikea-Remote-01","to":"Laundry-Exhaust-Light&Fan"},"status":"ok"}'

Tontze, 2022-12-01:
Try resetting ikea switch by tapping button behind five times fast ?

Johnny140, 2022-12-02:
> Try resetting ikea switch by tapping button behind five times fast ?

Hi.

Not sure what has happened since, but the link between the power button on the Ikea remote and the Xiaomi smart switch is no longer there, yeah. I was busy yesterday being out and about and then added 2 of the Ikea Tradfri LED bulbs that are dimmable and can change colour temperature as well. 

Thanks everyone.

Andrew

closed 2022-12-02 by Johnny140