Koenkk/zigbee2mqtt #15256: [Feature request]: Support quick open setting for Danfoss Ally thermostats

rsporsche, 2022-11-30:
### Is your feature request related to a problem? Please describe

The Danfoss Ally zigbee cluster specification describes the 'algorithm_scale_factor' attribute as such:

Range 1-10 (lower 4 bit allocated to scale factor)
Scale factor of setpoint filter timeconstant
("aggressiveness" of control algorithm) 1=5min(Quick) ...
5=30min(Moderate) ... 10=80min(Slow).
**Bit4=Quick open feature disable. 1=disable. 0=enable**

The algorithm_scale_factor attribute in zigbe2mqtt only accepts values between 1 and 10 and there is no attribute that corresponds to the quick open feature disable bit.

### Describe the solution you'd like

Please add an attribute that allows the quick open feature disable bit of the algorithm_scale_factor attribute to be set for danfoss ally thermostats.

### Describe alternatives you've considered

I thought I could experiment with the behaviour of this bit using the Dev Console but I don't see any way to set a 'raw' attribute value, it only allows me to select named attribute values in the Dev console with, I assume, the same 1-10 range restriction.

### Additional context

The reason I want to set this is that I find that the 'quick open' behaviour is too aggressive with my radiators and it results in a massive overshoot of the target temperature.
I have solved this for my HA integration by overriding the HA temperature_command_topic to use occupied_heating_setpoint_scheduled which doesn't trigger the quick open, but changes on the thermostat's physical dial still trigger it. It's my hope that disabling this feature via the quick open feature disable bit of algorithm_scale_factor will affect the behaviour of setpoint changes using the physical dial as well.

closed 2023-01-07 by github-actions[bot]

Wickedy123, 2023-11-08:
>>I have solved this for my HA integration by overriding the HA temperature_command_topic to use

Sorry for the late follow-up, but can you tell me how you did this override? I have the opposite problem, where I would like the valve to act rapidly, therefore getting the the request sent out on occupied_heating_setpoint topic._

rsporsche, 2023-11-08:
> Sorry for the late follow-up, but can you tell me how you did this override? I have the opposite problem, where I would like the valve to act rapidly, therefore getting the the request sent out on [...]

I added this content under the 'homeassistant' section for each of my thermostats:

`devices:`
`  '0x0123456789abcdef':`
`    friendly_name: Bedroom Thermostat`
`    homeassistant:`
`      climate:`
`        current_temperature_template: '{{ value_json.external_measured_room_sensor|float/100 }}'`
`        temperature_command_topic: >-`
`          zigbee2mqtt/Bedroom`
`          Thermostat/set/occupied_heating_setpoint_scheduled`
`        temperature_state_template: '{{ value_json.occupied_heating_setpoint_scheduled }}'`

Note the name of the thermostat in the value of temperature_command_topic.

There may be a better way but this is working for me.

But the quick open behaviour is enabled by default so I'm not sure what you want to change?