Koenkk/zigbee2mqtt #15258: Error: spawn udevadm ENOENT #420

platini76, 2022-11-30:
### What happened?

### Description of the issue

I dont know if is my bad config or something else...

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://192.168.15.10:1883
  user: USER
  password: PASSWORD
serial:
  serial:
    port: /dev/ttyACM0
    adapter: deconz
```

### Addon version

1.28.2-1 

### Platform

Home Assistant 2022.11.4
Supervisor 2022.11.2
Operating System 9.3
Frontend 20221108.0 - latest

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

26720700

### Adapter

Conbee II

### Debug log

```shell
[03:00:29] INFO: Preparing to start...
[03:00:29] INFO: Socat not enabled
[03:00:31] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2022-11-26 03:00:35: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-26.03-00-35' filename: log.txt
Zigbee2MQTT:info  2022-11-26 03:00:35: Starting Zigbee2MQTT version 1.28.2 (commit #unknown)
Zigbee2MQTT:info  2022-11-26 03:00:35: Starting zigbee-herdsman (0.14.68)
Zigbee2MQTT:error 2022-11-26 03:00:36: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-11-26 03:00:36: Failed to start zigbee
Zigbee2MQTT:error 2022-11-26 03:00:36: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-26 03:00:36: Exiting...
Zigbee2MQTT:error 2022-11-26 03:00:36: Error: spawn udevadm ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:283:19)
    at onErrorNT (node:internal/child_process:478:16)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
```

ChrisHae, 2022-11-30:
could you try `port: null` to auto detect the path or check the serial/by-id path and use it in your config after port: ...
`ls -l /dev/serial/by-id`
the name should start with: `usb-dresden_elektronik_ingenierutechnik_GmbH_ConBee_II_DE..`

platini76, 2022-11-30:
port is the only thing correct..

Koenkk, 2022-11-30:
The serial part is invalid, it should be:

```yaml
serial:
  port: /dev/ttyACM0
  adapter: deconz
```

platini76, 2022-11-30:
> The serial part is invalid, it should be:
> 
> ```yaml
> serial:
>   port: /dev/ttyACM0
>   adapter: deconz
> ```

no. not working.

ChrisHae, 2022-12-01:
> > The serial part is invalid, it should be:
> > ```yaml
> > serial:
> >   port: /dev/ttyACM0
> >   adapter: deconz
> > ```
> 
> no. not working.

Really? in your first post you wrote your config was:
```
serial:
  serial:
    port: ...
```

closed 2023-01-08 by github-actions[bot]