Koenkk/zigbee2mqtt #15265: [Feature request]: add vendor for Human Presence Sensor

joohann, 2022-11-30:
### Is your feature request related to a problem? Please describe

Vendor is not listed in Tuya Human Presence Sensor _TZE204_ztc6ggyl please add 😊 

### Describe the solution you'd like

adding _TZE204_ztc6ggyl

### Describe alternatives you've considered

Local converter is working base in original code

### Additional context

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_ztc6ggyl'
        },
    ],
    model: 'TS0601',
    vendor: '_TZE204_ztc6ggyl',
    description: 'Human Presence Sensor',
    fromZigbee: [fz.tuya_smart_human_presense_sensor],
    toZigbee: [tz.tuya_smart_human_presense_sensor],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
            e.illuminance_lux(), e.presence(),
            exposes.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
            exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1)
                .withDescription('sensitivity of the radar'),
            exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
                .withDescription('Minimum range').withUnit('m'),
            exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
                .withDescription('Maximum range').withUnit('m'),
            exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
                .withDescription('Detection delay').withUnit('s'),
            exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1)
                .withDescription('Fading time').withUnit('s'),
            // exposes.text('cli', ea.STATE).withDescription('not recognize'),
            exposes.enum('self_test', ea.STATE, Object.values(tuya.tuyaHPSCheckingResult))
                .withDescription('Self_test, possible resuts: checking, check_success, check_failure, others, comm_fault, radar_fault.'),
    ],
};

module.exports = definition;

Koenkk, 2022-11-30:
This device is already supported in the latest dev branch (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-30 by Koenkk

B1n00, 2022-12-07:
Just perform a test and this code works while the one in dev is not for me

Koenkk, 2022-12-08:
What doesn't work?