Koenkk/zigbee2mqtt #15266: Namron Heater - Model 5401395 hysteresis bug

TomasDJo, 2022-11-30:
### What happened?

I seem to be unable to set hysteresis lower than 5 degrees. 
Zigbee2mqtt shows "Hysterersis setting, range is 5-50, unit is 0.1oC, Default: 5.", this indicate that the adjustment is in 0.1° increments, and the lowest allowed hysteresis value is 0.5°. 

The interface indicates that the lowest value of the hysteresis is 5° celcius, and it is increased in 1° increments. This seems to end up in the oven overshooting the setpoint by several degrees. The oven also starts heating way to early (example: temperature sensor reports 17.74° celcius, the oven starts heating as soon as I set the temperature setpoint to 15.5° celcius or above)

### What did you expect to happen?

With the lowest hysteresis setting of 0.5°. The oven turns on when it is 0.5° below the setpoint. It turns off when 0.5° above the setpoint.

### How to reproduce it (minimal and precise)

Set the temperature to 2.5 degrees below the room temperature with the lowest hysteresis setting. This turns on the oven.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20210120

### Adapter

Slaesh CC2652RB

TomasDJo, 2022-12-05:
It looks to have been fixed in 1.28.4 (the hysteresis setting output in z2m is now showing correctly, and adjustable between 0.5 and 2 degrees). Unfortunately the temperature setting is way off. This is not related to the hysteresis setting according to auduny at https://github.com/Koenkk/zigbee2mqtt/issues/14412#issuecomment-1336460879

owangen, 2022-12-05:
> It looks to have been fixed in 1.28.4 (the hysteresis setting output in z2m is now showing correctly, and adjustable between 0.5 and 2 degrees).

Great. Could you close this issue since hysteresis problem is fixed?
The other problem you mentions looks to be from the device it self and cannot be fixed from z2m.

closed 2022-12-05 by TomasDJo