Koenkk/zigbee2mqtt #15275: [Feature request]: Set custom min/max brightness range for dimmers

b4dpxl, 2022-12-01:
### Is your feature request related to a problem? Please describe

I have 2 dimmer switches which allow you to set a minimum brightness value, but not a maximum. On both dimmers for the current bulbs I'm using they reach maximum brightness at around 60%, and minimum at around 15%. This makes the usable range of adjustment only about 50%.

### Describe the solution you'd like

I would like to be able to specify a range in the configuration which maps to 0-100%. E.g.

```
devices:
  '0x8cf681fffb527e0c':
    friendly_name: Samotech dimmer 2
    min_brightness: 10
    max_brightness: 60
```

This way, if the dimmer is adjusted directly it's value would be translated to the full range to send to Home Assistant, and vice versa.

### Describe alternatives you've considered

none

### Additional context

If you can point me in the right direction, I can have a go at making a PR. The range mapping is easy enough, but I'm not sure where it would be needed.

b4dpxl, 2022-12-01:
The value wouldn't be consistent, it depends on the bulbs used. When used with halogens the same dimmers works "normally", e.g. 0-100%. But with dimmable LEDs the range is different. 

The real culprit is probably the dimmable LEDs, but I'd like to be able to accomodate this.

I had different dimmable LEDs before which worked better from a range perspective, although they sometimes wouldn't wake up except at full brightness 🤷

danieledwardgeorgehitchcock, 2022-12-01:
Ah I see! This is something that could probably be added to the device settings part of Z2M (not the exposes)

b4dpxl, 2022-12-01:
If it makes any difference though, these are the dimmers I'm using: 

- https://www.zigbee2mqtt.io/devices/ZG2835RAC.html
https://www.zigbee2mqtt.io/devices/SM323.html

Robi1976, 2022-12-02:
> If it makes any difference though, these are the dimmers I'm using:
> 
> * https://www.zigbee2mqtt.io/devices/ZG2835RAC.html
>   https://www.zigbee2mqtt.io/devices/SM323.html

I'm using five ZG2835RAC with halogens. 3 work fine but 2 trigger light flickering as soon as I go higher than 95% or approx. 238 in the 0-254 scale. The dev console of Z2M doesn't allow to set a max value (unknown attribute), only the min value and the start-up value. Is there no way set a max value through Z2M? In my case it's because of the halogens, it's unfortunately because of the dimmer. The Z-Wave version of the ZG2835 allows the setting of a max value through the Z-Wave integration.

Regardless of the dimmer brand, it would be great to have a function in Z2M to specify an own brightness range. If it is possible and I don't know, please someone explain to me.

b4dpxl, 2022-12-02:
> The dev console of Z2M doesn't allow to set a max value (unknown attribute), only the min value and the start-up value. 

I didn't know you could do this through the console. Presumably you can set those values though as the hardware supports them, but not max.

Robi1976, 2022-12-02:
> > The dev console of Z2M doesn't allow to set a max value (unknown attribute), only the min value and the start-up value.
> 
> I didn't know you could do this through the console. Presumably you can set those values though as the hardware supports them, but not max.

In the dev console choose endpoint "1", cluster "LevelCtrl" and attribute "onlevel". Reading the value gives the output 255 but you can write your own and it works. Regarding the min value I've mistaken, that's as you wrote directly through the min-level button on the dimmer. So there's no direct connection between what the dimmer physically can and Z2M allows (you can't define a start up value on the dimmer directly). Unfortunately I'm not affine with IT or programming but I would have expected that this should be possible if someone adapts the converter to this device. Or ist that not possible?

b4dpxl, 2022-12-02:
> Or is is not possible to choose an own brightness scale between the standard 0-254?

Not currently, AFAIK, hence my feature request. Although as a hack, if you're using Home Assistant, you can create a template light. It's a pain, but I got it working last night as a stop gap:

```
- platform: template
  lights:
    bedroom_dimmer:
      # min = 10% = 26 (25.5)
      # max = ~57% = 145 (145.35)
      unique_id: some_random_string
      level_template: "{{ [[0, ((state_attr('light.master_bedroom_dimmer', 'brightness') or 0)  - 26) * (255 / (145 - 26) )] | max, 255] | min }}"
      value_template: "{{ states('light.master_bedroom_dimmer') }}"
      turn_on:
        service: light.turn_on
        target:
          entity_id: light.master_bedroom_dimmer
      turn_off:
        service: light.turn_off
        target:
          entity_id: light.master_bedroom_dimmer
      set_level:
        service: light.turn_on
        target:
          entity_id: light.master_bedroom_dimmer
        data:
          brightness: "{{ (brightness / (255 / (145 - 26))) + 26 }}"
```

Robi1976, 2022-12-02:
> Not currently, AFAIK, hence my feature request. Although as a hack, if you're using Home Assistant, you can create a template light. It's a pain, but I got it working last night as a stop gap:
>
> ``` [...]

Many thanks b4dpxl! I'll try this weekend :-)

I saw the following conversation on gitbub [https://github.com/Koenkk/zigbee2mqtt/issues/12793](url) and therefore thought that it should be possible in general, since a max value was also programmed via the converter for this link.

closed 2023-01-09 by github-actions[bot]

timothylui, 2024-01-05:
I also have this problem, I have the same dimmers and despite using the highest Watt GU10 bulbs available to me, maximum brightness is at about 40%.

BenJamesAndo, 2024-01-29:
If your device itself supports setting a minimum brightness then you might be able to get it to work by adding in `minBrightness: true` to the devices config or external converter e.g. `extend: [tuya.modernExtend.tuyaLight({minBrightness: true})]` I tried adding it to the ZB-DL01  0-10V dimmer but it didn't work. So it would be nice if Z2M could somehow handle it using software, instead of needing the hardware itself to support it.
[image]

coverthembrow, 2024-01-29:
My devices in question don't support this. TBH, max brightness would be more beneficial in my case, as my bulb reaches 100% brightness when the dimmer still says 60%. 

I've also got one bulb which whines annoyingly at 100%, but it fine at 95%, so it would be useful to be able to max that out.

kravemir, 2024-03-03:
Solved by [fredck/lightener](https://github.com/fredck/lightener).

It is quite a nice wrapper, that allows mapping custom ranges of HA brightness percentage to custom ranges of brightness percentage of dimmer(s). I.e. _1-100% (HA control)_ to _50-90% (Zigbee dimmer)_, and _0%_ means _0%_.