Koenkk/zigbee2mqtt #15279: [Feature request]:  Add OTA support to Eco-Dim.07/Eco-Dim.10

Kilill, 2022-12-01:
### Is your feature request related to a problem? Please describe

Have a acquired a new firmware file (`ZB21S3_HZC_Dimmer1_ECO_1.01_20221021.ota`) from ecodim.nl for Eco-Dim.07/Eco-Dim.10 ,
but OTA is not enabled in the device file it seems,  so it can't be loaded it via zigbee2mqtt.

### Describe the solution you'd like

Be able to load a firmware vi OTA

### Describe alternatives you've considered

create a local device file with ota enabled but seems not to work, probably due me not correctly understanding the proper procedure...

### Additional context

Dimmer does not work correctly, ecodim.nl has made new firmware available, and btw, been informed about the existence of the ota database so this and future updates should appear there.

Koenkk, 2022-12-01:
Can you upload the OTA file here? Then I can add it.

Kilill, 2022-12-01:
The problem is that OTA is not enabled in the device descriptor for ecodim at _all,_ ie. zigbee2mqtt does not recognize that ecodim is "OTA capable".
Also, I asked permission from ecodim.nl if it was ok to upload, and the response was that they wanted to do that themselves. I do have a mail adress to the fellow but i don't feel comfortable publish that this public...

Koenkk, 2022-12-02:
I can enable the OTA, but I need to have an OTA file first.

Kilill, 2022-12-02:
Ok, i'l contact Ecodim about that, i know that they should be i the process of adding it to the zigbee-OTA site

Kilill, 2022-12-05:
There should now be a pull request from me on  the ota file on the zigbee-oat repo

closed 2022-12-06 by Koenkk

Kilill, 2022-12-07:
Thanks. Installed the dev branch, works fine, and the z2mqtt reports the ecodim as beeing ota ready, but trying to check for an update reports: 
```
Info 2022-12-07 11:10:42MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Dimmer Up Livingroom Ceiling"},"error":"Failed to check if update available for 'Dimmer Up Livingroom Ceiling' (No image available for imageType '44')","status":"error","transaction":"esjdk-2"}'
Error 2022-12-07 11:10:42Failed to check if update available for 'Dimmer Up Livingroom Ceiling' (No image available for imageType '44')
```

Veldkornet, 2023-01-04:
My devices are already on 1.01 anyway, but when I try to check for updates, I get the following error:

```
No image available for imageType '60'
```

Same problem?

Kilill, 2023-01-04:
I think you might have a different device than what i have ?

Veldkornet, 2023-01-04:
> I think you might have a different device than what i have ?

I have this [ECO-DIM.07 Zigbee Pro](https://www.ecobright.nl/zigbee-led-dimmer-draai-0-250w-ecodim-07-zigbee-pro.html)

Kilill, 2023-01-04:
Yea , and i have this one: [EcoDim](https://www.eldirekte.no/ecodim-smart-vridimmer-zigbee-pro/cat-p/c/p1500014234)
which is not exactly the same not sure if you have an earlier or later version.

But that aside, you probably get an error since there are no firmware files for ecodim as of yet. except the one i'm trying to unsuccessfully upload

Kilill, 2023-01-04:
@Veldkornet just checking, you said that your software version is 1.0 ? which is interesting since mine is 3.09....

Kilill, 2023-01-04:
Right, so mine is an older version I have build date: 20210228. That will be highly interesting since I have 3.09 and the version needs to be higher to be accepted...

Veldkornet, 2023-01-04:
Seems like you have the "Standard" version, and I have the "Pro" version, which I suppose was released later. 

Not sure what the difference is exactly though.

Kilill, 2023-01-04:
No it is also a pro version:
[image]

Veldkornet, 2023-01-04:
Aha! Strange! I see the one I have, you can switch between leading edge and trailing edge. 

Not sure if any of this actually matters for the firmware though of course 😀

Kilill, 2023-01-10:
I have a suspicion that a all the ones with the the _3.09_  firmware are the previous version of the dimmer, and that _1.01_ version is for the newer version. And seeing as the hardware looks slightly different, one can only hope that it is similar enough so that the _1.01_ can be back ported... but I'm not holding my breath.

My _3.09_ version has the problem that it will not go to full intensity when turning the knob all up (or setting it to 100%). But when trying to set the _max dim_ value it goes all the way up. It ofc drops the moment the dimer value changes.

untitledlt, 2023-03-21:
@Kilill i have exactly the same issue with max brightness. I have 4 dimmers and 3 of them has this issue. Also one of them sometimes makes 2 bulbs in parallel slightly blink. Very annoyed.
All of them are on 3.09 and z2m says there's no available firmware update.

Kilill, 2023-03-25:
I would take them back to the supplier and get them exchanged  for the new ones. I'm very pessimistic about EcoDim ever coming out with a corrected software for the old version of the dimmers.

Kilill, 2023-04-27:
I got the successor to it, and it works quite well, the reason i chose the ecodim in the first place was that it came with the faceplate and knob , while most other i could find did not...

kenniani, 2023-09-07:
> I agree. I do not think Ecodim will ever update this dimmer, at least not the one with firmware 3.09, date 20200701. I have 10 of these dimmers (all the same version) and at least 3 of them are not reporting the state to z2m. I have no idea why there is a difference in reporting the state because all 10 of them have the same firmware. I hoped a firmware update would fix this problem, but I do not see that happening. Also tried to email Ecodim, but no response.
> 
> In search of a better push/turn dimmer I found this one: https://www.ionindustries.com/winkel/led-zigbee-dimmer-0-3-200-watt/, tuya based. It is recognized in z2m as TS0601, _TZE200_4mh6tyyo (https://www.zigbee2mqtt.io/devices/TS0601_dimmer_1.html#tuya-ts0601_dimmer_1) It seem to work fine, the state is updated correctly. I will monitor its function the upcoming days.

I have the same issue, the turn off action is not always sent to mine gateway, tried Conbee II with Phoscon and Sonof Stick with Zigbee2MQTT, both same issue, I have the basic dimmer with firmware version 3.09.

Searching for a new dimmer brings me to the ionindustries one, only the Peha Standard plate I have isn't supported.