Koenkk/zigbee2mqtt #15282: Update from 1.28.2-1 to 1.28.3-1 breaks Silvercrest (LIDL) switch

pixal-it, 2022-12-01:
### What happened?

Updated the Zigbee2MQTT Home Assistant addon from 1.28.2-1 to 1.28.3-1 and after the add-on restarted one of my devices was replaced by a device that I don't even own.

The below device was fully working in 1.28.2-1. (device name in the logs is 0xec1bbdfffeb73ad0)
Silvercrest 3 gang switch, with 4 USB (EU, FR, CZ, BS) (https://www.zigbee2mqtt.io/devices/HG06338.html#lidl-hg06338)

But after the update to 1.28.3-1, it was replaced by this device with limited functionality.

Tuya smart plug (https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html)

### What did you expect to happen?

I expected all my devices to remain intact.

### How to reproduce it (minimal and precise)

Update from 1.28.2-1 to 1.28.3-1 with a Silvercrest (LIDL) switch in the network.

### Zigbee2MQTT version

1.28.3-1

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10134992/log.txt)

Koenkk, 2022-12-01:
Fixed in 1.28.4!

closed 2022-12-01 by Koenkk

Safs, 2022-12-06:
Same problem on :
Device type: Router
Zigbee Model: TS011F
Zigbee Manufacturer: _TZ3000_ew3ldmgx
Description: Smart plug (with power monitoring)
Support status: Supported

Manufacturer: [TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model: [TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1)

Please fix problem for this model also!

nonamelee1003, 2022-12-18:
可以一同修復tuya ts0601 _TZE200_adquvgsi嗎？
顯示無法支持
謝謝

janptr, 2023-02-24:
My Lidl HG06338 is still identifying as TuYa TS011F_plug_3. I'm running Zigbee2MQTT Version
 1.30.1
 / zStack3x0

 / 20210708. What can I do to fix it?

Koenkk, 2023-02-25:
@janptr please provide the data/database.db entry of your device.

janptr, 2023-02-27:
> @janptr please provide the data/database.db entry of your device.

This is the entry:

{"id":10,"type":"Router","ieeeAddr":"0xec1bbdfffe7524f5","nwkAddr":4402,"manufId":4098,"manufName":"_TZ3000_oznonj5q","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}},"genBasic":{"attributes":{"65503":"5ۋ+\u0012","65506":30,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_oznonj5q","zclVersion":3,"appVersion":69,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[3,4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":266,"inClusterList":[3,4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2123427102},"lastSeen":1677272375768,"defaultSendRequestWhen":"immediate"}

janptr, 2023-02-27:
> `_TZ3000_oznonj5q` will be detected as `HG06338` with the next release on 1 February.

Sounds great, thank you very much! Do you mean 1st march?

Koenkk, 2023-02-27:
Woops, yes!

janptr, 2023-03-01:
> Woops, yes!

Your new release fixed it. It works like a charm. Thank you very much for the fast and kind support! 😊

borristhecat, 2023-04-30:
hi do we have the OTA file for this device? a bodged version messed mine up but its still communicating so i'm hoping the update will fix it.