Koenkk/zigbee2mqtt #15285: After 1.28.3 update Z2M does not start anymore

JackTalisker, 2022-12-01:
### What happened?

After 1.28.3 update Z2M does not start anymore. Going back to 1.28.2 it starts again.

### What did you expect to happen?

Z2M starting

### How to reproduce it (minimal and precise)

docker image update to 1.28.3

### Zigbee2MQTT version

1.28.3

### Adapter firmware version

6.7.9.0 build 405

### Adapter

EZSP v8 (Sonoff ZBBridge Tasmota flashed)

### Debug log

Docker log:

```
pi@raspberrypi:~/home-assistant $ docker container logs zigbee2mqtt -f
Using '/app/data' as data directory

/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
Using '/app/data' as data directory

/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
Using '/app/data' as data directory

/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
Using '/app/data' as data directory

/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

```

Koenkk, 2022-12-01:
You are using external converter which is not valid for Zigbee2MQTT 1.28.3, update your external converter accordingly.

closed 2022-12-01 by Koenkk

Koenkk, 2022-12-02:
`tz.legrand_settingAlwaysEnableLed` was renamed to `tz.legrand_settingEnableLedInDark`. 

If you rely on these changes it is recommended to make a PR to integrate it. External converters are not meant to be used across releases (only for temporarily testing)