Koenkk/zigbee2mqtt #15292: ClickSmart+ now recognised incorrectly as TS011F_plug_3

uvjim, 2022-12-01:
### What happened?

After upgrading the Home Assistant Add-on to 1.28.3-1, the ClickSmart+ CMA30036 devices that I have now identify as TS011F_plug_3 so I can no longer control both sockets.  It used to be TS011F_2_gang_wall.  The ClickSmart+ CMA30035 devices now also identify as TS011F_plug_3 when it used to be TS011F_wall_outlet.

### What did you expect to happen?

No changes to how the devices were identified or for them to be mapped to equivalent devices.

### Zigbee2MQTT version

1.28.3

### Adapter firmware version

0x26780700

### Adapter

RaspBee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10135821/log.txt)

Koenkk, 2022-12-01:
Should already be fixed in 1.28.4

uvjim, 2022-12-02:
Confirmed as fixed in 1.28.4

closed 2022-12-02 by uvjim