Koenkk/zigbee2mqtt #15299: Devices are disappearing from Device list.

erzhan89, 2022-12-01:
### What happened?

Devices  4 pc. (RTV https://www.zigbee2mqtt.io/devices/GS361A-H04.html )  disappearing from Device list. after 15-20 minutes. Before they worked without issues. 

### What did you expect to happen?

To have stable connection. 

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

cloudharps, 2022-12-01:
Same problem. Devices dissapearing

terierul, 2022-12-01:
I have the same issue after updated to 1.28.4

Koenkk, 2022-12-01:
Are you all using the EZSP adapter?

erzhan89, 2022-12-01:
> I have the same issue after updated to 1.28.4 downgraded to 1.28.2 and everything back to normal

I did same. so far works fine.

cloudharps, 2022-12-01:
> Are you all using the EZSP adapter?

Yes.

terierul, 2022-12-01:
conbee 2

rhvs, 2022-12-01:
> Are you all using the EZSP adapter?

Slaesh stick

brunolinsalves, 2022-12-01:
Same thing here. I can't readd the device that left network. What should I do ?
I already updated to version 1.28.4-1 to test and the problem was not solved...

My device was ZM-L03E-Z(TS0003)

kilski, 2022-12-02:
@erzhan89 still works with 1.28.2?

erzhan89, 2022-12-02:
> @erzhan89 still works with 1.28.2?

Yes. Since I deployed v1.28.2 there is no issue.

brunolinsalves, 2022-12-07:
I wasn't been able to simulate the error to post the logs here. But, I tryed again, in version 1.28.4-1 and was successful. The devices was added as expected and didn't left the network anymore. 

Thanks

terierul, 2022-12-07:
> Can someone provide the debug log when a device disappears?
> 
> See https://www.zigbee2mq
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10173703/log.txt)
tt.io/guide/usage/debug.html on how to enable debug logging.

> Can someone provide the debug log when a device disappears?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Hello @Koenkk . Here you have the file
 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10173712/log.txt)

Koenkk, 2022-12-07:
It looks to me that Zigbee2MQTT is not running, can you check the log page of the addon?

terierul, 2022-12-07:
I think this is the issue (check the LE2 comment file- log after restart)
The last lines in the log. Something get wrong and forced Zigbee2MQTT to stop

debug 2022-12-07 08:52:16: lumi.vibration.aq1: unknown key 6 with value 0,5
debug 2022-12-07 08:52:16: lumi.vibration.aq1: unknown key 153 with value 107
debug 2022-12-07 08:52:16: lumi.vibration.aq1: Processed data into payload {"voltage":3005,"battery":100,"device_temperature":21,"power_outage_count":8}
debug 2022-12-07 08:52:16: lumi.vibration.aq1: Processed data into payload {"voltage":3005,"battery":100,"device_temperature":21,"power_outage_count":8}
info  2022-12-07 08:52:16: MQTT publish: topic 'zigbee2mqtt/Sensor vibracion 1', payload '{"action":null,"angle":9,"angle_x":0,"angle_x_absolute":90,"angle_y":-1,"angle_y_absolute":91,"angle_z":-89,"battery":100,"device_temperature":21,"linkquality":255,"power_outage_count":8,"sensitivity":"high","strength":2,"vibration":false,"voltage":3005}'
debug 2022-12-07 08:52:20: Saving state to file /config/zigbee2mqtt/state.json
info  2022-12-07 08:52:20: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2022-12-07 08:52:20: Disconnecting from MQTT server
info  2022-12-07 08:52:20: Stopping zigbee-herdsman...
info  2022-12-07 08:52:22: Stopped zigbee-herdsman
info  2022-12-07 08:52:22: Stopped Zigbee2MQTT

Koenkk, 2022-12-07:
Interesting, can you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

terierul, 2022-12-11:
Hello @Koenkk 
Sorry have been out of town for some days
please give me some more instructions how to extract STDOUT. Is the first time
Thanks

jerrm, 2022-12-11:
https://www.zigbee2mqtt.io/guide/usage/debug.html

But it is a little easier now for the HA addon:

[image]

terierul, 2022-12-11:
well I do not have this button in add-on. could you please activate the button and show me the yaml code please?

[image]

terierul, 2022-12-11:
@Koenkk after I have activated herdsman debug and install again 1.28.4 2 logs have been generated

Log 1
[log (1) 2022-12-12.00-02-06.txt](https://github.com/Koenkk/zigbee2mqtt/files/10203521/log.1.2022-12-12.00-02-06.txt)

Log2
[log (2) 2022-12-12.00-02-35.txt](https://github.com/Koenkk/zigbee2mqtt/files/10203522/log.2.2022-12-12.00-02-35.txt)

Look again at log 1

debug 2022-12-12 00:02:23: Saving state to file /config/zigbee2mqtt/state.json
info  2022-12-12 00:02:23: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2022-12-12 00:02:23: Disconnecting from MQTT server
info  2022-12-12 00:02:23: Stopping zigbee-herdsman...
info  2022-12-12 00:02:24: Stopped zigbee-herdsman
info  2022-12-12 00:02:24: Stopped Zigbee2MQTT

Koenkk, 2022-12-12:
Found the issue, should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

erzhan89, 2022-12-21:
@Koenkk ,
 after 2 days... lost all devices again...

terierul, 2022-12-21:
@erzhan89 I think you have another issue. For you the list is working and suddenly disappear? Or this is happening (like in my case) only after a reboot/restart?
anyway the fix is not in production. Probably will be pushed with the next version. Right now is in dev branch.

erzhan89, 2022-12-21:
I had dev branch... 
now reverted to 1.28.2 back. 
in 1.28.2 works stable...

terierul, 2022-12-21:
well the true is that I did not test the fix because my zigbee network is installed in a location where I am under a CG-NAT network. Is my secondary home and cannot connect from internet.
But still, after reboot, after 10-15 min your network does not come back alive?

erzhan89, 2022-12-21:
I tried twice to join all devices back... but after 2 days... they are not in list again..

Koenkk, 2022-12-21:
@erzhan89 what coordinator are you using?

kilski, 2022-12-22:
I gave up. I've bought ZBDongle-P - everything works well on that.

vignez2991, 2022-12-23:
> I tried twice to join all devices back... but after 2 days... they are not in list again..

Same issue for me also on power restart, But not all the time.

I am using CC2652R using CC2652R_coordinator_20220219 firmware with Z2M version 1.28.2

erzhan89, 2022-12-23:
@Koenkk, 
ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

Koenkk, 2022-12-23:
@erzhan89 the dongle-e is currently experimental @kirovilya do you know if this issue has been fixed already in the latest dev?

@vignez2991 can you explain a bit more what happens exactly? CC2652R should be stable.

myuc, 2022-12-24:
Jumping in here as I seem to be having the same issue. Devices randomly dropping off the network after a while/after stopping and starting zigbee2mqtt. Sometimes the device will rejoin automatically, other times it does not. Do let me know if you prefer I  submit a new issue instead.

**Zigbee2MQTT version:**  1.28.4-dev (commit #8fc83ca) 
I noticed that this problem started with v1.28.1. I do not get this issue when I downgrade to 1.28.0

**Coordinator type/revision:** EZSP v8 6.10.3.0 build 297 - ZBDongle-E

Below is a snippet of my herdsman log. I've trimmed the log using grep. The first 5 digit of each line refers to the line number in the full log file. The 6th character (either a `:` or a `-` is a by product of `grep -n`). The original log starts from the 7th character. Do let me know if you require more info from my full logs. 

```
00787:Zigbee2MQTT:info  2022-12-25 03:00:29: M Light (0x00124b00257c778c): ZBMINI-L - SONOFF Zigbee smart switch (no neutral) (EndDevice)
00793-Zigbee2MQTT:info  2022-12-25 03:00:29: Zigbee: disabling joining new devices.
00844-Zigbee2MQTT:debug 2022-12-25 03:00:30: Received MQTT message on 'zigbee2mqtt/M Room TV Con/availability' with data '{"state":"online"}'
00894:Zigbee2MQTT:debug 2022-12-25 03:00:30: Active device 'M Light' was last seen '20.45' minutes ago.
00895:Zigbee2MQTT:info  2022-12-25 03:00:30: MQTT publish: topic 'zigbee2mqtt/M Light/availability', payload '{"state":"online"}'
01007-2022-12-24T19:00:30.648Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
01008-2022-12-24T19:00:30.648Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
01009-2022-12-24T19:00:30.649Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3680017e00
01010-2022-12-24T19:00:30.649Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x7e: {"_id_":126,"_cls_":"setExtendedTimeout","_isRequest_":false}
01011-2022-12-24T19:00:30.649Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
01012-2022-12-24T19:00:30.649Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setExtendedTimeout","_id_":126,"_isRequest_":true,"remoteEui64":{"_value":{"type":"Buffer","data":[0,18,75,0,36,191,242,240]}},"extendedTimeout":true}
01013-2022-12-24T19:00:30.650Z zigbee-herdsman:adapter:ezsp:uart --> DATA (7,1,0): 3700017e00f0f2bf24004b120001
01014-2022-12-24T19:00:30.650Z zigbee-herdsman:adapter:ezsp:uart --> [717521a92a2ae540e6b04a6eb8559363d67e]
01015-2022-12-24T19:00:30.650Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (0)
01016-2022-12-24T19:00:30.650Z zigbee-herdsman:adapter:ezsp:ezsp ==> sendUnicast: {"type":0,"indexOrDestination":4666,"apsFrame":{"clusterId":0,"profileId":260,"sequence":7,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":320},"messageTag":8,"message":{"type":"Buffer","data":[16,3,0,0,0]}}
01027:Zigbee2MQTT:info  2022-12-25 03:00:30: MQTT publish: topic 'homeassistant/switch/0x00124b00257c778c/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/M Light/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/M Light/set","device":{"identifiers":["zigbee2mqtt_0x00124b00257c778c"],"manufacturer":"SONOFF","model":"Zigbee smart switch (no neutral) (ZBMINI-L)","name":"M Light","sw_version":"1.1.1"},"name":"M Light","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/M Light","unique_id":"0x00124b00257c778c_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
01028:Zigbee2MQTT:info  2022-12-25 03:00:30: MQTT publish: topic 'homeassistant/sensor/0x00124b00257c778c/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/M Light/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00124b00257c778c"],"manufacturer":"SONOFF","model":"Zigbee smart switch (no neutral) (ZBMINI-L)","name":"M Light","sw_version":"1.1.1"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","name":"M Light power on behavior","state_topic":"zigbee2mqtt/M Light","unique_id":"0x00124b00257c778c_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
01029:Zigbee2MQTT:info  2022-12-25 03:00:30: MQTT publish: topic 'homeassistant/select/0x00124b00257c778c/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/M Light/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/M Light/set/power_on_behavior","device":{"identifiers":["zigbee2mqtt_0x00124b00257c778c"],"manufacturer":"SONOFF","model":"Zigbee smart switch (no neutral) (ZBMINI-L)","name":"M Light","sw_version":"1.1.1"},"entity_category":"config","icon":"mdi:power-settings","name":"M Light power on behavior","options":["off","previous","on"],"state_topic":"zigbee2mqtt/M Light","unique_id":"0x00124b00257c778c_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
01030:Zigbee2MQTT:info  2022-12-25 03:00:30: MQTT publish: topic 'homeassistant/sensor/0x00124b00257c778c/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/M Light/availability","value_template":"{{ value_json.state }}"}],"availab [... cut ...]
01596-2022-12-24T19:00:41.143Z zigbee-herdsman:adapter:ezsp:debg Device left network request received: 22028 00124b00257c778c
01597:2022-12-24T19:00:41.144Z zigbee-herdsman:controller:log Device leave '0x00124b00257c778c'
01598:2022-12-24T19:00:41.144Z zigbee-herdsman:controller:log Removing device from database '0x00124b00257c778c'
01599:Zigbee2MQTT:warn  2022-12-25 03:00:41: Device 'M Light' left the network
01600:Zigbee2MQTT:info  2022-12-25 03:00:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"M Light","ieee_address":"0x00124b00257c778c"},"type":"device_leave"}'

```

myuc, 2022-12-25:
Including logs for another device which disappeared ~12 hours after starting zigbee2mqtt. First 6 lines refers to the line no in the original log file.

```
000832:Zigbee2MQTT:info  2022-12-25 04:27:17: Kitchen Light (0x2c1165fffe4290a6): TS0013 - TuYa Smart light switch - 3 gang without neutral wire (EndDevice)
```

herdsman log when device disappear from zigbee2mqtt (including 20 lines before/after the first/last instance of `Kitchen Light` or `0x2c1165fffe4290a6`.
```
166484-2022-12-25T08:00:49.073Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
166485-2022-12-25T08:00:49.073Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
166486-2022-12-25T08:00:49.074Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
166487-2022-12-25T08:00:49.074Z zigbee-herdsman:adapter:ezsp:uart > DATA (0,5,0): e800010500
166488-2022-12-25T08:00:49.074Z zigbee-herdsman:adapter:ezsp:uart > [05aa21a9512afb8a7e]
166489-2022-12-25T08:00:49.074Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
166490-2022-12-25T08:00:49.079Z zigbee-herdsman:adapter:ezsp:uart < [51aaa1a9512a50877e]
166491-2022-12-25T08:00:49.079Z zigbee-herdsman:adapter:ezsp:uart < DATA (5,1,0): 51aaa1a9512a50877e
166492-2022-12-25T08:00:49.079Z zigbee-herdsman:adapter:ezsp:uart > ACK  (6)
166493-2022-12-25T08:00:49.079Z zigbee-herdsman:adapter:ezsp:uart > [8610be7e]
166494-2022-12-25T08:00:49.079Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: e880010500
166495-2022-12-25T08:00:49.079Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
166496-2022-12-25T08:00:49.079Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (1) success
166497-2022-12-25T08:00:55.057Z zigbee-herdsman:adapter:ezsp:uart < [61aab5a9702aef8cff0408db553083659c4d6f97579e7e]
166498-2022-12-25T08:00:55.057Z zigbee-herdsman:adapter:ezsp:uart < DATA (6,1,0): 61aab5a9702aef8cff0408db553083659c4d6f97579e7e
166499-2022-12-25T08:00:55.057Z zigbee-herdsman:adapter:ezsp:uart > ACK  (7)
166500-2022-12-25T08:00:55.057Z zigbee-herdsman:adapter:ezsp:uart > [87009f7e]
166501-2022-12-25T08:00:55.058Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: e894012400fa3ea69042feff65112c0003483c
166502-2022-12-25T08:00:55.058Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_id_":36,"_cls_":"trustCenterJoinHandler","_isRequest_":false,"newNodeId":16122,"newNodeEui64":{"type":"Buffer","data":[44,17,101,255,254,66,144,166]},"status":0,"policyDecision":3,"parentOfNewNodeId":15432}
166503-2022-12-25T08:00:55.058Z zigbee-herdsman:adapter:ezsp:debg Device join request received: 16122 2c1165fffe4290a6
166504-2022-12-25T08:00:55.058Z zigbee-herdsman:controller:log Device '0x2c1165fffe4290a6' joined
166505-2022-12-25T08:00:55.058Z zigbee-herdsman:controller:log Device '0x2c1165fffe4290a6' accepted by handler
166506:Zigbee2MQTT:info  2022-12-25 16:00:55: MQTT publish: topic 'zigbee2mqtt/Kitchen Light', payload '{"backlight_mode":null,"last_seen":"2022-12-25T16:00:55+08:00","linkquality":80,"power_on_behavior":null,"state_center":null,"state_left":null,"state_right":null}'
166507:2022-12-25T08:00:55.060Z zigbee-herdsman:controller:log Not interviewing '0x2c1165fffe4290a6', completed 'true', in progress 'false'
166508:Zigbee2MQTT:info  2022-12-25 16:00:55: MQTT publish: topic 'zigbee2mqtt/Kitchen Light/mode', payload '{"backlight":null}'
166509-2022-12-25T08:00:55.062Z zigbee-herdsman:adapter:ezsp:uart < [71aab1a9772a15b2a3aaecb5e8ab6d2c8d622382617e]
166510-2022-12-25T08:00:55.062Z zigbee-herdsman:adapter:ezsp:uart < DATA (7,1,0): 71aab1a9772a15b2a3aaecb5e8ab6d2c8d622382617e
166511-2022-12-25T08:00:55.062Z zigbee-herdsman:adapter:ezsp:uart > ACK  (0)
166512-2022-12-25T08:00:55.062Z zigbee-herdsman:adapter:ezsp:uart > [8070787e]
166513-2022-12-25T08:00:55.062Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: e8900123000000fa3ea69042feff65112c04
166514-2022-12-25T08:00:55.063Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x23: {"_id_":35,"_cls_":"childJoinHandler","_isRequest_":false,"index":0,"joining":0,"childId":16122,"childEui64":{"type":"Buffer","data":[44,17,101,255,254,66,144,166]},"childType":4}
166515-2022-12-25T08:00:55.063Z zigbee-herdsman:adapter:ezsp:debg Device left network request received: 16122 2c1165fffe4290a6
166516:2022-12-25T08:00:55.063Z zigbee-herdsman:controller:log Device leave '0x2c1165fffe4290a6'
166517:2022-12-25T08:00:55.063Z zigbee-herdsman:controller:log Removing device from database '0x2c1165fffe4290a6'
166518:Zigbee2MQTT:warn  2022-12-25 16:00:55: Device 'Kitchen Light' left the network
166519:Zigbee2MQTT:info  2022-12-25 16:00:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Kitchen Light","ieee_address":"0x2c1165fffe4290a6"},"type":"device_leave"}'
166520-2022-12-25T08:00:58.326Z zigbee-herdsman:adapter:ezsp:uart < [01aab1a97d312a15b6589e4a24ab5593499cec4f1d17f09874f8c60489fb7d5e3bc28d7e]
166521-2022-12-25T08:00:58.326Z zigbee-herdsman:adapter:ezsp:uart < DATA (0,1,0): 01aab1a9112a15b6589e4a24ab5593499cec4f1d17f09874f8c60489fb7e3bc28d7e
166522-2022-12-25T08:00:58.326Z zigbee-herdsman:adapter:ezsp:uart > ACK  (1)
166523-2022-12-25T08:00:58.326Z zigbee-herdsman:adapter:ezsp:uart > [8160597e]
166524-2022-12-25T08:00:58.326Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: e8900145000004010a00010100010000a268b6fa3effff05006700070004
[... cut ...]
157102-2022-12-25T07:25:19.200Z zigbee-herdsman:adapter:ezsp:uart > DATA (5,3,0): 2d0001340000fa3e04010000010140010000cccd0518580b0a00
157103-2022-12-25T07:25:19.200Z zigbee-herdsman:adapter:ezsp:uart > [536f21a9602a154867904b25aa5493099d4e276720cb7fd3f6cc633f7d337e]
157104-2022-12-25T07:25:19.200Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (6)
157105-2022-12-25T07:25:19.209Z zigbee-herdsman:adapter:ezsp:uart < [366fa1a9602a15d7b7837e]
157106-2022-12-25T07:25:19.209Z zigbee-herdsman:adapter:ezsp:uart < DATA (3,6,0): 366fa1a9602a15d7b7837e
157107-2022-12-25T07:25:19.209Z zigbee-herdsman:adapter:ezsp:uart > ACK  (4)
```

vignez2991, 2022-12-26:
> @Koenkk 
I have configured 10+ devices using zigbee2Mqtt with CC2652R Coordinator, Everything is working fine. When I restarted my Gateway sometimes working fine, but sometime devices disappeared in zigbee2Mqtt device list. At the same time I can see the device feedback log in zigbee-herdsman. At this stage when I restored the data folder from backup and restarted zigbee2Mqtt service everything comes to work normal.

vignez2991, 2022-12-28:
> > @Koenkk
> > I have configured 10+ devices using zigbee2Mqtt with CC2652R Coordinator, Everything is working fine. When I restarted my Gateway sometimes (approximately 9/10 times) working fine, but sometime(approximately 1/10 times) devices disappeared in zigbee2Mqtt device list. At the same time I can see the device response in zigbee-herdsman log. And also when I restore the data folder from backup and restarted zigbee2Mqtt service everything comes to work normal.
> > [zigbee2mqtt.txt](https://github.com/Koenkk/zigbee2mqtt/files/10302814/zigbee2mqtt.txt)
> 
> please find the attached log file with device response separated at the end of the file.
@Koenkk Hii Koenkk I have attached log data again with herdsman debug enabled.
[device-remove-log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10313322/device-remove-log.txt)

wiinxt, 2023-01-03:
I've the similar problem as myuc, I lost devices connected directly to coordinator randomly after long period or after restart (but I never lost routers connected to coordinator for now).
I tried disabling the USB auto-suspend feature but that didn't change anything. (I see this in [FAQ](https://www.zigbee2mqtt.io/guide/faq/#zigbee2mqtt-crashes-after-some-time))

I use Raspberry pi 3B with docker
Zigbee2MQTT version: 1.29 (commit: dddce57)
Coordinator type/revision: EZSP v8 6.10.3.0 build 297 - ZBDongle-E

I just configured the logs, I'll post them if it happens again

vignez2991, 2023-01-09:
Hii @Koenkk Any Update??

I am Using Ubuntu 20.04 OS.
And This happens only on power restart and not on soft reboot.

medictroy, 2023-01-11:
I have a similar problem with the Sonoff ZB Dongle-E in coordinator mode.
(ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"), 
Adapter firmware version: 6.10.3.0 build 297
Zigbee2MQTT version: 1.29.1

When I get a device to join the network, it says connected for approx. 10-15 seconds and then leaves the network.
When it is connected I can control the device with no problems.
This happens every time I try to get a TuYa TS0013 switch to join.
https://www.zigbee2mqtt.io/devices/TS0013.html
These switches worked fine with my old Conbee II. 
Any thoughts?

Koenkk, 2023-01-12:
@vignez2991 which device left in this log?

vignez2991, 2023-01-12:
> @vignez2991 which device left in this log?

All devices.

Koenkk, 2023-01-12:
@vignez2991 it seems your `data/database.db` is cleared for some reason. Since you indicated that this only happens on a hard restart, could it be that this creates problems with your file system dropping the files?

vignez2991, 2023-01-12:
Yes, Exactly this is was the issue.

vignez2991, 2023-01-12:
When I restore data folder which was backed up early, Everything works normal.

Koenkk, 2023-01-13:
@vignez2991 I don't expect z2m to be the root cause of this. You have to check what your system is doing.

myuc, 2023-01-17:
@Koenkk could you take a look at my above comment? Link: https://github.com/Koenkk/zigbee2mqtt/issues/15299#issuecomment-1364577939

It seems unrelated to vignez2991's problem. Should I move my comments to a separate issue?

vignez2991, 2023-01-18:
> @vignez2991 I don't expect z2m to be the root cause of this. You have to check what your system is doing.

@Koenkk This happens only when the z2m service was active. It looks this file was kept opened while running z2m at that time when I restart power this file gets truncated. (Note: I am using ubuntu 20.04 OS)

Koenkk, 2023-01-18:
What if you:
- start z2m; pair devices
- restart z2m -> devices should still be there
- restart power
- start z2m -> are devices missing now?

vignez2991, 2023-01-19:
> What if you:
> 
> * start z2m; pair devices
> * restart z2m -> devices should still be there
> * restart power
> * start z2m -> are devices missing now?

Yes, After restarting power and starting z2m devices are missing (1 out of 3 tries).

Koenkk, 2023-01-19:
Can you check the following:
- restart z2m -> devices should still be there
  - check the data/database.db, are the devices still there?
- restart power
  - check the data/database.db, are the devices still there?

vignez2991, 2023-01-20:
> 

1. restart z2m  --> Yes, All devices exist.

2. restart power  --> No, The File was empty

Koenkk, 2023-01-21:
And what in case of:
- start z2m
- stop z2m
- check the data/database.db, are the devices still there?
- restart power
- check the data/database.db, are the devices still there?

vignez2991, 2023-01-21:
> 
- Start z2m --> Yes, All devices exist.
- Stop z2m --> Yes, All devices exist.
- Power restart z2m --> Yes, All devices exist (Disabled Auto start z2m and restarted 5 times)

Koenkk, 2023-01-22:
Interesting, so that goes well.

In the scenario:
1. start z2m
2. restart power
3. stat z2m

Can you provide the herdsman debug logging of both 1 and 3?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

vignez2991, 2023-01-23:
> Interesting, so that goes well.
>
> In the scenario:
>
> 1. start z2m
> 2. restart power
> 3. stat z2m
>
> Can you provide the herdsman debug logging of both 1 and 3?
>
> See [...]

I have attached log file for both the cases
[after_power_restart.log](https://github.com/Koenkk/zigbee2mqtt/files/10478931/after_power_restart.log)
[before_power_restart.log](https://github.com/Koenkk/zigbee2mqtt/files/10478932/before_power_restart.log)

1 - File name before_power_restart.log
3 - File name after_power_restart.log

riversideneyon, 2023-01-26:
I have same issue since couple weeks. Always specific 1 device disappearing after some time. It's also the most recent added device since couple weeks.

Device is philips hue motion sensor. 
Stick is zigstar V4 if I'm not mistaken. 

I can enable debug but will that not be to big file since it happens after couple days, i can try reboot since it can be related to it.

Koenkk, 2023-01-29:
@vignez2991 at point 3, it seems the database is indeed gone, I don't see anything about z2m clearing it. Between step 1 and 2, is the data/database.db filled with the paired devices?

vignez2991, 2023-01-30:
Step 1 - data/database.db contains paired devices
Step 2 - data/database.db file is empty.

In Linux sometimes the file gets truncated While doing a power restart when the file is opened with write mode (Or opened by any process).
This could be the reason for the database.db file getting empty.

Koenkk, 2023-01-30:
@vignez2991 z2m only writes/open the database occasionally, the chance that every time your power restart z2m is doing a write on the database (which should take a very short amount of time) is astronomically low.

vignez2991, 2023-01-31:
Yes But, When i stopped z2m and tried power restart several times database file contains actual data. The database file gets empty only when i do power restart after running z2m service

Grazulex, 2023-02-05:
Hello, do we have a fixed, I have the same issue under Jeedom and 1.30.0  Version PHP installée : 7.3.31-1~deb10u2.
my controler is a "Zigbee 3.0 USB Dongle Plus Gateway Con Antena, EFR32MG21 + CH9102F Coordinator EFR32MG21" and it's always the 3 sames devices (Aqara E1) that are disappearing after a reboot.......If I can help to find a quick fixed, let me know :-(

Wattrider, 2023-02-05:
Hi,

same for me

Virtual machine Home Assistant 2023.1.7
Sonoff Zigbee 3.0 V2 dongle (ezsp)
Zigbee2mqtt version 1.30.0-1

I have 20 zigbee devices (most sonoff and moes) and 3 are disapearing during the night for example.

If i can help please tell

[52 more replies]

closed 2023-04-14 by github-actions[bot]