Koenkk/zigbee2mqtt #15300: Cannot start Zigbee2mqtt after updating to 1.28.4-1

Schmee323, 2022-12-01:
### What happened?

Updated to latest Zigbee2MQTT version 1.28.4-1.
Now cannot start integration in Home Assistant.

### What did you expect to happen?

Integration to start as it was before version upgrade.

### How to reproduce it (minimal and precise)

clikc start and monitor log

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

cant find this data

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

[08:45:55] INFO: Preparing to start...
[08:45:55] INFO: Socat not enabled
[08:45:56] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

troelde, 2022-12-02:
Same for me when I try to restart after upgrading to 1.28.4-1. No entries in zigbee2mqtt log. Supervisor log: "Image zigbee2mqtt/zigbee2mqtt-aarch64:1.28.4-1 does not exist for addon_45df7312_zigbee2mqtt".  Please fix this fast, or give me some assistance how to downgrade.

schmielinski, 2022-12-02:
@troelde I have exactly the same problem. Only this info and no additional logs.

Schmee323, 2022-12-02:
Seems to be related to the following custom converter:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3210_raqjcxo5'}],
    zigbeeModel: ['TS011F'],
    model: 'SPP02',
    vendor: 'Mercator',
    description: 'Mercator Ikuu Double Indoor Power Point',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off],
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right'), e.power().withEndpoint('left'), e.current().withEndpoint('left'), e.voltage().withEndpoint('left').withAccess(ea.STATE), e.energy()],
    endpoint: (device) => {
            return {left: 1, right: 2};
    },
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    meta: {multiEndpoint: true}, 
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genBasic', 'genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint1);
        await reporting.onOff(endpoint1);
        await reporting.onOff(endpoint1);
        await reporting.rmsVoltage(endpoint1, {change: 5});
        await reporting.rmsCurrent(endpoint1, {change: 50});
        await reporting.activePower(endpoint1, {change: 1});
        await reporting.onOff(endpoint1);
        await reporting.onOff(endpoint2);
        endpoint1.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint1.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
	device.save();
    },
};

module.exports = definition;

This device is now detected as a single smart plug by zigbee2mqtt if i disable the converter and i lose the second outlet as it is a Mercator double GPO.

JackTalisker, 2022-12-02:
coing from #15285 
the following is the external converter which prevents z2m from starting (modified from a @kvazis idea):

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_u5u4cakc'},],
        model: 'TS011F_plug_3',
        description: 'Blitzwolf BW SHP-15',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1, acVoltageDivisor: 1, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1,
                acPowerDivisor: 1,
            });
        },
        // This device doesn't support reporting correctly.
        // https://github.com/Koenkk/zigbee-herdsman-converters/pull/1270
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage')],
        onEvent: (type, data, device, options) => {
            const endpoint = device.getEndpoint(1);

            if (type === 'stop') {
                clearInterval(globalStore.getValue(device, 'interval'));
                globalStore.clearValue(device, 'interval');
            } else if (!globalStore.hasValue(device, 'interval')) {
                const seconds = options && options.measurement_poll_interval ? options.measurement_poll_interval : 20;
                const interval = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                        await endpoint.read('seMetering', ['currentSummDelivered']);
                    } catch (error) {/* Do nothing*/}
                }, seconds*1000);
                globalStore.putValue(device, 'interval', interval);
            }
        }
};

module.exports = definition;
```

ph0nebooth, 2022-12-02:
@Schmee323 I have the exact same issue. My Mercator Ikuu SPP02G is being detected as a Tuya Smart Plug (single socket). I was using the same converter and it now doesn't allow Z2M to start.

I'll let you know if I work out what's wrong with the converter.

liquidacid, 2022-12-02:
> Same for me when I try to restart after upgrading to 1.28.4-1. No entries in zigbee2mqtt log. Supervisor log: "Image zigbee2mqtt/zigbee2mqtt-aarch64:1.28.4-1 does not exist for addon_45df7312_zigbee2mqtt". Please fix this fast, or give me some assistance how to downgrade!!

@troelde 
this is not the same issue as mentioned in OP but you can downgrade to the previous version.

settings->system->backups->select the backup with the size of about 0.1MB which has addon....zigbee2mqtt in its name. It will for you fix the issue. Im not sure if you can update to the latest version again with no isssues, i did not try that.

impankratov, 2022-12-02:
For me issue of `TypeError: Cannot read properties of undefined (reading 'options')`
error  was resolved by changing export statement in custom converter file (one at the very end) to an array.

```diff
const device = {
  ...(custom converter data)
}

-module.exports = device;
+module.exports = [device];
```

troelde, 2022-12-02:
> @troelde this is not the same issue as mentioned in OP but you can downgrade to the previous version.
>
> settings->system->backups->select the backup with the size of about 0.1MB which has [...]

Thank you. I created a new issue

YevgenZhylenkov, 2022-12-02:
Same problem for me :(

novacekji, 2022-12-24:
Have the same problem on my boih installation with SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

closed 2023-01-31 by github-actions[bot]