Koenkk/zigbee2mqtt #15302: [New device support]: 2 gang switch with Energy Monitoring - _TZ3000_irrmjcgi - TS0002

cloudbr34k84, 2022-12-01:
### Link

https://www.aliexpress.com/item/1005004345234213.html?spm=a2g0o.order_list.order_list_main.105.10ff18024SmnOC

### Database entry

IEEE:  0xa4c1389a34414fb0 Nwk:358D Device Type: Router LQI: 145 RSSI: Unknown Last Seen: 2022-01-07T08:41:17 Power Source: Wired

### Comments

I have installed this zigbee module and i was hoping we could look at exposing the energy monitoring. Im new to Z2M, but liking it better than ZHA.

MityaDVG, 2022-12-03:
Duplicate of #4842

MityaDVG, 2022-12-03:
Duplicate of [https://github.com/Koenkk/zigbee-herdsman-converters/issues/4842#issue-1427073573](url)

cloudbr34k84, 2022-12-04:
I don't see anything in the duplicate the that relates to my request?

closed 2022-12-17 by cloudbr34k84