Koenkk/zigbee2mqtt #15303: IKEA E1525/E1745 Motion sensor pairs, but does not return occupancy state

EricPalmquist, 2022-12-01:
### What happened?

I am migrating a Conbee from Deconz to z2m.  All of my devices repaired after being power cycled, and all are working great except both of my IKEA motion sensors, which worked fine on Deconz and now only return the battery and LQI.  I had a third one still in a box, paired it and get the same result.

IKEA firmware = 2.0.022

State is reported as:
{
    "battery": 100,
    "linkquality": 255
}

### What did you expect to happen?

Should be reading an occupancy state.

### How to reproduce it (minimal and precise)

It happens with all of these devices.

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

0x26400500

### Adapter

Conbee

JochemLangerak, 2022-12-29:
Same over here after pairing:

```
{
    "last_seen": "2022-12-29T15:10:01+01:00",
    "linkquality": 110,
    "battery": 47
}
```

mozzbozz, 2023-01-05:
Can confirm. Seems to be related to #12611

closed 2023-02-13 by github-actions[bot]