Koenkk/zigbee2mqtt #15304: TypeError: argument of type 'float' is not iterable

UPEngineer, 2022-12-02:
### What happened?

I have noticed that since I have updated Zigbee2Mqtt, I am getting a lot of TypeErrors for my zigbee devices. 

Here are a couple of examples:

`Exception in handle_state_message_received when handling msg on 'zigbee2mqtt/Christmas Tree': '{"linkquality":255,"state":"ON","update":{"installed_version":"2.1","state":"available"},"update_available":true}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/update.py", line 191, in handle_state_message_received
    if "installed_version" in json_payload:
TypeError: argument of type 'float' is not iterable`

and

`2022-12-01 18:45:10.001 ERROR (MainThread) [homeassistant.util.logging] Exception in handle_state_message_received when handling msg on 'zigbee2mqtt/Light - Back Yard': '{"action":null,"activeEnergyReports":null,"activePowerReports":null,"autoTimerOff":null,"buttonDelay":null,"defaultLed1ColorWhenOff":null,"defaultLed1ColorWhenOn":null,"defaultLed1IntensityWhenOff":null,"defaultLed1IntensityWhenOn":null,"defaultLed2ColorWhenOff":null,"defaultLed2ColorWhenOn":null,"defaultLed2IntensityWhenOff":null,"defaultLed2IntensityWhenOn":null,"defaultLed3ColorWhenOff":null,"defaultLed3ColorWhenOn":null,"defaultLed3IntensityWhenOff":null,"defaultLed3IntensityWhenOn":null,"defaultLed4ColorWhenOff":null,"defaultLed4ColorWhenOn":null,"defaultLed4IntensityWhenOff":null,"defaultLed4IntensityWhenOn":null,"defaultLed5ColorWhenOff":null,"defaultLed5ColorWhenOn":null,"defaultLed5IntensityWhenOff":null,"defaultLed5IntensityWhenOn":null,"defaultLed6ColorWhenOff":null,"defaultLed6ColorWhenOn":null,"defaultLed6IntensityWhenOff":null,"defaultLed6IntensityWhenOn":null,"defaultLed7ColorWhenOff":null,"defaultLed7ColorWhenOn":null,"defaultLed7IntensityWhenOff":null,"defaultLed7IntensityWhenOn":null,"defaultLevelLocal":null,"defaultLevelRemote":null,"dimmingSpeedDownLocal":null,"dimmingSpeedDownRemote":null,"dimmingSpeedUpLocal":null,"dimmingSpeedUpRemote":null,"doubleTapClearNotifications":null,"doubleTapUpForFullBrightness":null,"energy":null,"firmwareUpdateInProgressIndicator":null,"invertSwitch":null,"ledColorWhenOff":null,"ledColorWhenOn":null,"ledIntensityWhenOff":null,"ledIntensityWhenOn":null,"linkquality":255,"loadLevelIndicatorTimeout":null,"localProtection":null,"maximumLevel":null,"minimumLevel":null,"onOffLedMode":null,"outputMode":null,"periodicPowerAndEnergyReports":null,"power":91.8,"powerType":null,"rampRateOffToOnLocal":null,"rampRateOffToOnRemote":null,"rampRateOnToOffLocal":null,"rampRateOnToOffRemote":null,"relayClick":null,"remoteProtection":null,"smartBulbMode":null,"state":null,"stateAfterPowerRestored":null,"switchType":null,"update":{"installed_version":"2.00","state":null},"update_available":null}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/update.py", line 191, in handle_state_message_received
    if "installed_version" in json_payload:
TypeError: argument of type 'float' is not iterable`

Never noticed it before, but I am seeing it quite a bit today.

### What did you expect to happen?

No errors.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

 0x26720700

### Adapter

Conbee II

UPEngineer, 2022-12-02:
I think this is related to https://github.com/Koenkk/zigbee2mqtt/issues/15290 as well.

Koenkk, 2022-12-05:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/15290

closed 2022-12-05 by Koenkk