Koenkk/zigbee2mqtt #15306: week programming string not saved for workdays  any more since 1.28.3 for Tuya TS0601 thermostate

ebeneezer, 2022-12-02:
### What happened?

- week programming could be done for workday and holiday independently - does not work any more button is ghosted until both settings are filled even if i do not want to change the holiday setting just the workday setting

- week programming now needs "" around the configuragtion string like "06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C" that wasnt the case before (minor)

- week programming now needsto always select the type of program (7 days, 5+2 etc) even if i do not want to change the item

- week programming string is not saved to device config (checked via status function retrieving the devices json) MAJOR problem

- holiday setting is saved correctly !

### What did you expect to happen?

workday and holiday string should be saved to the device

the holiday mode (7 days 5+2) should be pre selected from actual dev settings

working day setting and holiday setting should be configurable independently as it was before

### How to reproduce it (minimal and precise)

as described

### Zigbee2MQTT version

 1.28.3

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Debug log

2022-12-02 04:26:45MQTT publish: topic 'zigbee2mqtt/Heizkörper Büro/device-softwareBuildID', payload ''
Info 2022-12-02 04:26:45MQTT publish: topic 'zigbee2mqtt/Heizkörper Büro/valve_detection', payload ''
Debug 2022-12-02 04:26:52Received Zigbee message from 'Heizkörper Diele', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":3594}' from endpoint 1 with groupID null
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCK","comfort_temperature":22,"current_heating_setpoint":21,"device":{"applicationVersion":91,"dateCode":"","friendlyName":"Heizkörper Diele","hardwareVersion":1,"ieeeAddr":"0x804b50fffe1a8652","manufacturerID":4098,"manufacturerName":"_TZE200_cwnjrr72","model":"TS0601_thermostat","networkAddress":2775,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"eco_temperature":19,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"holidays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C","last_seen":"2022-12-02T04:26:51+01:00","linkquality":255,"local_temperature":20,"local_temperature_calibration":-1,"max_temperature":30,"min_temperature":15,"position":5,"preset":"schedule","running_state":"heat","system_mode":"auto","update":{"installed_version":"unknown","state":"idle"},"update_available":false,"valve_detection":null,"week":"7","window_detection":"OFF","window_detection_params":{"minutes":101,"temperature":87},"window_open":false,"workdays":[{"hour":6,"minute":0,"temperature":22},{"hour":9,"minute":0,"temperature":20},{"hour":11,"minute":30,"temperature":20},{"hour":12,"minute":30,"temperature":20},{"hour":16,"minute":30,"temperature":21},{"hour":20,"minute":0,"temperature":20}],"workdays_schedule":"06:00/22°C 09:00/20°C 11:30/20°C 12:30/20°C 16:30/21°C 20:00/20°C"}'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/min_temperature', payload '15'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/max_temperature', payload '30'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/current_heating_setpoint', payload '21'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/local_temperature', payload '20'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/preset', payload 'schedule'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/away_mode', payload 'OFF'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/child_lock', payload 'UNLOCK'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/local_temperature_calibration', payload '-1'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/window_detection', payload 'OFF'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/window_detection_params-temperature', payload '87'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/window_detection_params-minutes', payload '101'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/boost_time', payload '300'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/system_mode', payload 'auto'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/force', payload 'normal'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/comfort_temperature', payload '22'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/eco_temperature', payload '19'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/position', payload '5'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/running_state', payload 'heat'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/battery_low', payload 'false'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/week', payload '7'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/workdays', payload '[object Object],[object Object],[object Object],[object Object],[object Object],[object Object]'
Info 2022-12-02 04:26:52MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/workdays_schedule', payload '06:00/22°C 09:00/20°C 11:30/20°C 12:30/20°C 16:30/21°C 20:00/20°C'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/holidays', payload '[object Object],[object Object],[object Object],[object Object],[object Object],[object Object]'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/holidays_schedule', payload '06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/away_preset_temperature', payload '15'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/window_open', payload 'false'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/auto_lock', payload 'MANUAL'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/away_preset_days', payload '1'
[... 44 lines cut ...]
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/holidays_schedule', payload '06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/away_preset_temperature', payload '15'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/window_open', payload 'false'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/auto_lock', payload 'MANUAL'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/away_preset_days', payload '1'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/linkquality', payload '255'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/update-state', payload 'idle'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/update-installed_version', payload 'unknown'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/update_available', payload 'false'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/last_seen', payload '2022-12-02T04:26:51+01:00'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/device-friendlyName', payload 'Heizkörper Diele'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/device-model', payload 'TS0601_thermostat'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/device-ieeeAddr', payload '0x804b50fffe1a8652'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/device-networkAddress', payload '2775'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/device-type', payload 'EndDevice'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/device-manufacturerID', payload '4098'
[... cut ...]
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/device-zclVersion', payload '3'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/device-hardwareVersion', payload '1'
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/device-dateCode', payload ''
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/device-softwareBuildID', payload ''
Info 2022-12-02 04:26:53MQTT publish: topic 'zigbee2mqtt/Heizkörper Diele/valve_detection', payload ''

neovinter, 2022-12-04:
Same problem. The apply button is disabled until you select "Week format user for schedule". But "workdays_schedule" cannot be set anyway.

PotcFdk, 2022-12-05:
Didn't the two schedule text fields contain the current active schedule strings in the past? I haven't edited any for quite a long time, but it seems odd to me that they start empty in the UI.

ebeneezer, 2022-12-06:
I used them to copy the correspondig composite string out of the status json, then changing after my needs and writing back to the device - worked a charm besides would have been nice to be pre filled by the UI. Now I have to publish a string via an mqtt client everytime to amend the week program in all thermostats - this is a pain in the a...

m0wlheld, 2022-12-09:
Just bought a thermostat of that type and facing issue when setting workday/holidays schedules. What's the intendent format for values in `workdays_schedule` and `holidays_schedule` anyhow? I tried "06:00/21°C 19:00/17°C" but that gives me error:

> '0x00...' failed: 'TypeError: Cannot read properties of undefined (reading 'split')'

PotcFdk, 2022-12-09:
> I tried "06:00/21°C 19:00/17°C"

The format you've tried is correct, but you need 6 values instead of just two.

m0wlheld, 2022-12-09:
> > I tried "06:00/21°C 19:00/17°C"
> 
> The format you've tried is correct, but you need 6 values instead of just two.

Six? Is that a fixed count or is it related to filed "Week format user for schedule" for which I choose "5+2"?

PotcFdk, 2022-12-09:
> Six? Is that a fixed count or is it related to filed "Week format user for schedule" for which I choose "5+2"?

That is a fixed count, six values per day. It's a vendor limitation.
If you pick 5+2 that means you have two schedules, one for working days, one for weekends. Both have six fields.

KalleKal, 2022-12-14:
> 

do not write "°C", just HH:MM/C like 19:00/17

ebeneezer, 2022-12-15:
Tnx for the hint - but it should be fixed anyways to accept at least the format that is used by the devices status json. Else it would somewhat be inconsistent.

vcirstea, 2023-01-02:
Same issue happens on my end as well. I have 6 TRVs.. used to work like a charm 3 weeks ago or so..

pakagi, 2023-01-28:
The same for me, using 1.29.2. The workdays_schedule and holidays_schedule used to show current settings, but now they are empty, and if values are copied to them from State and modified and applied, only the holidays_schedule takes effect. The workdays_schedule does not work and old settings are still there in the thermostat valve (and show in state).

pakagi, 2023-02-12:
This bug is still there using new 1.30.1.

ottoketney, 2023-02-15:
I have the same behaviour with 1.30.1-1

pakagi, 2023-03-03:
This bug is still there even in new 1.30.2 :-(

ottoketney, 2023-03-03:
I have the same behaviour with 1.30.2

ebeneezer, 2023-04-03:
No changee - still nothing was fixed.

ottoketney, 2023-04-03:
is still present in curent version 1.30.3-1

[image]

ebeneezer, 2023-05-05:
Still no correct function :-(

ottoketney, 2023-05-05:
No changee - still nothing was fixed.

pakagi, 2023-06-06:
AFAIK still not fixed :-(

PotcFdk, 2023-07-10:
> This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 7 days

bump

ebeneezer, 2023-07-10:
Please devs fix this. It is bot working for soooo long.Pleeeeeeeeaaaazzzzzyyyy!

ebeneezer, 2023-08-11:
Crap does not work  (like the months before)

pakagi, 2023-09-11:
ping ! still valid :-(

git-tail-d, 2023-10-03:
it is still valid

Koenkk, 2023-10-15:
> Tnx for the hint - but it should be fixed anyways to accept at least the format that is used by the devices status json. Else it would somewhat be inconsistent.
> 
> BTW: Did not work for me.

@ebeneezer 

- Is this the only issue are are there others aswell?
- Can you provide the data/database.db entry of your device?

ebeneezer, 2024-01-01:
Fixed after 1 year.

Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you! Thank you!

closed 2024-01-01 by ebeneezer