Koenkk/zigbee2mqtt #15308: Wrong Battery calculation with Xiaomi-Devices

Diet1976, 2022-12-02:
### What happened?

Since the last update of IOBRoker.Zigbee, my battery percentages of all of my windows-contacts/motion sensors are really low, even the Volt-Value is near 100%.
I didn't find a change logged in the adapter history?

Greetings, Dietmar

### Zigbee2MQTT version

1.7.6

### Adapter firmware version

0x26690700

### Adapter

ZigBee-Adapter 1.7.6

Koenkk, 2022-12-02:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/8499

closed 2022-12-02 by Koenkk

Diet1976, 2022-12-02:
Hello!Are you sure, that it is a duplicate? Because the other issue describes the opposite? My Devices were at 90-100% and suddenly they are all low?Greets, Dietmar