Koenkk/zigbee2mqtt #15309: After upgrade to 1.28.4-1 the add-on failed to start up

troelde, 2022-12-02:
### What happened?

No entries in zigbee2mqtt log. Supervisor log: "Image zigbee2mqtt/zigbee2mqtt-aarch64:1.28.4-1 does not exist for addon_45df7312_zigbee2mqtt". 

### What did you expect to happen?

Please fix this fast, or give me some assistance how to downgrade!!

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

Not available 

### Adapter

ConBee II

### Debug log

22-12-02 10:40:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:15 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
[... cut ...]
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 WARNING (SyncWorker_3) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
22-12-02 10:40:18 ERROR (SyncWorker_3) [supervisor.docker.manager] Image zigbee2mqtt/zigbee2mqtt-aarch64:1.28.4-1 does not exist for addon_45df7312_zigbee2mqtt

troelde, 2022-12-03:
I managed to downgrade, so I am back on 1.28.2-1 which works.

rodrigogbs, 2022-12-04:
same here:

```
[10:45:52] INFO: Preparing to start...
[10:45:53] INFO: Socat not enabled
[10:45:55] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
```

Gerd2809, 2022-12-09:
I have the same problem

Koenkk, 2022-12-09:
If you get an error like:

```
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
```

It means that any of your external converters are not compatible with the current z2m release. Update or remove your external converter to fix the issue.

rodrigogbs, 2022-12-10:
> If you get an error like:
> 
> ```
> /app/node_modules/zigbee-herdsman-converters/index.js:91
>         if (converter.options) {
> ```
> 
> It means that any of your external converters are not compatible with the current z2m release. Update or remove your external converter to fix the issue.

@Koenkk, thanks for your reply, but I am not understanding what is wrong with the external converters...I can't just remove them, otherwise my devices won't work...

Where can I find this information?

Koenkk, 2022-12-10:
@rodrigogbs external converters are not made for permanent usage, only while testing (since they can break over releases). The correct solution would be to integrate the external converter for out-of-the-box support. 

- Can you post the code of this external converter?
- Can you provide a link of the device?

Gerd2809, 2022-12-14:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS011SB'],
    model: 'Power Monitor Switch',
    vendor: 'Aubess',
    description: '[Fucking device]',
       fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory, fz.tuya_relay_din_led_indicator],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.tuya_relay_din_led_indicator],
        whiteLabel: [{vendor: 'Aubess', model: 'ATMS1602Z'}],
        ota: ota.zigbeeOTA,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE), e.energy(), exposes.enum('power_outage_memory', 
        	ea.ALL, ['on', 'off', 'restore']).withDescription('Recover state after power outage'),
            exposes.enum('indicator_mode', ea.STATE_SET, ['off', 'on_off', 'off_on']).withDescription('Relay LED indicator mode')],
        
};
module.exports = definition;
`

Koenkk, 2022-12-15:
@Gerd2809 is there a model number on the back of this device?

Gerd2809, 2022-12-15:
@Koenkk on the back is only the pairing button and on the board is FL-S104A-V1. 0 printed

Koenkk, 2022-12-15:
@Gerd2809 please check if everything works with https://gist.github.com/Koenkk/d7c5ccc2c11877e72f4d66330f30ce08

Gerd2809, 2022-12-20:
@Koenkk 
system is runing with this script, but the device is detected as 
[image]

Koenkk, 2022-12-20:
@Gerd2809 can you provide the data/database.db entry of this device?

Gerd2809, 2022-12-21:
@Koenkk 

`{"id":29,"type":"Router","ieeeAddr":"0xa4c138108a4d11fa","nwkAddr":12760,"manufId":4417,"manufName":"_TZ3000_xkap8wtb","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"65503":"�_�*\u0003","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_xkap8wtb","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"seMetering":{"attributes":{"currentSummDelivered":[0,1212],"divisor":100,"multiplier":1}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":232,"rmsCurrent":0,"activePower":0,"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2,"tuyaBacklightMode":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2834279,"53251":127,"53252":10912,"53253":1270,"switchType":0,"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0022811af8","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0022811af8","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0022811af8","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1671617339924,"defaultSendRequestWhen":"immediate"}`

The Story of this device.
- bought from aliexpress
- it was recognized as TS0001_power when first pared, but it kept turning itself off like TS011F
- I added the ota update for the device to the tuya.js file and after the update the device responded with TS011F
- now it works very well, but is shown as TS011F
- I took a second device and I connected it directly to a tuya app/hub, it received an update and is now also reporting as TS011F

Koenkk, 2022-12-21:
@Gerd2809 since the device itself advertises as a TS011F and it is detected as a TS011F, what is the problem?

Gerd2809, 2022-12-21:
@Koenkk 
the only problem was that the upgrade to 1.28.4-1 with which the external converter didn't start.
I'm currently running it without the external adapter, it just shows the wrong picture and the wrong manufacturer

[image]

Koenkk, 2022-12-21:
Aubess is just a whitelabel, inside is a TuYa device. I'll check about the picture.

rodrigogbs, 2022-12-22:
> @rodrigogbs external converters are not made for permanent usage, only while testing (since they can break over releases). The correct solution would be to integrate the external converter for [...]

@Koenkk, these ate the external converters that I use here in my installation, for example, I am developing 20A power plugs with Zemismart, they are being released to the market now, so I am still testing the products and can ask for official integration on Z2M only at that time.

```
>> YMC-420D_ext_converter.js

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const fzLocal = {
  action_source_name: {
    cluster: "closuresDoorLock",
    type: "raw",
    convert: (model, msg, publish, options, meta) => {
      const lookup = {
          0: 'password_unlock', //'unknown', // password 15/07
          1: 'unlock', //'lock', unlock system 15/07
          2: 'auto_lock', // 'unlock', // one touch lock and autolock 15/07
          3: 'RFID_unlock', //'lock_failure_invalid_pin_or_id', // card 15/07
          4: 'fingerprint_unlock', //'lock_failure_invalid_schedule',  // fingerprint 15/07
          5: 'unlock_failure_invalid_pin_or_id',
          6: 'unlock_failure_invalid_schedule',
          7: 'one_touch_lock',
          8: 'key_lock',
          9: 'key_unlock',
          10: 'auto_lock',
          11: 'schedule_lock',
          12: 'schedule_unlock',
          13: 'manual_lock',
          14: 'manual_unlock',
          15: 'non_access_user_operational_event',
      };
      const value = lookup[msg.data[3]];
      return { action_source_name: value }
    },
  },
  action_source_user: {
    cluster: "closuresDoorLock",
    type: "raw",
    convert: (model, msg, publish, options, meta) => {
    return { action_source_user: msg.data[5] };
    },
  },
};

const lockExtend = (meta) => {
  return {
    fromZigbee: [fzLocal.action_source_user, fzLocal.action_source_name, fz.lock, fz.battery, fz.lock_operation_event, fz.lock_programming_event, fz.lock_pin_code_response, fz.lock_user_status_response],
    toZigbee: [tz.lock, tz.pincode_lock, tz.lock_userstatus],
    meta: {pinCodeCount: 250, ...meta},
    exposes: [e.lock(), e.lock_action(), e.battery(), e.pincode()],
    configure: async (device, coordinatorEndpoint, logger) => {
      const endpoint = device.getEndpoint(1);
      await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
      await reporting.lockState(endpoint);
      await reporting.batteryPercentageRemaining(endpoint);
    },
  };
};

module.exports = [
  {
    zigbeeModel: ["YMC 420 D"],
    model: "YMC 420 D",
    vendor: "Yale",
    description: "YMC 420 D",
[... 78 lines cut ...]
e.switch().withEndpoint('l3'),
exposes.enum('power_on_behavior', ea.ALL, ['on', 'off', 'previous']),
exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']),
],
endpoint: () => {
return {'l1': 1, 'l2': 2, 'l3': 3};
},
meta: {multiEndpoint: true},
configure: async (device, coordinatorEndpoint) => {
const firstEndpoint = device.getEndpoint(1);
await firstEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(firstEndpoint, coordinatorEndpoint, ['genOnOff']);

const secondEndpoint = device.getEndpoint(1);
await secondEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(secondEndpoint, coordinatorEndpoint, ['genOnOff']);

const thirdEndpoint = device.getEndpoint(1);
await thirdEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(thirdEndpoint, coordinatorEndpoint, ['genOnOff']);
}
};

module.exports = definition;
```

Koenkk, 2022-12-22:
Can you remove the external converter 1-by-1, which one causes the error?

rodrigogbs, 2022-12-22:
> Can you remove the external converter 1-by-1, which one causes the error?

I have just realized that with the first one (YMC-420D_ext_converter.js) the Z2M starts without the `/app/node_modules/zigbee-herdsman-converters/index.js:91` error, but even without any external converter configured, the Z2m configuration screen doesn't open, showing Bad gateway error. I don't know if it is related to this error below on log, that was already fixed (https://github.com/Koenkk/zigbee2mqtt/issues/15386):

```
Zigbee2MQTT:error 2022-12-22 14:38:00: Failed to configure 'Interruptores Banheiro', attempt 1 (Error: Read 0xa4c13806f841ffe9/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32))
```

So, the error related to this topic occurs with these 2 converters:

```
>> ZM-TB242-20A_ext_converter.js

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [
{
modelID: 'TS011F',
manufacturerName: '_TZ3000_6s5dc9lx'
}
],
model: 'TS011F_2_gang',
vendor: 'TuYa',
description: '2 gang wall outlet 20A',
toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_backlight_mode]),
fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior, fz.tuya_backlight_mode]),
exposes: [
e.switch().withEndpoint('l1'),
e.switch().withEndpoint('l2'),
exposes.enum('power_on_behavior', ea.ALL, ['on', 'off', 'previous']),
exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']),
],
endpoint: () => {
return {'l1': 1, 'l2': 2};
},
meta: {multiEndpoint: true},
configure: async (device, coordinatorEndpoint) => {
const firstEndpoint = device.getEndpoint(1);
await firstEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(firstEndpoint, coordinatorEndpoint, ['genOnOff']);

const secondEndpoint = device.getEndpoint(1);
await secondEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(secondEndpoint, coordinatorEndpoint, ['genOnOff']);
}
};

module.exports = definition;
```

```
>> ZM-L03E-Z-20A_ext_converter.js

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [
{
modelID: 'TS0003',
manufacturerName: '_TZ3000_f09j9qjb'
}
],
model: 'ZM-L03E-Z',
vendor: 'Zemismart',
description: 'Smart light switch - 3 gang with neutral 20A wire CUSTOM',
toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_backlight_mode]),
fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior, fz.tuya_backlight_mode]),
exposes: [
e.switch().withEndpoint('l1'),
e.switch().withEndpoint('l2'),
e.switch().withEndpoint('l3'),
exposes.enum('power_on_behavior', ea.ALL, ['on', 'off', 'previous']),
exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']),
],
endpoint: () => {
return {'l1': 1, 'l2': 2, 'l3': 3};
},
meta: {multiEndpoint: true},
configure: async (device, coordinatorEndpoint) => {
const firstEndpoint = device.getEndpoint(1);
await firstEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(firstEndpoint, coordinatorEndpoint, ['genOnOff']);

const secondEndpoint = device.getEndpoint(1);
await secondEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(secondEndpoint, coordinatorEndpoint, ['genOnOff']);

const thirdEndpoint = device.getEndpoint(1);
await thirdEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(thirdEndpoint, coordinatorEndpoint, ['genOnOff']);
}
};

module.exports = definition;
```

rodrigogbs, 2022-12-23:
I am sorry @Koenkk, I forgot to send the products reference...here it is:

>> ZM-TB242-20A_ext_converter.js (this is the real product that had been just launched to themarket)
https://pt.aliexpress.com/item/1005003822349807.html?spm=a2g0o.order_list.order_list_main.4.21efcaa4XGDVOs&gatewayAdapt=glo2bra

>> ZM-L03E-Z-20A_ext_converter.js (this is just a reference, because I am developing a new product with zemismart with 20A socket)
https://pt.aliexpress.com/item/1005003955910335.html?spm=a2g0o.order_list.order_list_main.352.21efcaa4XGDVOs&gatewayAdapt=glo2bra

Koenkk, 2022-12-23:
This part has to be changed:

```js
toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_backlight_mode]),
fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior, fz.tuya_backlight_mode]),
exposes: [
e.switch().withEndpoint('l1'),
e.switch().withEndpoint('l2'),
exposes.enum('power_on_behavior', ea.ALL, ['on', 'off', 'previous']),
exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']),
],
```

to

```js
extend: tuya.extend.switch({backlightMode: true, endpoints: ['l1', 'l2']}),

```

rodrigogbs, 2022-12-23:
Is this code right?

>> ZM-TB242-20A_ext_converter.js
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [
{
modelID: 'TS011F',
manufacturerName: '_TZ3000_6s5dc9lx'
}
],
model: 'TS011F_2_gang',
vendor: 'TuYa',
description: '2 gang wall outlet 20A',
extend: tuya.extend.switch({backlightMode: true, endpoints: ['l1', 'l2']}),
endpoint: () => {
return {'l1': 1, 'l2': 2};
},
meta: {multiEndpoint: true},
configure: async (device, coordinatorEndpoint) => {
const firstEndpoint = device.getEndpoint(1);
await firstEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(firstEndpoint, coordinatorEndpoint, ['genOnOff']);

const secondEndpoint = device.getEndpoint(1);
await secondEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(secondEndpoint, coordinatorEndpoint, ['genOnOff']);
}
};

module.exports = definition;
```

The error with it is:
```
[15:49:45] INFO: Preparing to start...
[15:49:45] INFO: Socat not enabled
[15:49:48] INFO: Starting Zigbee2MQTT...
/app/dist/util/externally-loaded.js:19
extend: tuya.extend.switch({backlightMode: true, endpoints: ['l1', 'l2']}),
        ^
ReferenceError: tuya is not defined
    at /app/dist/util/externally-loaded.js:19:9
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```

rodrigogbs, 2022-12-23:
It worked !!!

Thanks for your attention and effort @Koenkk.

closed 2023-01-31 by github-actions[bot]