Koenkk/zigbee2mqtt #15310: Heiman HS2SK is missing energy consumption parameter

geelenbert, 2022-12-02:
### What happened?

I've switched from deconz to Z2MQTT with great results. But my Heiman HS2SK plugs don't report energy consumption under Zigbee 2 mqtt. 

When i was using Deconz, this paramter was available. 
[image]

### What did you expect to happen?

I expected the device to also have reporting of energy.

Tried to modify my own converter but it does not result un energy metering:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['SmartPlug', 'SmartPlug-EF-3.0'],
        model: 'HS2SK',
        description: 'Smart metering plug',
        vendor: 'HEIMAN',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering],
        toZigbee: [tz.on_off],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement','seMetering']);
            await reporting.onOff(endpoint);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.rmsVoltage(endpoint);
            await reporting.rmsCurrent(endpoint);
            await reporting.activePower(endpoint);
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage(),e.energy()],
    },
];
`

### How to reproduce it (minimal and precise)

Addming device to Zigbee 2 mqtt

### Zigbee2MQTT version

1.28.4-1 

### Adapter firmware version

20220219

### Adapter

Zigstar Lan

geelenbert, 2022-12-02:
In addidtion, i found this device under the zigbee certified devices:

[](https://csa-iot.org/csa_product/smart-metering-plug/
)

And the compliance documentation states the following clusters/attributes are supported:
 `<cluster>
<name>Simple Metering</name>
<code>0x0702</code>
<define>SIMPLE_METERING_CLUSTER</define>
<attribute side="server" code="0x0000" define="CURRENT_SUMMATION_DELIVERED" mandatory="true" name="current summation delivered">TRUE</attribute>
<attribute side="server" code="0x0200" define="STATUS" mandatory="true" name="status">TRUE</attribute>
<attribute side="server" code="0x0300" define="UNIT_OF_MEASURE" mandatory="true" name="unit of measure">TRUE</attribute>
<attribute side="server" code="0x0301" define="MULTIPLIER" mandatory="false" name="multiplier">TRUE</attribute>
<attribute side="server" code="0x0302" define="DIVISOR" mandatory="false" name="divisor">TRUE</attribute>
<attribute side="server" code="0x0303" define="SUMMATION_FORMATTING" mandatory="true" name="summation formatting">TRUE</attribute>
<attribute side="server" code="0x0304" define="DEMAND_FORMATTING" mandatory="false" name="demand formatting">TRUE</attribute>
<attribute side="server" code="0x0306" define="METERING_DEVICE_TYPE" mandatory="true" name="metering device type">TRUE</attribute>
<command receivedBy="client" code="0x00" name="GetProfileResponse" mandatory="false"/>
<command receivedBy="client" code="0x01" name="RequestMirror" mandatory="false"/>
<command receivedBy="client" code="0x02" name="RemoveMirror" mandatory="false"/>
<command receivedBy="client" code="0x03" name="RequestFastPollModeResponse" mandatory="false"/>
<command receivedBy="server" code="0x00" name="GetProfile" mandatory="false">FALSE</command>
<command receivedBy="server" code="0x01" name="RequestMirrorResponse" mandatory="false">FALSE</command>
<command receivedBy="server" code="0x02" name="MirrorRemoved" mandatory="false">FALSE</command>
<command receivedBy="server" code="0x03" name="RequestFastPollMode" mandatory="false">FALSE</command>
</cluster>`

geelenbert, 2022-12-02:
Manualy reading these attributes results in:
[image]

Info 2022-12-02 12:08:50Read result of 'seMetering': {"unitOfMeasure":0,"multiplier":1,"currentSummDelivered":[0,241843],"divisor":10000,"summaFormatting":42,"demandFormatting":44,"meteringDeviceType":0}

geelenbert, 2022-12-28:
I created a custom converter:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['SmartPlug', 'SmartPlug-EF-3.0'],
        model: 'HS2SK',
        description: 'Smart metering plug',
        vendor: 'HEIMAN',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering],
        toZigbee: [tz.on_off],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement','seMetering']);
            await reporting.onOff(endpoint);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.rmsVoltage(endpoint);
            await reporting.rmsCurrent(endpoint);
            await reporting.activePower(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.currentSummDelivered(endpoint);
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage(),e.energy()],
    },
];
`
[image]

This results in the expected behaviour.  
@Koenkk  Suggestion to fix this in the main branch?

Koenkk, 2022-12-28:
Done, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-28 by Koenkk