Koenkk/zigbee2mqtt #15316: Some devices disappeared from list but are still working

Kefut, 2022-12-02:
### What happened?

I have a few devices that dissapeared from the device list but are still visible in the config and are also still working. However as they are not in the list, I cannot do new configurations wih them.
Here you can see it in the RAW data:
[image]
But it cannot be found under devices
[image]

Additionally, you can see 7 devices in the group RAW data:
[image]
But in the list, you can only find 2:
[image]

### What did you expect to happen?

I would expect to see all devices in the list that are also in the config. 

### How to reproduce it (minimal and precise)

I don't know

### Zigbee2MQTT version

1.28.2 commit: e66fe85

### Adapter firmware version

20201127

### Adapter

 Texas Instruments, Inc. CC2531 ZigBee

Koenkk, 2022-12-02:
@nurikk this seems to be an issue in the frontend

Kefut, 2022-12-02:
> @nurikk this seems to be an issue in the frontend

I also connect to the instance through node red and there I can only select the devices that are listed in the device list

Kefut, 2023-01-02:
Any feedback on the issue?

Koenkk, 2023-01-02:
Can you provide the ieee address of these device, your `data/database.db` and your zigbee2mqtt start debug log?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Kefut, 2023-01-05:
> Can you provide the ieee address of these device, your `data/database.db` and your zigbee2mqtt start debug log?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Hi,

files are attached: [Archiv.zip](https://github.com/Koenkk/zigbee2mqtt/files/10351658/Archiv.zip)

The missing devices are (I can find all of them in the raw config):
0xa4c13850a29eaaba
0xa4c13858c5bf3b68
0xa4c1386bec3453fb
0xa4c1388167dcca52
0xa4c138dbbb8d27db

Thanks for your help.

Koenkk, 2023-01-06:
`0xa4c13850a29eaaba` is not in your database.db file. Probably it has left the network. Can you provide the herdsman debug log starting from the point where the device is paired until is disappears? 

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Kefut, 2023-01-06:
> 

Hi,
do you mean that I should try to pair it again? Because the stupid thing for me is that it is actually controlled, but doesn't appear anymore.
The only time it currently appears in the herdsman debug is here:
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Zigbee2MQTT:error 2023-01-06 13:40:36: Cannot find '0xa4c13850a29eaaba/1' of group 'OG2Carl Deckenlich'
Zigbee2MQTT:error 2023-01-06 13:40:36: Cannot find '0xa4c1388167dcca52/1' of group 'OG2Carl Deckenlich'
Zigbee2MQTT:error 2023-01-06 13:40:36: Cannot find '0xa4c13858c5bf3b68/1' of group 'OG2Carl Deckenlich'
Zigbee2MQTT:error 2023-01-06 13:40:37: Cannot find '0xa4c1386bec3453fb/1' of group 'OG2Carl Deckenlich'
Zigbee2MQTT:error 2023-01-06 13:40:37: Cannot find '0xa4c138dbbb8d27db/1' of group 'OG2Carl Deckenlich'

Thanks,
Christoph

Koenkk, 2023-01-06:
Yes, pair them again.

closed 2023-02-13 by github-actions[bot]