Koenkk/zigbee2mqtt #15319: TS0601 Tuya TRV _TZE200_bvu2wnxz

Althir1, 2022-12-02:
### Link

https://de.aliexpress.com/item/1005003903096003.html?spm=a2g0o.productlist.main.43.558e24d8fI1W4y&algo_pvid=648328d8-8b1c-4c95-8e47-eb4b54dfd04f&algo_exp_id=648328d8-8b1c-4c95-8e47-eb4b54dfd04f-21&pdp_ext_f=%7B%22sku_id%22%3A%2212000027424321645%22%7D&pdp_npi=2%40dis%21EUR%2145.59%2129.63%21%21%21%21%21%402100b69816699087058582432d074b%2112000027424321645%21sea&curPageLogUid=lSa5tpSMSnsf

### Database entry

{"id":46,"type":"EndDevice","ieeeAddr":"0x003c84fffe267e2c","nwkAddr":16019,"manufId":4098,"manufName":"_TZE200_bvu2wnxz","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000f\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_bvu2wnxz","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1669908328472,"defaultSendRequestWhen":"immediate"}

### Comments

I'm running Zigbee2Mqtt Edge latest version and get the followin error:
2022-12-01 16:25:28Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_bvu2wnxz'

closed 2022-12-02 by Koenkk

Althir1, 2022-12-02:
Seems to be the wrong device. It's not the TV02-Zigbee model,
it's the SEA802-Zigbee one instead.
https://www.zigbee2mqtt.io/devices/SEA801-Zigbee_SEA802-Zigbee.html

Koenkk, 2022-12-03:
It doesn't seem to work with that converter: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4936

Althir1, 2022-12-03:
I got it working with this converter: https://github.com/twhittock/avatto_me167

woody4165, 2022-12-09:
Hi
I got it working with me167, but when I set system_mode to heat and set a temperature, it always returns to auto and changed the temperature sat before.
Any suggestion?
Thanks