Koenkk/zigbee2mqtt #15322: stopped working after nodejs / npm upgrade

dwindey1, 2022-12-02:
### What happened?

For upgrade Node-red an upgrade to nodejs 16 and npm needed to be done.
Node-red was doing fine after the update, but Zigbee2mqtt stopped working.

I tried to re-install, but with no luck.  And now I can't turn off the lights anymore...

### What did you expect to happen?

How can I make it run again

### How to reproduce it (minimal and precise)

`npm ci`

### Zigbee2MQTT version

1.28

### Adapter firmware version

no idea

### Adapter

sonoff

### Debug log

`

pi@theone:/opt/zigbee2mqtt $ npm ci
npm ERR! code 1
npm ERR! path /opt/zigbee2mqtt/node_modules/@serialport/bindings
npm ERR! command failed
npm ERR! command sh -c -- prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! prebuild-install warn install No prebuilt binaries found (target=18.12.1 runtime=node arch=arm libc= platform=linux)
npm ERR! node:internal/modules/cjs/loader:998
npm ERR!   throw err;
npm ERR!   ^
npm ERR!
npm ERR! Error: Cannot find module 'object-assign'
npm ERR! Require stack:
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/render-template.js
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/plumbing.js
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/index.js
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/npmlog/log.js
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/lib/node-gyp.js
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js
npm ERR!     at Module._resolveFilename (node:internal/modules/cjs/loader:995:15)
npm ERR!     at Module._load (node:internal/modules/cjs/loader:841:27)
npm ERR!     at Module.require (node:internal/modules/cjs/loader:1061:19)
npm ERR!     at require (node:internal/modules/cjs/helpers:103:18)
npm ERR!     at Object.<anonymous> (/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/render-template.js:4:20)
npm ERR!     at Module._compile (node:internal/modules/cjs/loader:1159:14)
npm ERR!     at Module._extensions..js (node:internal/modules/cjs/loader:1213:10)
npm ERR!     at Module.load (node:internal/modules/cjs/loader:1037:32)
npm ERR!     at Module._load (node:internal/modules/cjs/loader:878:12)
npm ERR!     at Module.require (node:internal/modules/cjs/loader:1061:19) {
npm ERR!   code: 'MODULE_NOT_FOUND',
npm ERR!   requireStack: [
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/render-template.js',
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/plumbing.js',
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/index.js',
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/npmlog/log.js',
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/lib/node-gyp.js',
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js'
npm ERR!   ]
npm ERR! }
npm ERR!
npm ERR! Node.js v18.12.1

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-12-02T18_33_36_443Z-debug-0.log
pi@theone:/opt/zigbee2mqtt $ rm -rf node_modules && npm install
npm ERR! code 1
npm ERR! path /opt/zigbee2mqtt/node_modules/@serialport/bindings
npm ERR! command failed
npm ERR! command sh -c -- prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! prebuild-install warn install No prebuilt binaries found (target=18.12.1 runtime=node arch=arm libc= platform=linux)
npm ERR! node:internal/modules/cjs/loader:998
npm ERR!   throw err;
npm ERR!   ^
npm ERR!
npm ERR! Error: Cannot find module 'object-assign'
npm ERR! Require stack:
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/render-template.js
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/plumbing.js
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/index.js
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/npmlog/log.js
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/lib/node-gyp.js
npm ERR! - /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js
npm ERR!     at Module._resolveFilename (node:internal/modules/cjs/loader:995:15)
npm ERR!     at Module._load (node:internal/modules/cjs/loader:841:27)
npm ERR!     at Module.require (node:internal/modules/cjs/loader:1061:19)
npm ERR!     at require (node:internal/modules/cjs/helpers:103:18)
npm ERR!     at Object.<anonymous> (/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/render-template.js:4:20)
npm ERR!     at Module._compile (node:internal/modules/cjs/loader:1159:14)
npm ERR!     at Module._extensions..js (node:internal/modules/cjs/loader:1213:10)
npm ERR!     at Module.load (node:internal/modules/cjs/loader:1037:32)
npm ERR!     at Module._load (node:internal/modules/cjs/loader:878:12)
npm ERR!     at Module.require (node:internal/modules/cjs/loader:1061:19) {
npm ERR!   code: 'MODULE_NOT_FOUND',
npm ERR!   requireStack: [
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/render-template.js',
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/plumbing.js',
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/gauge/index.js',
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/npmlog/log.js',
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/lib/node-gyp.js',
npm ERR!     '/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js'
npm ERR!   ]
npm ERR! }
npm ERR!
npm ERR! Node.js v18.12.1

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-12-02T18_39_38_448Z-debug-0.log

`

dwindey1, 2022-12-02:
I downgrade to npm 6.x .  This error is now gone.
There are some other issue now, but probably unrelated to this.

lux73, 2022-12-03:
nodeJS up to v19 is supported -> https://github.com/Koenkk/zigbee2mqtt/issues/14833

even npm up to 8.x is supported too

z2m has some issues with npm 9.x as you can see here -> https://github.com/Koenkk/zigbee2mqtt/issues/15172

closed 2023-01-10 by github-actions[bot]

lux73, 2023-02-05:
i can confirm that npm 9.x works without Problem with at least z2m v1.30.0

today i have **upgraded npm from 8.19.3 to 9.4.1** - works like a charm