Koenkk/zigbee2mqtt #15323: [Feature request]: Complete Unocupied features to Sinope TH1123ZB-G2

MattL0, 2022-12-02:
### Is your feature request related to a problem? Please describe

-

### Describe the solution you'd like

Testing

### Describe alternatives you've considered

Test

### Additional context

testing

rigorm, 2022-12-02:
@MattL0 :
Tried it and getting this error message :
[image]
when clicking on the refresh icon

Getting this error when clicking on "occupied" or "unoccupied"

[image]

its a start :)

MattL0, 2022-12-02:
can you try this ?

```js
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    sinope_thermostat: {
        cluster: 'hvacThermostat',
        type: ['readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            const cycleOutputLookup = {15: '15_sec', 300: '5_min', 600: '10_min',
                900: '15_min', 1200: '20_min', 1800: '30_min', 65535: 'off'};
            if (msg.data.hasOwnProperty('1024')) {
                const lookup = {0: 'unoccupied', 1: 'occupied'};
                result.thermostat_occupancy = lookup[msg.data['1024']];
            }
            if (msg.data.hasOwnProperty('1025')) {
                result.main_cycle_output = cycleOutputLookup[msg.data['1025']];
            }
            if (msg.data.hasOwnProperty('1026')) {
                const lookup = {0: 'on_demand', 1: 'sensing'};
                result.backlight_auto_dim = lookup[msg.data['1026']];
            }
            if (msg.data.hasOwnProperty('1028')) {
                result.aux_cycle_output = cycleOutputLookup[msg.data['1028']];
            }
            return result;
        },
    }
};

module.exports = [
    {
        zigbeeModel: ['TH1123ZB-G2'],
        model: 'TH1123ZB-G2',
        vendor: 'Sinopé',
        description: 'Zigbee line volt thermostat',
        meta: {thermostat: {dontMapPIHeatingDemand: true}},
        fromZigbee: [fz.legacy.sinope_thermostat_att_report, fz.legacy.hvac_user_interface, fz.electrical_measurement, fz.metering,
            fz.ignore_temperature_report, fz.legacy.sinope_thermostat_state, fzLocal.sinope_thermostat],
        toZigbee: [tz.thermostat_local_temperature, tz.thermostat_occupied_heating_setpoint, tz.thermostat_temperature_display_mode,
            tz.thermostat_keypad_lockout, tz.thermostat_system_mode, tz.thermostat_running_state, tz.sinope_thermostat_backlight_autodim_param, 
            tz.sinope_thermostat_time, tz.sinope_thermostat_enable_outdoor_temperature, tz.sinope_thermostat_outdoor_temperature,
            tz.thermostat_pi_heating_demand, tz.acvoltage, tz.sinope_time_format],
        exposes: [
            exposes.climate()
                .withSetpoint('occupied_heating_setpoint', 5, 30, 0.5)
                .withSetpoint('unoccupied_heating_setpoint', 5, 30, 0.5)
                .withLocalTemperature()
                .withSystemMode(['off', 'heat'], ea.ALL, 'Mode of the thermostat')
                .withPiHeatingDemand(ea.STATE_GET)
                .withRunningState(['idle', 'heat']),
            exposes.enum('thermostat_occupancy', ea.ALL, ['unoccupied', 'occupied'])
                .withDescription('Occupancy state of the thermostat'),
            exposes.enum('backlight_auto_dim', ea.ALL, ['on demand', 'sensing'])
                .withDescription('Control backlight dimming behavior'),
            exposes.enum('keypad_lockout', ea.ALL, ['unlock', 'lock1'])
                .withDescription('Enables or disables the device’s buttons'),
            e.power(), e.current(), e.voltage().withAccess(ea.STATE_GET), e.energy(),
        ],

        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const binds = [
                'genBasic', 'genIdentify', 'genGroups', 'hvacThermostat', 'hvacUserInterfaceCfg',
                'msTemperatureMeasurement', 'haElectricalMeasurement', 'seMetering',
                'manuSpecificSinope'];
            await reporting.bind(endpoint, coordinatorEndpoint, binds);
            await reporting.thermostatTemperature(endpoint);
            await reporting.thermostatPIHeatingDemand(endpoint);
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
            try {
                await reporting.thermostatUnoccupiedHeatingSetpoint(endpoint);
            } catch (error) {/* Not all support this */}

            try {
                await reporting.thermostatSystemMode(endpoint);
            } catch (error) {/* Not all support this */}
            try {
                await reporting.thermostatRunningState(endpoint);
            } catch (error) {/* Not all support this */}

            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.currentSummDelivered(endpoint, {min: 10, max: 303, change: [1, 1]});

            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            try {
                await reporting.activePower(endpoint, {min: 10, max: 305, change: 1});
            } catch (error) {/* Do nothing*/}
            try {
                await reporting.rmsCurrent(endpoint, {min: 10, max: 306, change: 100}); // divider 1000: 0.1Arms
            } catch (error) {/* Do nothing*/}
            try {
                await reporting.rmsVoltage(endpoint, {min: 10, max: 307, change: 5}); // divider 10: 0.5Vrms
            } catch (error) {/* Do nothing*/}

            await reporting.temperature(endpoint, {min: 1, max: 0xFFFF}); // Disable default reporting
        },
    }
];
```

rigorm, 2022-12-02:
> can you try this ? I forgot to put tz.sinope_thermostat_occupancy ...
> 
> ```js
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const constants = require('zigbee-herdsman-converters/lib/constants');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const fzLocal = {
>     sinope_thermostat: {
>         cluster: 'hvacThermostat',
>         type: ['readResponse'],
>         convert: (model, msg, publish, options, meta) => {
>             const result = {};
>             const cycleOutputLookup = {15: '15_sec', 300: '5_min', 600: '10_min',
>                 900: '15_min', 1200: '20_min', 1800: '30_min', 65535: 'off'};
>             if (msg.data.hasOwnProperty('1024')) {
>                 const lookup = {0: 'unoccupied', 1: 'occupied'};
>                 result.thermostat_occupancy = lookup[msg.data['1024']];
>             }
>             if (msg.data.hasOwnProperty('1025')) {
>                 result.main_cycle_output = cycleOutputLookup[msg.data['1025']];
>             }
>             if (msg.data.hasOwnProperty('1026')) {
>                 const lookup = {0: 'on_demand', 1: 'sensing'};
>                 result.backlight_auto_dim = lookup[msg.data['1026']];
>             }
>             if (msg.data.hasOwnProperty('1028')) {
>                 result.aux_cycle_output = cycleOutputLookup[msg.data['1028']];
>             }
>             return result;
>         },
>     }
> };
> 
> module.exports = [
>     {
>         zigbeeModel: ['TH1123ZB-G2'],
>         model: 'TH1123ZB-G2',
>         vendor: 'Sinopé',
>         description: 'Zigbee line volt thermostat',
>         meta: {thermostat: {dontMapPIHeatingDemand: true}},
>         fromZigbee: [fz.legacy.sinope_thermostat_att_report, fz.legacy.hvac_user_interface, fz.electrical_measurement, fz.metering,
>             fz.ignore_temperature_report, fz.legacy.sinope_thermostat_state, fzLocal.sinope_thermostat],
>         toZigbee: [tz.thermostat_local_temperature, tz.thermostat_occupied_heating_setpoint, tz.thermostat_temperature_display_mode,
>             tz.thermostat_keypad_lockout, tz.thermostat_system_mode, tz.thermostat_running_state, tz.sinope_thermostat_backlight_autodim_param, 
>             tz.sinope_thermostat_time, tz.sinope_thermostat_enable_outdoor_temperature, tz.sinope_thermostat_outdoor_temperature,
>             tz.thermostat_pi_heating_demand, tz.acvoltage, tz.sinope_time_format, tz.sinope_thermostat_occupancy],
>         exposes: [
>             exposes.climate()
>                 .withSetpoint('occupied_heating_setpoint', 5, 30, 0.5)
>                 .withSetpoint('unoccupied_heating_setpoint', 5, 30, 0.5)
>                 .withLocalTemperature()
>                 .withSystemMode(['off', 'heat'], ea.ALL, 'Mode of the thermostat')
>                 .withPiHeatingDemand(ea.STATE_GET)
>                 .withRunningState(['idle', 'heat']),
>             exposes.enum('thermostat_occupancy', ea.ALL, ['unoccupied', 'occupied'])
>                 .withDescription('Occupancy state of the thermostat'),
>             exposes.enum('backlight_auto_dim', ea.ALL, ['on demand', 'sensing'])
>                 .withDescription('Control backlight dimming behavior'),
>             exposes.enum('keypad_lockout', ea.ALL, ['unlock', 'lock1'])
>                 .withDescription('Enables or disables the device’s buttons'),
>             e.power(), e.current(), e.voltage().withAccess(ea.STATE_GET), e.energy(),
>         ],
> 
>         configure: async (device, coordinatorEndpoint, logger) => {
[... 18 lines cut ...]
>             } catch (error) {/* Not all support this */}
> 
>             await reporting.readMeteringMultiplierDivisor(endpoint);
>             await reporting.currentSummDelivered(endpoint, {min: 10, max: 303, change: [1, 1]});
> 
>             await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
>             try {
>                 await reporting.activePower(endpoint, {min: 10, max: 305, change: 1});
>             } catch (error) {/* Do nothing*/}
>             try {
>                 await reporting.rmsCurrent(endpoint, {min: 10, max: 306, change: 100}); // divider 1000: 0.1Arms
>             } catch (error) {/* Do nothing*/}
>             try {
>                 await reporting.rmsVoltage(endpoint, {min: 10, max: 307, change: 5}); // divider 10: 0.5Vrms
>             } catch (error) {/* Do nothing*/}
> 
>             await reporting.temperature(endpoint, {min: 1, max: 0xFFFF}); // Disable default reporting
>         },
>     }
> ];
> ```

Getting the same errors.. as if it was not seeing the fzLocal method

FYI: before your changes above, I tried adding tz.sinope_thermostat_occupancy to the "toZigbee" list and it gave me a UNKNOWN_ATTRIBUTE when clicking either "occupied" or "unoccupied"

MattL0, 2022-12-02:
Oh … not again…Let me check so i give you an mqtt command to try

rigorm, 2022-12-02:
this is what I got : 
[image]

rigorm, 2022-12-02:
weirdly, I also tried on first gen in my office and got nothing back.. but occupency is working fine from Exposes tab or HA.

MattL0, 2022-12-02:
Not good. 

Can you send this as a mqtt message ? ( bugged in dev console). This attribute can't be written but i am curious. 

topic : zigbee2mqtt/thermostat_etage_salle-de-bain/1/set

message: 
{"read":{"cluster":"hvacThermostat", "attributes":["occupancy]}}

rigorm, 2022-12-02:
this show up in the Z2M logs: 

[image]

MattL0, 2022-12-02:
ok. So this wont work :S Seems like Sinope removed/changed/bugged all theird custum attributes... on the g2

rigorm, 2022-12-02:
to bad.. 
there must be a way to snif out the cluster and attribute addr of some sort ?
Just like the auto_dim works as it's in the neviweb interface so there must be cluster/attribute for it.

MattL0, 2022-12-02:
I think nobody have tried to sniff the cluster/attribute from sinope. All was available from various source/forum ( ex. smarthings ) on the web. 

I used deconz before. what thing that was nice is that they show all the attribute available form the ui . 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2104

I don't know if there is a method like this for z2m ?

MattL0, 2022-12-02:
> weirdly, I also tried on first gen in my office and got nothing back.. but occupency is working fine from Exposes tab or HA.

Could you try to send theses mqtt messages and send me the log output of z2m?  :

```
  -topic 
   zigbee2mqtt/yourG1Sinope/set
 -message 
 {"thermostat_occupancy": "occupied"}
```

```
-topic 
zigbee2mqtt/yourG1Sinope/get
 -message
 {"thermostat_occupancy": ""}

```

rigorm, 2022-12-02:
First request:

```
Info 2022-12-02 18:36:41MQTT publish: topic 'zigbee2mqtt/thermostat_soussol_bureau', payload '{"backlight_auto_dim":"sensing","current":8.6,"device":{"applicationVersion":68,"dateCode":"20210903\u0000\u0000\u0000\u0000\u0000\u0000\u0000","friendlyName":"thermostat_soussol_bureau","hardwareVersion":2,"ieeeAddr":"0x500b914000029683","manufacturerID":4508,"manufacturerName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"TH1123ZB","networkAddress":9639,"powerSource":"Mains (single phase)","softwareBuildID":"2231\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","stackVersion":1,"type":"Router","zclVersion":2},"energy":1051.13,"keypad_lockout":null,"last_seen":"2022-12-02T23:36:41.418Z","linkquality":83,"local_temperature":19.02,"occupied_heating_setpoint":21,"operation":"heating","pi_heating_demand":100,"power":2073,"running_state":"heat","system_mode":"heat","thermostat_occupancy":"unoccupied","unoccupied_heating_setpoint":17,"voltage":241.1}'
Info 2022-12-02 18:36:41MQTT publish: topic 'zigbee2mqtt/thermostat_soussol_bureau/auto_dim', payload '{"backlight":"sensing"}'
Info 2022-12-02 18:36:41MQTT publish: topic 'zigbee2mqtt/thermostat_soussol_bureau', payload '{"backlight_auto_dim":"sensing","current":8.6,"device":{"applicationVersion":68,"dateCode":"20210903\u0000\u0000\u0000\u0000\u0000\u0000\u0000","friendlyName":"thermostat_soussol_bureau","hardwareVersion":2,"ieeeAddr":"0x500b914000029683","manufacturerID":4508,"manufacturerName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"TH1123ZB","networkAddress":9639,"powerSource":"Mains (single phase)","softwareBuildID":"2231\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","stackVersion":1,"type":"Router","zclVersion":2},"energy":1051.13,"keypad_lockout":null,"last_seen":"2022-12-02T23:36:41.418Z","linkquality":83,"local_temperature":19.02,"occupied_heating_setpoint":21,"operation":"heating","pi_heating_demand":100,"power":2073,"running_state":"heat","system_mode":"heat","thermostat_occupancy":"occupied","unoccupied_heating_setpoint":17,"voltage":241.1}'
Info 2022-12-02 18:36:41MQTT publish: topic 'zigbee2mqtt/thermostat_soussol_bureau/auto_dim', payload '{"backlight":"sensing"}'
```

Second request:

```
Info 2022-12-02 18:37:54MQTT publish: topic 'zigbee2mqtt/thermostat_soussol_bureau', payload '{"backlight_auto_dim":"sensing","current":0,"device":{"applicationVersion":68,"dateCode":"20210903\u0000\u0000\u0000\u0000\u0000\u0000\u0000","friendlyName":"thermostat_soussol_bureau","hardwareVersion":2,"ieeeAddr":"0x500b914000029683","manufacturerID":4508,"manufacturerName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"TH1123ZB","networkAddress":9639,"powerSource":"Mains (single phase)","softwareBuildID":"2231\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","stackVersion":1,"type":"Router","zclVersion":2},"energy":1051.13,"keypad_lockout":null,"last_seen":"2022-12-02T23:37:54.806Z","linkquality":80,"local_temperature":19.4,"occupied_heating_setpoint":21,"operation":"idle","pi_heating_demand":0,"power":0,"running_state":"idle","system_mode":"heat","thermostat_occupancy":"","unoccupied_heating_setpoint":17,"voltage":241.1}'
Info 2022-12-02 18:37:54MQTT publish: topic 'zigbee2mqtt/thermostat_soussol_bureau/auto_dim', payload '{"backlight":"sensing"}'
Info 2022-12-02 18:37:54MQTT publish: topic 'zigbee2mqtt/thermostat_soussol_bureau', payload '{"backlight_auto_dim":"sensing","current":0,"device":{"applicationVersion":68,"dateCode":"20210903\u0000\u0000\u0000\u0000\u0000\u0000\u0000","friendlyName":"thermostat_soussol_bureau","hardwareVersion":2,"ieeeAddr":"0x500b914000029683","manufacturerID":4508,"manufacturerName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"TH1123ZB","networkAddress":9639,"powerSource":"Mains (single phase)","softwareBuildID":"2231\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","stackVersion":1,"type":"Router","zclVersion":2},"energy":1051.13,"keypad_lockout":null,"last_seen":"2022-12-02T23:37:54.806Z","linkquality":80,"local_temperature":19.4,"occupied_heating_setpoint":21,"operation":"idle","pi_heating_demand":0,"power":0,"running_state":"idle","system_mode":"heat","thermostat_occupancy":"","unoccupied_heating_setpoint":17,"voltage":241.1}'
Info 2022-12-02 18:37:54MQTT publish: topic 'zigbee2mqtt/thermostat_soussol_bureau/auto_dim', payload '{"backlight":"sensing"}'
```

closed 2022-12-03 by MattL0