Koenkk/zigbee2mqtt #15324: OTA check for Osram plug raising error after upgrade to 1.28.4

satter, 2022-12-02:
### What happened?

After upgrading z2m from 1.28.2 to 1.28.4 today I noticed that OTA check for Osram plugs raising errors

### What did you expect to happen?

Device has up to date firmware so this check should return 'no update available'
```
Apr 20 19:37:00 homie1 npm[10191]: Zigbee2MQTT:debug 2022-04-20 19:37:00: Is new image available for '0x7cb03eaa00ae5ff1', current '{"fieldControl":0,"manufacturerCode":4364,"imageType":39,"fileVersion":>
Apr 20 19:37:00 homie1 npm[10191]: Zigbee2MQTT:debug 2022-04-20 19:37:00: Update available for '0x7cb03eaa00ae5ff1': NO
```

### How to reproduce it (minimal and precise)

I can reproduce it with every 'Check for new updates' button click

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle

### Debug log

```
Dec 02 19:46:27 homie1 npm[3001]: Zigbee2MQTT:debug 2022-12-02 19:46:27: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"osram_plug1","transaction":"tjrjr-3"}'
Dec 02 19:46:27 homie1 npm[3001]: Zigbee2MQTT:info  2022-12-02 19:46:27: Checking if update available for 'osram_plug1'
Dec 02 19:46:27 homie1 npm[3001]: Zigbee2MQTT:debug 2022-12-02 19:46:27: Check if update available for '0x7cb03eaa00ae5ff1' (Plug 01)
Dec 02 19:46:27 homie1 npm[3001]: Zigbee2MQTT:debug 2022-12-02 19:46:27: Using endpoint '3'
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.913Z zigbee-herdsman:controller:endpoint CommandResponse 0x7cb03eaa00ae5ff1/3 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.913Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x7cb03eaa00ae5ff1:3709/3 (0,0,1)
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.913Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":3709,"destendpoint":3,"srcendpoint":1,"clusterid":25,"transid":3,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[25,4,0,0,100]}}
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.913Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,125,14,3,1,25,0,3,0,30,5,25,4,0,0,100,35]
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.930Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.930Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.930Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.930Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.930Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.935Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,3,197]
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.935Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,3,197]
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.935Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,3] - 197
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.935Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":3}
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.935Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.956Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,125,14,0,241]
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.956Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,125,14,0,241]
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.956Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [125,14,0] - 241
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.956Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":3709,"relaycount":0,"relaylist":[]}
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.956Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.984Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,125,14,3,1,0,69,0,110,197,163,0,0,12,17,142,1,0,12,17,39,0,9,5,2,1,125,14,29,9]
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.984Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,125,14,3,1,0,69,0,110,197,163,0,0,12,17,142,1,0,12,17,39,0,9,5,2,1,125,14,29,9]
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.984Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,125,14,3,1,0,69,0,110,197,163,0,0,12,17,142,1,0,12,17,39,0,9,5,2,1,125,14,29] - 9
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.984Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":3709,"srcendpoint":3,"dstendpoint":1,"wasbroadcast":0,"linkquality":69,"securityuse":0,"timestamp":10732910,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[17,142,1,0,12,17,39,0,9,5,2,1]}}
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.987Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":142,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4364,"imageType":39,"fileVersion":16909577},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":3709,"endpoint":3,"linkquality":69,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Dec 02 19:46:27 homie1 npm[3001]: Zigbee2MQTT:debug 2022-12-02 19:46:27: Received Zigbee message from 'osram_plug1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16909577,"imageType":39,"manufacturerCode":4364}' from endpoint 3 with groupID 0
Dec 02 19:46:27 homie1 npm[3001]: Zigbee2MQTT:info  2022-12-02 19:46:27: MQTT publish: topic 'zigbee2mqtt/osram_plug1', payload '{"last_seen":"2022-12-02T19:46:27+00:00","linkquality":69,"state":"OFF","update":{"state":"idle"}}'
Dec 02 19:46:27 homie1 npm[3001]: 2022-12-02T19:46:27.995Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Dec 02 19:46:27 homie1 npm[3001]: Zigbee2MQTT:debug 2022-12-02 19:46:27: Got OTA request '{"fieldControl":0,"manufacturerCode":4364,"imageType":39,"fileVersion":16909577}'
Dec 02 19:46:28 homie1 npm[3001]: Zigbee2MQTT:info  2022-12-02 19:46:28: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"osram_plug1"},"error":"Failed to check if update available for 'osram_plug1' (No image available for manufacturerCode '4364' imageType '39')","status":"error","transaction":"tjrjr-3"}'
Dec 02 19:46:28 homie1 npm[3001]: Zigbee2MQTT:error 2022-12-02 19:46:28: Failed to check if update available for 'osram_plug1' (No image available for manufacturerCode '4364' imageType '39')
Dec 02 19:46:28 homie1 npm[3001]: Zigbee2MQTT:debug 2022-12-02 19:46:28: AssertionError [ERR_ASSERTION]: No image available for manufacturerCode '4364' imageType '39'
Dec 02 19:46:28 homie1 npm[3001]:     at getImageMeta (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/ledvance.js:17:5)
Dec 02 19:46:28 homie1 npm[3001]:     at processTicksAndRejections (node:internal/process/task_queues:96:5)
Dec 02 19:46:28 homie1 npm[3001]:     at isNewImageAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:265:18)
Dec 02 19:46:28 homie1 npm[3001]:     at Object.isUpdateAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:256:23)
Dec 02 19:46:28 homie1 npm[3001]:     at OTAUpdate.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/otaUpdate.ts:192:39)
```

ecov773, 2022-12-08:
After upgrading I also have the same problem but with "No image available for manufacturerCode '4364' imageType '19'". This OSRAM bulb https://www.zigbee2mqtt.io/devices/AC03641.html#osram-ac03641

Does someone know what changed?

sotima, 2022-12-08:
Same here, fun fact: Homeassistant informs me correctly, that one OSRAM plug is out of date. But as soon as I try to do an OTA I get the same error "No image available for manufacturerCode '4364' imageType '39'".

ahoeh, 2022-12-13:
if updating your OSRAM bulbs is your priority then here is how i solved this temporarily by going back to `1.28.2`

* first stop your npm and save your personal data folder!
* then open a git bash or similar
* `cd` into zigbee2mqtt
* `git fetch --tags --all`
* `git checkout tags/1.28.2` ==> message: `HEAD is now at 360a777a 1.28.2`
* `npm install`
* eventually restore your personal data folder
* `npm start`
* happy firmware-updating without these errors

satter, 2023-01-03:
I cannot reproduce this error anymore after z2m upgrade to version 1.29.0

sotima, 2023-01-03:
> I cannot reproduce this error anymore after z2m upgrade to version 1.29.0

Same here! Seems to be fixed in 1.29.0! I could update my Osram plug via OTA without problems! Thanks a lot!

closed 2023-01-04 by Koenkk