Koenkk/zigbee2mqtt #15328: Last update broke Lixee communication (OTA error: TypeError: releasesLIST.sort is not a function or its return value is not iterable)

JeanMars, 2022-12-02:
### What happened?

Energy consumption no longer showing in HA after container image updated to latest

### What did you expect to happen?

Energy consumtion still reported in HA after Zigbee2mqtt update

### How to reproduce it (minimal and precise)

Update was performed through Watchtower coontainer:
time="2022-12-02T17:28:49Z" level=info msg="Found new docker.io/koenkk/zigbee2mqtt:latest image (e6bf13d00050)"
time="2022-12-02T17:29:08Z" level=info msg="Stopping /Zigbee2MQTT (d6699f8b80ed) with SIGTERM"
time="2022-12-02T17:29:09Z" level=info msg="Creating /Zigbee2MQTT"
time="2022-12-02T17:29:10Z" level=info msg="Removing image bf4576703bf7"
time="2022-12-02T17:29:14Z" level=info msg="Session done" Failed=0 Scanned=7 Updated=1 notify=no
When I go to "OTA"-->Check for new updates of Lixee device, I can see the same error "TypeError: releasesLIST.sort is not..."
Using MQTT Explorer, I can see that consumptions are still sent by Lixee device
According to Zigbee2mqtt logs, they are received by Zigbee2mqtt, no idea why HA does not show them; looks like the OTA error prevents this but not sure.

### Zigbee2MQTT version

1.28.4 commit: 52e545f

### Adapter firmware version

6.10.0.0 build 169

### Adapter

Zigbee GPIO ZB-Shield

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10144895/log.txt)

JeanMars, 2022-12-03:
Hi again,
just to mention that back to version 1.28.2, this error is not present; it is in 1.28.3 and 1.28.4
Hope that helps,
Jean

JeanMars, 2022-12-03:
Update: communication broken (consumption no longer reported) seems to be a consequence (not even sure it's on zigbee2mqtt or HA); I needed to pair again the Lixee to restore proper functionning.
Anyway, the original error (TypeError: releasesLIST.sort is not a function or its return value is not iterable) is caused by version 1.28.3 and 1.28.4

SilentT-FR, 2022-12-04:
You need to open issue here : https://github.com/Koenkk/zigbee-herdsman-converters/issues/new

i thinks its about : https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/ota/lixee.js#L13

@arenoux & @vk496

JeanMars, 2022-12-04:
Thanks,
I have open the following:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/5082

Koenkk, 2022-12-06:
Probably already fixed in the latest-dev, can you check?  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

JeanMars, 2022-12-06:
Hi,
switching to latest-dev ends up with continuous restart of container, see attached log.
[_Zigbee2MQTT_logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/10170342/_Zigbee2MQTT_logs.txt)
Let me know if I can help more.

Koenkk, 2022-12-07:
I think this has just been fixed by https://github.com/Koenkk/zigbee2mqtt/pull/15400

JeanMars, 2022-12-07:
Just tested latest-dev docker release, back to regular behavior :-)
Thanks, this has been fixed.

closed 2022-12-08 by Koenkk