Koenkk/zigbee2mqtt #15331: [New device support]:

daizusikatabenai, 2022-12-03:
### Link

https://www.philips-hue.com/ja-jp/p/hue-white-b39---e17-smart-bulb----2-pack-/8719514320642

### Database entry

{"id":18,"type":"Router","ieeeAddr":"0x001788010d5c73d9","nwkAddr":46937,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LWE008","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWE008","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20201012","swBuildId":"1.76.10"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20201012","swBuildId":"1.76.10","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1670056088274,"defaultSendRequestWhen":"immediate"}

### Comments

Device join but not supported
Please help 
Thanks

Zigbee2MQTT 
Current version: 1.28.4-1
Mosquitto broker 
Current version: 6.1.3

Warning 2022-12-03 14:21:30Received message from unsupported device with Zigbee model 'LWE008' and manufacturer name 'Signify Netherlands B.V.'
Warning 2022-12-03 14:21:30Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### Supported color modes

light_onoff.brightness

Koenkk, 2022-12-03:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

daizusikatabenai, 2022-12-03:
thanks for your prompt reply
I tried but,brightness only?

Error 2022-12-03 18:59:22Publish 'set' 'read' to '0x001788010d5c73d9' failed: 'Error: Read 0x001788010d5c73d9/11 lightingColorCtrl(["colorTemperature","startUpColorTemperature","colorTempPhysicalMin"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 46937 - 11 - 151 - 768 - 1 after 10000ms)'

Maybe... colorTemp is no function
I'll look into it

daizusikatabenai, 2022-12-03:
[image]

see you tomorrow
I can see the direction

daizusikatabenai, 2022-12-03:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {    {
        zigbeeModel: ['LWE008'],
        model: '‎929003021302',
        vendor: 'Philips',
        description: 'Hue Smart Light E17 470 lumen',
        extend: hueExtend.light_onoff_brightness(),
};

module.exports = definition;

daizusikatabenai, 2022-12-04:
It seems highly skilled to me

closed 2022-12-04 by daizusikatabenai