Koenkk/zigbee2mqtt #15333: TuYa Power monitoring sockets _TZ3000_gvn91tmx and _TZ3000_cehuw1lw report 0W power consumption  once in a while

kecksk0enig, 2022-12-03:
### What happened?

I have multiple sockets of type  TS011F_plug_3 and TS011F_plug_1

the ones requirng polling to get the power measurements work flawless.
however the other sockets which push the power measurements are a bit wonky.
they seem to report  the power consumption finde, however in regular intervalls they will report 0W/0A even when the load is not changed. ( see debug log) the next updated values are then normal (and correct) again.

This results in very ugly graphs in HA since the graph contains regular drops to 0W( vor one reporting interval)

I assume this is a bug within the firmware of the affected device and thus not realy fixable by zigbe2mqtt.
However maybe there is a workaround to  not publish these single 0W reports within a row of correct measurements.
any hints where i could find a solution to this or should i dump the affected sockets altogether?
Is there a reliable source of powermeasuring sockets wher it is clear which revision of sockets one gets?

thank you in advance

flawlass working sockets (by polling):
_TZ3000_ew3ldmgx
 affected sockets:
_TZ3000_gvn91tmx
[image]

### What did you expect to happen?

Correct rporting of power consumption

### How to reproduce it (minimal and precise)

plug constant load to socket , observe single reports of 0W measurements within a row ofcorrect measurements once in a while

### Zigbee2MQTT version

 1.28.4 commit: 52e545f

### Adapter firmware version

20210708

### Adapter

sONOFF ZIGBEE 3.0 usb dONGLE pLUS

### Debug log

`Info 2022-12-03 11:30:48MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":4.18,"energy":467.15,"indicator_mode":"off/on","linkquality":33,"power":962,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'
Info 2022-12-03 11:30:48MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":4.18,"energy":467.15,"indicator_mode":"off/on","linkquality":33,"power":973,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'
Info 2022-12-03 11:30:49MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":4.18,"energy":467.16,"indicator_mode":"off/on","linkquality":30,"power":973,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'
Info 2022-12-03 11:30:57MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":0,"energy":467.16,"indicator_mode":"off/on","linkquality":33,"power":973,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'
Info 2022-12-03 11:30:57MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":0,"energy":467.16,"indicator_mode":"off/on","linkquality":27,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'
Info 2022-12-03 11:31:02MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":4.22,"energy":467.16,"indicator_mode":"off/on","linkquality":27,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'
Info 2022-12-03 11:31:02MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":4.22,"energy":467.16,"indicator_mode":"off/on","linkquality":27,"power":984,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'
Info 2022-12-03 11:31:12MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":4.27,"energy":467.16,"indicator_mode":"off/on","linkquality":30,"power":984,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'
Info 2022-12-03 11:31:16MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":4.27,"energy":467.16,"indicator_mode":"off/on","linkquality":27,"power":996,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'
Info 2022-12-03 11:31:26MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":4.27,"energy":467.17,"indicator_mode":"off/on","linkquality":33,"power":996,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'
Info 2022-12-03 11:31:33MQTT publish: topic 'zigbee2mqtt/Socket 06', payload '{"child_lock":"LOCK","current":4.32,"energy":467.17,"indicator_mode":"off/on","linkquality":48,"power":996,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":230}'`

frabnet, 2022-12-05:
I got the same problem with a [TS011F_din_smart_relay](https://www.zigbee2mqtt.io/devices/TS011F_din_smart_relay.html#tuya-ts011f_din_smart_relay).

Reporting 0 w every  ~5 minutes.

Adapter: zig-a-zig-ah! CC2652R with zStack3x0 20220219
Zigbee2mqtt:  1.28.4

[image]

sakirsek, 2022-12-17:
i have the same problem. It does not give accurate and stable values. I have automations that process according to these value changes and I am constantly having problems.

[image]

sakirsek, 2022-12-27:
> i have the same problem. It does not give accurate and stable values. I have automations that process according to these value changes and I am constantly having problems.
> 
> Device: TS011F_plug_1 Adapter: sonoff Zigbee 3.0 Dongle-E
> 
> [image]

Debug 2022-12-27 09:54:54Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2022-12-27 09:54:54MQTT publish: topic 'zigbee2mqtt/Charger Plug', payload '{"child_lock":"UNLOCK","current":1.05,"energy":0.53,"indicator_mode":"off/on","last_seen":"2022-12-27T06:54:53.727Z","linkquality":180,"power":5,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":224}'

Debug 2022-12-27 09:55:06Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
Info 2022-12-27 09:55:06MQTT publish: topic 'zigbee2mqtt/Charger Plug', payload '{"child_lock":"UNLOCK","current":1.05,"energy":0.53,"indicator_mode":"off/on","last_seen":"2022-12-27T06:55:06.004Z","linkquality":180,"power":5,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":224}'

Debug 2022-12-27 09:56:30Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
Info 2022-12-27 09:56:30MQTT publish: topic 'zigbee2mqtt/Charger Plug', payload '{"child_lock":"UNLOCK","current":0,"energy":0.53,"indicator_mode":"off/on","last_seen":"2022-12-27T06:56:29.741Z","linkquality":160,"power":5,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":224}'

Debug 2022-12-27 09:56:35Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":53}' from endpoint 1 with groupID 0
Info 2022-12-27 09:56:35MQTT publish: topic 'zigbee2mqtt/Charger Plug', payload '{"child_lock":"UNLOCK","current":1.05,"energy":0.53,"indicator_mode":"off/on","last_seen":"2022-12-27T06:56:34.742Z","linkquality":172,"power":5,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":224}'

Debug 2022-12-27 09:57:10Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,53]}' from endpoint 1 with groupID 0
Info 2022-12-27 09:57:10MQTT publish: topic 'zigbee2mqtt/Charger Plug', payload '{"child_lock":"UNLOCK","current":1.05,"energy":0.53,"indicator_mode":"off/on","last_seen":"2022-12-27T06:57:10.025Z","linkquality":188,"power":5,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":224}'

Debug 2022-12-27 09:57:10Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":5,"rmsCurrent":53,"rmsVoltage":224}' from endpoint 1 with groupID 0
Info 2022-12-27 09:57:10MQTT publish: topic 'zigbee2mqtt/Charger Plug', payload '{"child_lock":"UNLOCK","current":1.05,"energy":0.53,"indicator_mode":"off/on","last_seen":"2022-12-27T06:57:10.108Z","linkquality":196,"power":5,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":224}'

Debug 2022-12-27 09:57:38Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
Info 2022-12-27 09:57:38MQTT publish: topic 'zigbee2mqtt/Charger Plug', payload '{"child_lock":"UNLOCK","current":1.05,"energy":0.53,"indicator_mode":"off/on","last_seen":"2022-12-27T06:57:38.481Z","linkquality":204,"power":5,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":224}'

Debug 2022-12-27 10:00:21Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
Info 2022-12-27 10:00:21MQTT publish: topic 'zigbee2mqtt/Charger Plug', payload '{"child_lock":"UNLOCK","current":1.05,"energy":0.53,"indicator_mode":"off/on","last_seen":"2022-12-27T07:00:21.220Z","linkquality":204,"power":5,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":224}'

Debug 2022-12-27 10:00:59Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
Info 2022-12-27 10:00:59MQTT publish: topic 'zigbee2mqtt/Charger Plug', payload '{"child_lock":"UNLOCK","current":0,"energy":0.53,"indicator_mode":"off/on","last_seen":"2022-12-27T07:00:58.771Z","linkquality":196,"power":5,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":224}'

Debug 2022-12-27 10:01:04Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":53}' from endpoint 1 with groupID 0
[... cut ...]
Debug 2022-12-27 10:07:15Received Zigbee message from 'Charger Plug', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,53]}' from endpoint 1 with groupID 0
Info 2022-12-27 10:07:15MQTT publish: topic 'zigbee2mqtt/Charger Plug', payload '{"child_lock":"UNLOCK","current":1.05,"energy":0.53,"indicator_mode":"off/on","last_seen":"2022-12-27T07:07:15.389Z","linkquality":188,"power":5,"power_outage_memory":"restore","state":"ON","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":225}'

sakirsek, 2023-01-02:
The problem still exists after Zigbee2MQTT version1.29.0

Koenkk, 2023-01-02:
@sakirsek I don't think this can be fixed by a z2m update, looks like some hardware issue of the plug

sakirsek, 2023-01-02:
> @sakirsek I don't think this can be fixed by a z2m update, looks like some hardware issue of the plug

I'm so sorry to hear. According to your experience, which power monitoring socket is the most stable for z2m?

Koenkk, 2023-01-02:
@sakirsek I have good experiences with the Xiaomi/Aqara/Innr ones.

sakirsek, 2023-01-02:
> Innr

Thank you for inf

> @sakirsek I have good experiences with the Xiaomi/Aqara/Innr ones.

Thank you for advice. I think you can close issue.

frabnet, 2023-01-03:
Maybe, before closing the issue (this one and also #13246 and #13222 seems related), I think a good idea would be write into "Supported Devices" (_TZ3000_cehuw1lw _TZ3000_gvn91tmx TS011F_din_smart_relay S011F_plug_1) that some issues exists about 0 w reports, just to warn new buyers.

Dema, 2023-01-05:
Mine started reporting 0KW after issuing command specified here https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#reset-energy

kecksk0enig, 2023-01-06:
> Mine started reporting 0KW after issuing command specified here https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#reset-energy. It did reset counter to zero but it never changed to any number since.

Thats interesting, how long did you run the plug before resetting the counter?
I'm pretty sure I did not reset the counter prior.

Dema, 2023-01-06:
It was showing about 67kW when I hooked up my gas boiler to it, wanted to measure it, reset it and the reading never changed again, that was about a month ago. The boiler consumes about 30-70W constantly

TarjeiD, 2023-01-08:
One of my plugs from BlitzWolf (_TZ3000_amdymr7l) is disassembled and found to use the BL0937 chip for energy measurement. The BL0937 outputs high frequency pulses (38uS typical) for current, voltage and active power. The frequency of the output determines the value.
The application for the BlitzWolf plug is probably using the API described [here](https://developer.tuya.com/en/docs/iot-device-dev/Electricity-Statistics?id=Kaomiw064atlm) and rely on the Tuya implementation of the driver, which might be where the problem occurs. Either the driver responds with a 0 value or the application does not handle the response correctly when no value is available.
As seen from the flowchart, the cycle timer used by the driver for voltage and power is 30s, while it is 10min for consumption (energy). Using the following settings, the plug is found to produce a lot less drops in values compared to the default values set by Zigbee2MQTT.
```
Settings for rmsCurrent, rmsVoltage og activePower
Min rep interval 60
Max rep interval 3600
Min rep change   0
```
```
Settings for currSummDelivered
Min rep interval 600
Max rep interval 3600
Min rep change   0
```
I have another plug from Aubess (_TZ3000_gjnozsaz) which by inspection is found to be using the BL0942 chip for energy measurement. The BL0942 outputs data through an SPI interface, which easily interfaces an MCU.
The Aubess plug (_TZ3000_gjnozsaz) works flawlessly with no drops in any values.

kecksk0enig, 2023-01-10:
Thanks for the input, im trying it out, however prolonging the reporting intervals also reduces the resolution of the data and thus reducing the possibility of the wrong reporting. The core underlying problem is not touched by this.

TarjeiD, 2023-01-11:
The core underlying problem is the Tuya driver (plug FW). My fix only reduce the number of false values while still getting valid and representative data.
I am monitoring a wall-mounted heater which turns on/off by its own thermostat. As seen from the figure below, it does so fairly often. To have a cleaner view of the data I have created a filter for the power data.
```
- platform: filter
  name: " tuya_contact4_power_filtered"
  entity_id: sensor.tuya_contact4_power
  filters:
    - filter: time_simple_moving_average
      window_size: "00:15"
      precision: 2
```
[image]
To reduce the load on the filter and have a cleaner view of the reported data, I have the following settings for the plug.
`Settings for rmsCurrent, rmsVoltage og activePower`
```
Min rep interval 30
Max rep interval 3600
Min rep change   0
```
`Settings for currSummDelivered`
```
Min rep interval 600
Max rep interval 3600
Min rep change   0
```
As seen from the figure below, the data is lot cleaner and the reported data takes into account the duration of on state in the 30s interval when reporting power.
[image]

closed 2023-02-18 by github-actions[bot]