Koenkk/zigbee2mqtt #15334: [New device support]: Tuya 4 button Scene Switch - Wired

evangelo18, 2022-12-03:
### Link

https://www.alibaba.com/product-detail/Tuya-Zigbee-smart-4-button-scene_62368109165.html?spm=a2700.shop_plser.41413.16.2c55551f7AcP7F

### Database entry

{"id":63,"type":"Router","ieeeAddr":"0xa4c13861b2b8ce2c","nwkAddr":16554,"manufId":4417,"manufName":"_TZ3002_z0ywd8c8","powerSource":"Mains (single phase)","modelId":"TS0024","epList":[1,2,242,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":4,"inClusterList":[4,5,6,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":4,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":4,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":4,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1669894681472,"defaultSendRequestWhen":"immediate"}

### Comments

is there any existing tuya device converter that would work with this switch?

vignez2991, 2022-12-23:
@evangelo18  Did you Found the support for this device??

evangelo18, 2022-12-23:
> @evangelo18 Did you Found the support for this device??

Not yet unfortunately!

evangelo18, 2023-02-16:
Anyone able to assist?

closed 2023-04-02 by github-actions[bot]