Koenkk/zigbee2mqtt #15337: 1.28.4 added UNSUPPORTED_ATTRIBUTE to _TZ3210_9q49basr

stoinov, 2022-12-03:
### What happened?

After updating to 1.28.4 I saw there is new option for my [TuYa TS0501B](https://www.zigbee2mqtt.io/devices/TS0501B.html#tuya-ts0501b) (manufacturer _TZ3210_9q49basr) to set the power_on_behavior.
Upon testing I immediately get:
```
Error 2022-12-03 17:11:50Publish 'get' 'power_on_behavior' to 'test/lights/zig027' failed: 'Error: Read 0xa4c1382xxxxxxxxx/1 genOnOff(["startUpOnOff"],
{"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```
In the release notes I saw that #14287 enabled this to all devices that supports it, but not sure this is the case with mine.

### What did you expect to happen?

Either do not show power_on_behavior or show it and be able to use it.

### How to reproduce it (minimal and precise)

Join _TZ3210_9q49basr to Z2M

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

zzh

Koenkk, 2022-12-04:
My TS0505B also does not support this, so I guess the 504, 503 and 502 also don't. Removed it.

Koenkk, 2022-12-04:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0501B'],
    model: 'TS0501B',
    description: 'Zigbee light',
    vendor: 'TuYa',
    toZigbee: extend.light_onoff_brightness({disablePowerOnBehavior: true}).toZigbee.concat([tuya.tz.power_on_behavior]),
    fromZigbee: extend.light_onoff_brightness({disablePowerOnBehavior: true}).fromZigbee.concat([tuya.fz.power_on_behavior]),
    exposes: extend.light_onoff_brightness({disablePowerOnBehavior: true}).exposes.concat([tuya.exposes.powerOnBehavior()]),
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

closed 2023-01-11 by github-actions[bot]