Koenkk/zigbee2mqtt #15339: Moes ZTS-US_4gang

thanhvn693, 2022-12-03:
### Link

https://www.moeshouse.com/products/zigbee-wall-touch-smart-light-switch-with-neutral-wire-no-neutral-wire-no-capacitor-needed-3-way-us?variant=39797108965457

### Database entry

{"id":9,"type":"EndDevice","ieeeAddr":"0xdc8e95fffe4dbfcd","nwkAddr":45844,"manufId":4098,"manufName":"_TZ3000_mrduubod","powerSource":"Mains (single phase)","modelId":"TS0014","epList":[1,2,3,4,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0014","manufacturerName":"_TZ3000_mrduubod","stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c16690","endpointID":1}],"configuredReportings":[],"meta":{"scenes":{}}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c16690","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c16690","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":256,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c16690","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1871679396},"lastSeen":1670082420134,"defaultSendRequestWhen":"immediate"}

### Comments

i tried to follow supporting new device page, the device showed supported with full functions, but the physical device didn't sync
[image]

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;

const exposesLocal = {
    hour: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('h').withValueMin(0).withValueMax(23),
    minute: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('m').withValueMin(0).withValueMax(59),
    program_temperature: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('°C')
        .withValueMin(5).withValueMax(35).withValueStep(0.5),
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS0014', manufacturerName: '_TZ3000_mrduubod'}],
        model: 'ZTS-US_4gang',
        vendor: 'Moes',
        description: 'Wall touch light switch (4 gang)',
        exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
            e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
            e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
            e.switch().withEndpoint('l4').setAccess('state', ea.STATE_SET),
            exposes.enum('indicate_light', ea.STATE_SET, Object.values(tuya.moesSwitch.indicateLight))
                .withDescription('Indicator light status'),
            exposes.enum('power_on_behavior', ea.STATE_SET, Object.values(tuya.moesSwitch.powerOnBehavior))
                .withDescription('Controls the behavior when the device is powered on')],
        fromZigbee: [fz.ignore_basic_report, fz.tuya_switch, fz.moes_switch],
        toZigbee: [tz.tuya_switch_state, tz.moes_switch],
        onEvent: tuya.onEventSetLocalTime,
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            // Endpoint selection is made in tuya_switch_state
            return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1};
        },
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(3)) await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(4)) await reporting.bind(device.getEndpoint(4), coordinatorEndpoint, ['genOnOff']);
            // Reports itself as battery which is not correct: https://github.com/Koenkk/zigbee2mqtt/issues/6190
            device.powerSource = 'Mains (single phase)';
            device.save();
        },
    },
];
```

thanhvn693, 2022-12-07:
I have changed External converter, switches work well, however, indicate_light and power_on_behavior don't live.

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;

module.exports = [
    {
        fingerprint: [{modelID: 'TS0014', manufacturerName: '_TZ3000_mrduubod'}],
        model: 'ZTS-US_4gang',
        vendor: 'Moes',
        description: 'Wall touch light switch (4 gang)',
        exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
            e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
            e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
            e.switch().withEndpoint('l4').setAccess('state', ea.STATE_SET),
            exposes.enum('indicate_light', ea.STATE_SET, Object.values(tuya.moesSwitch.indicateLight))
                .withDescription('Indicator light status'),
            exposes.enum('power_on_behavior', ea.STATE_SET, Object.values(tuya.moesSwitch.powerOnBehavior))
                .withDescription('Controls the behavior when the device is powered on')],
        toZigbee: extend.switch().toZigbee.concat([tz.tuya_switch_type, tz.moes_switch, tz.tuya_switch_state, tz.on_off, tz.tuya_operation_mode]),
        fromZigbee: extend.switch().fromZigbee.concat([fz.ignore_basic_report, fz.tuya_switch, fz.moes_switch, fz.on_off, fz.tuya_on_off_action]),
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            // Endpoint selection is made in tuya_switch_state
            return {'l1': 1, 'l2': 2, 'l3': 3, 'l4': 4};
        },
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(3)) await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(4)) await reporting.bind(device.getEndpoint(4), coordinatorEndpoint, ['genOnOff']);
            device.powerSource = 'Mains (single phase)';
            device.save();
        },
    },
];`

thanhvn693, 2022-12-08:
Without external converter, this device is interviewed completely, as model TS0014 and support status is not supported.
 [image]

thanhvn693, 2022-12-08:
I also have a similar 3 gang moes (model: ZTS-US_3gang). It is recognised as model: TS0013, TuYa, but that 3 gang works perfectly with full functions!

Koenkk, 2022-12-08:
I see, added the device now!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-08 by Koenkk

thanhvn693, 2022-12-12:
After updating addon, my device can work as TS0014 (with no backlight mode). I have added backlightMode in extend line TS0014 (copied from tuya.js), saved in extenal converter and it can works with full exposes: switch (state), power_on_behavior, backlight_mode, linkquality. I don't know wheather orginal TS0014 has this expose.

```javascript
extend: tuya.extend.switch({backlightMode: true, endpoints: ['l1', 'l2', 'l3', 'l4']}),'

thanhvn693, 2022-12-12:
After updating addon, my device can work as TS0014 (with no backlight mode). I have added backlightMode in extend line of TS0014 (copied from tuya.js), saved in extenal converter and it can works with full exposes: switch (state), power_on_behavior, backlight_mode, linkquality. I don't know wheather orginal TS0014 has this expose.

```javascript
extend: tuya.extend.switch({backlightMode: true, endpoints: ['l1', 'l2', 'l3', 'l4']}),'

thanhvn693, 2022-12-12:
After updating addon, my device can work as TS0014 (with no **backlight mode**). I have added backlightMode in extend line TS0014 (copied from tuya.js), saved in extenal converter and it can work with full exposes: switch (state), power_on_behavior, **backlight_mode**, linkquality. I don't know wheather orginal TS0014 has this expose.

```javascript
extend: tuya.extend.switch({backlightMode: true, endpoints: ['l1', 'l2', 'l3', 'l4']}),'

thanhvn693, 2022-12-12:
After updating addon, my device can work as TS0014 (with no **backlight mode**). I have added backlightMode in extend line of TS0014 (copied from tuya.js), saved in extenal converter and it can work with full exposes: switch (state), power_on_behavior, **backlight_mode**, linkquality. I don't know wheather orginal TS0014 has this expose.

```javascript
extend: tuya.extend.switch({backlightMode: true, endpoints: ['l1', 'l2', 'l3', 'l4']}),'

Koenkk, 2022-12-12:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)