Koenkk/zigbee2mqtt #15340: [New device support]:

kamms, 2022-12-03:
### Link

https://www.aliexpress.com/item/1005003432430621.html?spm=a2g0o.productlist.main.1.3162793eId0Oh1&algo_pvid=5385e072-9bb0-4a8e-9b70-ec8a48564a06&algo_exp_id=5385e072-9bb0-4a8e-9b70-ec8a48564a06-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000025779660263%22%7D&pdp_npi=2%40dis%21EUR%2132.2%2114.81%21%21%21%21%21%4021227d8316700919742394680d076e%2112000025779660263%21sea&curPageLogUid=c5MYAxuzEGeL

### Database entry

{"id":35,"type":"EndDevice","ieeeAddr":"0xa4c1386da1304880","nwkAddr":11568,"manufId":4417,"manufName":"_TZ3000_iszegwpd","powerSource":"Battery","modelId":"TS0046","epList":[1,2,3,4,5,6],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[1,6,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"epId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"epId":6,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1670092476482,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to add support, but I'm extremely new to doing anything under the hood on zigbee2mqtt. I followed the adding support guide, but got stumped as to what to do next for a 6-button remote. 

I then tried to monkey the configuration from Tuya TS0044 4-button remote, however the events I get do not distinguish between different buttons. The attached log file contains the logs from events captured by first pressing a button once, then twice, then moving on to the next button from button 1 to button 6. Except for the double click of button 5, since I fumbled a bit.
[6-button.log](https://github.com/Koenkk/zigbee2mqtt/files/10147176/6-button.log)

I don't know how to continue from here. I don't really understand what I'm doing

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;

const definition = {
        zigbeeModel: ['TS0046'],
        model: 'TS0046',
        vendor: '_TZ3000_iszegwpd',
        description: 'Wireless switch with 6 buttons',
        whiteLabel: [{vendor: 'LoraTap', model: 'TS0046'}],
        fromZigbee: [fz.tuya_on_off_action, fz.battery],
        exposes: [e.battery(), e.action(['1_single', '1_double', '1_hold', '2_single', '2_double', '2_hold',
            '3_single', '3_double', '3_hold', '4_single', '4_double', '4_hold',
            '5_single', '5_double', '5_hold','6_single', '6_double', '6_hold'])],
        toZigbee: [],
        configure: tuya.configureMagicPacket,
        /*
         * reporting.batteryPercentageRemaining removed as it was causing devices to fall of the network
         * every 1 hour, with light flashing when it happened, extremely short battery life, 2 presses for
         * action to register: https://github.com/Koenkk/zigbee2mqtt/issues/8072
         * Initially wrapped in a try catch: https://github.com/Koenkk/zigbee2mqtt/issues/6313
         */
    }

module.exports = definition;
```

Koenkk, 2022-12-04:
Try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    tuya_on_off_action: {
        cluster: 'genOnOff',
        type: 'raw',
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg, model, msg.data[1])) return;
            const clickMapping = {0: 'single', 1: 'double', 2: 'hold'};
            let buttonMapping = null;
            if (model.model === 'TS0042') {
                buttonMapping = {1: '1', 2: '2'};
            } else if (model.model === 'TS0043') {
                buttonMapping = {1: '1', 2: '2', 3: '3'};
            } else if (['TS0044', 'YSR-MINI-Z', 'TS004F'].includes(model.model)) {
                buttonMapping = {1: '1', 2: '2', 3: '3', 4: '4'};
            } else if (['TS0046'].includes(model.model)) {
                buttonMapping = {1: '1', 2: '2', 3: '3', 4: '4', 5: '5', 6: '6'};
            }
            const button = buttonMapping ? `${buttonMapping[msg.endpoint.ID]}_` : '';
            // Since it is a non standard ZCL command, no default response is send from zigbee-herdsman
            // Send the defaultResponse here, otherwise the second button click delays.
            // https://github.com/Koenkk/zigbee2mqtt/issues/8149
            msg.endpoint.defaultResponse(0xfd, 0, 6, msg.data[1]).catch((error) => {});
            return {action: `${button}${clickMapping[msg.data[3]]}`};
        },
    }
}

const definition = {
        zigbeeModel: ['TS0046'],
        model: 'TS0046',
        vendor: '_TZ3000_iszegwpd',
        description: 'Wireless switch with 6 buttons',
        whiteLabel: [{vendor: 'LoraTap', model: 'TS0046'}],
        fromZigbee: [fzLocal.tuya_on_off_action, fz.battery],
        exposes: [e.battery(), e.action(['1_single', '1_double', '1_hold', '2_single', '2_double', '2_hold',
            '3_single', '3_double', '3_hold', '4_single', '4_double', '4_hold',
            '5_single', '5_double', '5_hold','6_single', '6_double', '6_hold'])],
        toZigbee: [],
        configure: tuya.configureMagicPacket,
        /*
         * reporting.batteryPercentageRemaining removed as it was causing devices to fall of the network
         * every 1 hour, with light flashing when it happened, extremely short battery life, 2 presses for
         * action to register: https://github.com/Koenkk/zigbee2mqtt/issues/8072
         * Initially wrapped in a try catch: https://github.com/Koenkk/zigbee2mqtt/issues/6313
         */
    }

module.exports = definition;

```

kamms, 2022-12-04:
That works wonderfully, thank you! 
Only modification I had to make was 
`utils.hasAlreadyProcessedMessage`

kamms, 2022-12-04:
Sorry, I mean added the utils. prefix

Koenkk, 2022-12-04:
Thanks, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-04 by Koenkk

giostefax89, 2023-02-05:
> Prova con:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const fzLocal = {
>     tuya_on_off_action: {
>         cluster: 'genOnOff',
>         type: 'raw',
>         convert: (model, msg, publish, options, meta) => {
>             if (hasAlreadyProcessedMessage(msg, model, msg.data[1])) return;
>             const clickMapping = {0: 'single', 1: 'double', 2: 'hold'};
>             let buttonMapping = null;
>             if (model.model === 'TS0042') {
>                 buttonMapping = {1: '1', 2: '2'};
>             } else if (model.model === 'TS0043') {
>                 buttonMapping = {1: '1', 2: '2', 3: '3'};
>             } else if (['TS0044', 'YSR-MINI-Z', 'TS004F'].includes(model.model)) {
>                 buttonMapping = {1: '1', 2: '2', 3: '3', 4: '4'};
>             } else if (['TS0046'].includes(model.model)) {
>                 buttonMapping = {1: '1', 2: '2', 3: '3', 4: '4', 5: '5', 6: '6'};
>             }
>             const button = buttonMapping ? `${buttonMapping[msg.endpoint.ID]}_` : '';
>             // Since it is a non standard ZCL command, no default response is send from zigbee-herdsman
>             // Send the defaultResponse here, otherwise the second button click delays.
>             // https://github.com/Koenkk/zigbee2mqtt/issues/8149
>             msg.endpoint.defaultResponse(0xfd, 0, 6, msg.data[1]).catch((error) => {});
>             return {action: `${button}${clickMapping[msg.data[3]]}`};
>         },
>     }
> }
> 
> const definition = {
>         zigbeeModel: ['TS0046'],
>         model: 'TS0046',
>         vendor: '_TZ3000_iszegwpd',
>         description: 'Wireless switch with 6 buttons',
>         whiteLabel: [{vendor: 'LoraTap', model: 'TS0046'}],
>         fromZigbee: [fzLocal.tuya_on_off_action, fz.battery],
>         exposes: [e.battery(), e.action(['1_single', '1_double', '1_hold', '2_single', '2_double', '2_hold',
>             '3_single', '3_double', '3_hold', '4_single', '4_double', '4_hold',
>             '5_single', '5_double', '5_hold','6_single', '6_double', '6_hold'])],
>         toZigbee: [],
>         configure: tuya.configureMagicPacket,
>         /*
>          * reporting.batteryPercentageRemaining removed as it was causing devices to fall of the network
>          * every 1 hour, with light flashing when it happened, extremely short battery life, 2 presses for
>          * action to register: https://github.com/Koenkk/zigbee2mqtt/issues/8072
>          * Initially wrapped in a try catch: https://github.com/Koenkk/zigbee2mqtt/issues/6313
>          */
>     }
> 
> module.exports = definition;
> ```

Good morning,
I used the mentioned converter but it gives me this error when I press the buttons.

Error 2023-02-05 09:45:43Exception while calling fromZigbee converter: hasAlreadyProcessedMessage is not defined}

How can I fix?

kamms, 2023-02-05:
Replace hasAlreadyProcessedMessage with utils.hasAlreadyProcessedMessage

chapeau84, 2023-11-17:
Hi,
i bought this device and i got one with model number: SS9600ZB-V2 (note the V2).
When i add it it gets added but without any "actions" or "battery". Just an empty box.
I guess V2 is not added to zigbee2mqtt? How to fix this?