Koenkk/zigbee2mqtt #15347: Eurotronic SPZB0001 problem after upgrading zigbee2mqtt

Maurizion, 2022-12-04:
### What happened?

Hundreds of messages like this in the log: 

Exception in handle_state_message_received when handling msg on 'zigbee2mqtt/Sala': '{"battery":127.5,"battery_low":false,"boost":false,"child_protection":false,"current_heating_setpoint":7,"error_status":0,"linkquality":0,"local_temperature":18,"local_temperature_calibration":null,"mirror_display":false,"occupied_heating_setpoint":7,"pi_heating_demand":100,"running_state":"heat","system_mode":"auto","trv_mode":null,"unoccupied_heating_setpoint":16,"update":{"installed_version":"22190930","state":null},"update_available":null,"valve_position":null,"window_open":false}' Traceback (most recent call last): File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper msg_callback(msg) File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/mqtt/update.py", line 191, in handle_state_message_received if "installed_version" in json_payload: TypeError: argument of type 'int' is not iterable 

### What did you expect to happen?

this to be fixed ....

### How to reproduce it (minimal and precise)

I have upgraded zigbee2mqtt to version 1.28.4 and those message started to appears.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

CC2652R_coordinator_20221102.zip

### Adapter

zzh!

itkama, 2022-12-05:
[image]
This is the only messages I see from my SPZB0001

Is this maybe related to the new function that makes update entities available in Home Assistant?

Koenkk, 2022-12-05:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/15290

closed 2022-12-05 by Koenkk