Koenkk/zigbee2mqtt #15348: TRADFRIbulbGU10WS345lm returning unsupported

thejman1989, 2022-12-04:
### What happened?

I Paired two new Tradfri bulbs of the following [type ](https://www.zigbee2mqtt.io/devices/LED1923R5.html). They pair but show up as unsupported: 

[image]

The type number is "TRADFRIbulbGU10WS345lm" which can be found in the [ikea.js ](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/ikea.js) so I would expect them to work. I have repaired them, but to no avail. Another post said that for them it automatically went to supported after some time but it has been about 18 hours now. Any other thoughts? I dont mind fixing the code, but then I would need to know what's missing. 

### What did you expect to happen?

The light bulbs would connect to zigbee2Mqtt

### How to reproduce it (minimal and precise)

Get TRADFRIbulbGU10WS345lm light bulb, pair and wiat. 

### Zigbee2MQTT version

1.18.1-1

### Adapter firmware version

20201026

### Adapter

Electrolama zzh!

### Debug log

Received message from unsupported device with Zigbee model 'TRADFRIbulbGU10WS345lm' and manufacturer name 'IKEA of Sweden'
warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2022-12-04:
You are using an outdated z2m version, this device is supported.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2022-12-04 by Koenkk

thejman1989, 2022-12-04:
Sorry, I did check everything for updates but as you commented, my zigbee2mqtt was showing it was on latest while in fact it was far behind! Thanks the bulbs are working now, and thanks in general for zigbee2mqtt!